ESPresense/ESPresense #289: "distance":0 after update to v2.0.27

moblsu, 2022-02-16:
**Describe the bug**

Due to reboot error, updated devices to v2.0.27 via web automatic install. Dropdown "Generic ESP 32: ESP 32" came back as not found. Used "Generic ESP 32: ESP 32 Verbose Logging"

**Config**

- Version: 2.0.27
- Active scan enabled: tried both, yes and no
- Include filter: none
- Exclude filter: none
- Query filter: none

**Hardware**

- Firmware flavor:
- Device (be specific): Installed on 4 of these units: HiLetgo ESP-WROOM-32 ESP32 ESP-32S Development Board 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller Processor Integrated with Antenna RF AMP Filter AP STA for Arduino IDE (https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

**Logs**

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (**:**:**:**:**)...
I (74) wifi:wifi driver task: 3ffd494c, prio:23, stack:3584, core=0
I (368) wifi:wifi firmware version: dc30037
I (369) wifi:config NVS flash: enabled
I (369) wifi:config nano formating: disabled
I (369) wifi:Init data frame dynamic rx buffer num: 32
I (374) wifi:Init management frame dynamic rx buffer num: 32
I (379) wifi:Init management short buffer num: 32
I (384) wifi:Init dynamic tx buffer num: 32
I (388) wifi:Init static rx buffer size: 1600
I (392) wifi:Init static rx buffer num: 16
I (395) wifi:Init dynamic rx buffer num: 32
I (500) wifi:mode : sta (c8:c9:a3:c5:a8:b0)
Connecting to WiFi SSID '**************'..............I (2567) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2569) wifi:state: init -> auth (b0)
I (2582) wifi:state: auth -> assoc (0)
I (2588) wifi:state: assoc -> run (10)
I (2606) wifi:connected with **************, aid = 23, channel 6, BW20, bssid = ec:c3:02:f3:a9:44
I (2606) wifi:security type: 3, phy: bgn, rssi: -41
.I (2659) wifi:pm start, type: 1

I (2777) wifi:AP's beacon interval = 102400 us, DTIM period = 3
192.168.1.74
Version:    v2.0.27
IP address:   192.168.1.74
DNS address:  192.168.1.254
Hostname:     espresense-kn_p_esp
Room:         kn_p_esp
MQTT server:  192.168.1.105:1883
Max Distance: 16.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      
Exclude:      
Verbose | a4c138e1212e                                              MD: 0188ec000101
[... 51 lines cut ...]
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | 7ca6b09cf72c                                              MD: 0288ec00020101
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | 7ca6b09cf72c                                              MD: 0288ec00020101
Verbose | fa0dac76f287                                              SD: 0xfdf7/01bcef8938c84d58ba81261e777038f7af0000000003
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | msft:cdp:0902                                             MD: 06000109200241318b98beafb97fbe3fd5e799f008d8af4f6df3855d76
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | md:4c00:25                                                MD: 004c02152686f39cbada4658854aa62e7e5e8b8d00010000c9
Verbose | 7ca6b09cf72c                                              MD: 0288ec00020101
Verbose | fa0dac76f287                                              SD: 0xfdf7/01bcef8938c84d58ba81261e777038f7af0000000003
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | msft:cdp:0902                                             MD: 06000109200241318b98beafb97fbe3fd5e799f008d8af4f6df3855d76
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | fa0dac76f287                                              SD: 0xfdf7/01bcef8938c84d58ba81261e777038f7af0000000003
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | 7ca6b09cf72c                                              MD: 0288ec00020101
Verbose | msft:cdp:0902                                             MD: 06000109200241318b98beafb97fbe3fd5e799f008d8af4f6df3855d76
Verbose | a4c138e1212e                                              MD: 0188ec000101
Verbose | samsung:5cc1d783c59b                                      MD: 7500420401207e200f0002014101010001000000000000000000

**Screenshots**

[image]

lelalexi, 2022-02-16:
Hi! Im having the same issue with ESP32 Nodemcu board, using ESP32 Flavour

dewoert, 2022-02-16:
Same here.
All my devices (iPhones etc) are now reported at 1 location (ESP32 with v2.0.27)
(My other ESP32's are still at v2.0.25)

indomitu5, 2022-02-16:
not sure if its helpful or not, my c++ skills are absolute garbage, but i did notice that both `0` reported values (distance and speed) both seem to be using `round(output.value.XXXX)` in `lib/BleFingerprint/BleFingerprint.cpp` on lines 427 and 428:

```cpp
    (*doc)[F("distance")] = round(output.value.position * 100.0f) / 100.0f;
    (*doc)[F("speed")] = round(output.value.speed * 1e7f) / 10.0f;
```

jimmyeao, 2022-02-16:
I see the values in MQTT, distance and raw usually match but not anymore:
`lounge = {"id":"apple:1005:9-24","disc":"0a1856bc81","idType":35,"rssi@1m":-65,"rssi":-70,"mac":"60afec29fa75","raw":1.39,"distance":0,"speed":0}`

gooselk, 2022-02-16:
I can confirm the behaviour on all my ESP-WROOM-32

dewoert, 2022-02-16:
**Fixed** for me with **v2.0.28**

closed 2022-02-16 by moblsu