ESPresense/ESPresense #291: espresense not automatically updating

jimmyeao, 2022-02-16:
**Describe the bug**

espresense not automatically updating - have tried restarting the device several times (including cold boot)
**Config**
[image]

- Version:ESPResense Version: v2.0.26
- Active scan enabled: No
- Include filter: None
- Exclude filter: None
- Query filter: None

**Hardware**

- Firmware flavor:
- Device (be specific): ESP32

**Logs**

No logs, device hard to reach

cbhiii, 2022-02-16:
Same - v2.0.26 kept them all in a reboot so they aren't checking for the latest version.

hellcry37, 2022-02-22:
this is solved in later versions you can close the issue

closed 2022-02-22 by jimmyeao