ESPresense/ESPresense #294: A list of Apple Identifiers...

ctschach, 2022-02-16:
Just stumbled cross this project a couple of days ago. 

I currently try to understand how I can identify different Apple devices. With having things like HomePods, iPads, iPhones, Macs and Watches at home, I find it pretty hard to identify the correct devices. I try querying the devices, but this does not succeed.

I See the following Apple IDs:

apple:1005:9-24
apple:1005:9-26
apple:1006:10-10
apple:1006:10-7
apple:1007:11-7

Can someone help me identifying the devices? Maybe we could add a list to the project, I'm pretty sure I'm not the only one....

FreakyBoy, 2022-02-16:
I had the same problem. 

I then added all devices as "Test-Device-X" in home assistant and then removed myself from ESP32 to see which device updates the distance.

Zipties, 2022-02-16:
The only reliable way is to physically touch the devices to the ESP and wait for the "near" and "left" messages in the terminal log. It also helps if you move to a less noisy rea of the house to do the setup.

duceduc, 2022-02-17:
Not sure if it will help, but this is my findings.

    apple:1005:9-24 [apple:watch3-3(series3)]
    apple:1005:9-26 [apple:watch5-9(series7)]
    apple:1006:10-10
    apple:1006:10-7
    apple:1007:11-7
    
    apple:1005:9-12 [apple:iphone10-1(iphone8) apple:iphone7-1(iphone6+) apple:iphone7-2(iphone6)]

ozett, 2022-02-18:
this seems to be the fingerprinting for apple:
https://github.com/ESPresense/ESPresense/blob/19cde52b2f8588c9c05454a09823168c13cbe6d5/lib/BleFingerprint/BleFingerprint.cpp#L259-L291

ozett, 2022-02-21:
the fingerprint ID is the same for different devices/mac-adresses.
how do i know which one is wich? 
that way with short-fingerprints on esppresence is not very helpful to identify devices...
or do i do it the wrong way?

[image]

DTTerastar, 2022-02-21:
Apple is big on privacy, so they thwart us as best they can.  There is always the room assistant app to assign a guid.

ozett, 2022-02-21:
> Apple is big on privacy, so they thwart us as best they can. There is always the room assistant app to assign a guid.

i wanted to use espresence with my node-red. never heard of room-assistant-app. but will google it.

but whats the use of the fingerprint-ID in the mqtt-topics,
if one can not divide devices, because they are all under the same short fingerprint-id.
that seems not only for apple the case, [also for fitbit and garmin](https://github.com/ESPresense/ESPresense/issues/260) (and may some others). is this maybe a desing-flaw of the fingerprinting? or intended?

ozett, 2022-02-21:
secret difficulties.. now i understand: mac adress changes... 
how to overcome?

[image]

Zipties, 2022-02-26:
> the devs should have a look 👀 ->
> 
> https://github.com/dalehumby/ESPHome-Apple-Watch-detection
> 
> edit: and the extension for iphones:
> 
> [dalehumby/ESPHome-Apple-Watch-detection#4](https://github.com/dalehumby/ESPHome-Apple-Watch-detection/issues/4)

I've used that project, and it is very good at consistently detecting devices, but it is behind in a lot of other ways

ozett, 2022-02-27:
i though i could help differentiate different devices from the same manufacturer in espresense.
still lacking this for some other brands besides apple.

but i am no coder... can only supply links...

dashdrum, 2022-03-08:
After I setup two ESP32 devices in different rooms, I walked the Apple devices from one room to the other to determine which was the matching apple entry.  Sneaker config?

closed 2022-05-14 by stale[bot]