ESPresense/ESPresense #295: v2.0.27 reboots continue, distance in MQTT reported as zero in HA

cbhiii, 2022-02-16:
**Describe the bug**

Reboots under the latest version continue every  seconds or so.

**Config**

- Version: 2.0.27
- Active scan enabled: yes
- Include filter: apple:1005:9-24 apple:1005:9-26
- Exclude filter:
- Query filter: apple:1005:9-24 apple:1005:9-26

**Hardware**

- Firmware flavor: 2.0.27 (or whatever your website gave my if you mean that)
- Device (be specific): ESP WROOM 32 (4 of them all doing the same)

**Logs**

1 of 2 logs -- From MQTT
 
```
Topic to subscribe to
espresense/rooms/office/#
 
Message 118 received on espresense/rooms/office at 12:19:
{
    "id": "apple:1005:9-24",
    "disc": "0e9821688b",
    "idType": 35,
    "rssi@1m": -65,
    "rssi": -64,
    "mac": "7a52f5040b28",
    "raw": 0.94,
    "distance": 0,
    "speed": 0
}
QoS: 0 - Retain: false
Message 117 received on espresense/rooms/office at 12:19:
{
    "id": "apple:1005:9-26",
    "disc": "26982fa4f3",
    "idType": 35,
    "rssi@1m": -65,
    "rssi": -87,
    "mac": "78322a4031b2",
    "raw": 4.25,
    "distance": 0,
    "speed": 0
}
QoS: 0 - Retain: false
```

2 of 2 logs -- Webpage terminal

```
Connected 115200 8N1...
1 Query | MAC: 44e14f79962b, ID: apple:1005:9-24                                              rssi -70
1 QryErr| MAC: 44e14f79962b, ID: apple:1005:9-24                                              rssi -70, try 9, retry after 30000ms
1 Query | MAC: 54d5456047f2, ID: apple:1005:9-26                                              rssi -90
1 QryErr| MAC: 54d5456047f2, ID: apple:1005:9-26                                              rssi -90, try 8, retry after 30000ms
1 Query | MAC: 44e14f79962b, ID: apple:1005:9-24                                              rssi -70
1 QryErr| MAC: 44e14f79962b, ID: apple:1005:9-24                                              rssi -70, try 10, retry after 30000ms
0 New   | MAC: 510d12260f5d, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-1177-0          
0 New   | MAC: 6a27d8510d55, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-391-0           
1 Query | MAC: 54d5456047f2, ID: apple:1005:9-26                                              rssi -88
1 QryErr| MAC: 54d5456047f2, ID: apple:1005:9-26                                              rssi -88, try 9, retry after 30000ms
1 Query | MAC: 44e14f79962b, ID: apple:1005:9-24                                              rssi -70
1 QryErr| MAC: 44e14f79962b, ID: apple:1005:9-24                                              rssi -70, try 11, retry after 30000ms
1 Query | MAC: 54d5456047f2, ID: apple:1005:9-26                                              rssi -88
[... 392 lines cut ...]
0 New   | MAC: 7752b92332fe, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-1180-0          
1 Query | MAC: 78322a4031b2, ID: apple:1005:9-26                                              rssi -88
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
1 QryErr| MAC: 78322a4031b2, ID: apple:1005:9-26                                              rssi -88, try 1, retry after 5000ms
0 New   | MAC: 4be58422ab99, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-394-0           
0 New   | MAC: 45f8fb64ce17, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-1895-0          
1 Query | MAC: 78322a4031b2, ID: apple:1005:9-26                                              rssi -88
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
1 QryErr| MAC: 78322a4031b2, ID: apple:1005:9-26                                              rssi -88, try 2, retry after 6000ms
1 Query | MAC: 78322a4031b2, ID: apple:1005:9-26                                              rssi -87
1 Model | MAC: 78322a4031b2, ID: apple:1005:9-26                                              Watch6,2
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
0 New   | MAC: 2f839c56cfe0, ID: msft:cdp:0902                                                3c10126338bddf29bf45145075c305b4091e0f50
0 New   | MAC: 5e93a9fcfb35, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-1828-0          
0 New   | MAC: 4d75da623cdf, ID: apple:1006:10-8                                              311eb5602fdf
0 New   | MAC: 2221a22e913a, ID: exp:20                                                       b277657de6e5ce9335ad445d588494023d2d8e5a
0 New   | MAC: 6ef78533e95d, ID: apple:1005:9-3                                               0618de484a
Not updating from (sc=302): https://github.com/ESPresense/ESPresense/releases/download/v2.0.27/esp32.bin
1 Del   | MAC: 610ffe9b9a12, ID: apple:1002:6-12                                              0300
1 Del   | MAC: 71d781af492b, ID: apple:1005:9-26                                              2998f87105
1 Del   | MAC: 7752b92332fe, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-1180-0          
1 Del   | MAC: 45f8fb64ce17, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-1895-0          
1 Del   | MAC: 4be58422ab99, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-394-0           

```

siiiilky, 2022-02-16:
Turn off query to stop the reboots.

There is already a bug for the incorrect distance.

cbhiii, 2022-02-16:
@siiiilky - Removing the text in the 'Query device ids for characteristics' field did not prevent future reboots. Are you talking about some other switch of place in the setup?

cbhiii, 2022-02-16:
v2.0.28 is out now, but I still see the device rebooting.

closed 2022-02-16 by cbhiii