ESPresense/ESPresense #297: Add RGB light support

chrisron95, 2022-02-16:
**Is your feature request related to a problem? Please describe.**
Considering I have one of these in each room, I'd love if I could add an RGB LED to my esp32 so that I could control it through Home Assistant for things such as notifications.

**Describe the solution you'd like**
Since several sensors have been added recently, I'm wondering if it would be possible to include support for an RGB LED such as WS2812B. Personally, I only want a single pixel so I don't need support for multiple pixels. ESPHome offers FastLED Clockless that looks like it would accomplish everything if it's possible to implement here: https://esphome.io/components/light/fastled.html

**Describe alternatives you've considered**
Alternatives would be either to have no status light in each room, or to add an additional esp device to each room for a status light.

**Additional context**
Add any other context or screenshots about the feature request here.

laemen, 2022-03-17:
Please also add support for the SK6812 RGB led on the [M5 Pico Stamp](https://docs.m5stack.com/en/core/stamp_pico)

DTTerastar, 2022-06-04:
Yes, we should also support multiple led strips since this has 2: https://t.co/9kLrbhq0RI

strid3r21, 2022-07-14:
This is something I'm very interested in as well. It would be a really handy feature to have to be able to flash different colors to an RGB LED on the ESP32 for various notifications.

PR #595 (2022-08-23): Selectable LEDs and Update Button
Fixes #297 and #533
commit 07228feb (2022-08-23): Selectable LEDs and Update Button (#595)

* RGB LEDs supported (closes #297) 
* Add button to manually update (closes #533)
* Manual Wifi setup via web installer (closes #593, #172)
* Add esp32c3-verbose flavor (#134)
* Make scan all wifi the default (#594)
* Improvements to M5Atom/M5StickCs
closed 2022-08-23 by DTTerastar