ESPresense/ESPresense #298: No devices reported after update to 2.0.28

gooselk, 2022-02-16:
After 2.0.26 with the boot loop bug and 2.0.27 with distance reported as 0, all my devices got updated to 2.0.28. Now non of my ESPs report the devices that I'm looking for. In the logs only one device got reported but that's not the device I'm looking for. With 2.0.25 all worked like a charm.

Config:
[image]

- Version: 2.0.28
- Active scan enabled: Yes
- Include filter: mifit:d0876800704c mifit:f1387628d25d
- Exclude filter: None
- Query filter: None

**Hardware**

- Firmware flavor: ESP32 - 2.0.28
- Device (be specific): ESP-WROOM-32

Logs:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (C8:C9:A3:C9:4B:BC)...
Connecting to WiFi SSID 'Olymp'...............192.168.2.62
Version:    v2.0.28
IP address:   192.168.2.62
DNS address:  192.168.2.1
Hostname:     espresense-schlafzimmer
Room:         Schlafzimmer
MQTT server:  192.168.2.4:1883
Max Distance: 16.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      mifit:d0876800704c mifit:f1387628d25d
Exclude:      
0 New   | MAC: 5f988206a0fe, ID: exp:20                                                       537998c5d9a7b531347ebe77503f1d38d7bc2cff

**Screenshots**

[image]

DeekB1983, 2022-02-16:
I also have the same issue, I can see my mifit devices being detected using the terminal but the messages never arrive in mqtt for the mifit devices.

xwox, 2022-02-17:
same here. Two MiFit 6 watches and one Samsung watch do not show up anymore.

gnome4711, 2022-02-17:
Same here! Almost no ble devices are found!

gnome4711, 2022-02-17:
only the white printed devices are reported (mqtt).

[image]

casaerreti, 2022-02-17:
I opened a new bug report but I see now is the same issue:

Hi guys,

fresh install to .28 on a ESP32 Wroom32. Been rock solid since then but I see (in MQTT Explore) way less devices than .26 and .27.

All the miband (3) are one. All the ble Filo Tag (3) are gone. All my neighbours devices are gone. The only trusty is ibeacon from HA app companion that is quite fast and reliable.

Config

Version: ESPResense Version: v2.0.28
Active scan enabled: yes
Include filter: -
Exclude filter: -
Query filter: -
Hardware

Firmware flavor: esp32
Device (be specific): azdelivery wroom32

DeekB1983, 2022-02-17:
The devices showing in red I believe are devices being excluded, I downgraded to V2.0.25 as that was the most stable for me, the query and exclude option I have prefilled to ignore Samsung devices and I can see they are showing in red:
[image]

Feels like a bug with the option to exclude sending to mqtt
[image]

ASNNetworks, 2022-02-17:
This explains why since a few days all my ESP32 beacons are starting to go haywire and tracking keeps missing my devices. Any way to downgrade?

DeekB1983, 2022-02-17:
See this discussion thread - https://github.com/ESPresense/ESPresense/discussions/293

I downloaded the V2.0.25 binary and then downloaded the ESP Flasher tool linked and uploaded the file.

The only bad thing is you need to do each one manually, would be good if there was an upload functionality on the web page for the device to downgrade in situations like this as it would save running round grabbing each one to be downgraded

wifispray, 2022-02-18:
How can you downgrade, I dont see an option in the web UI?

ASNNetworks, 2022-02-18:
> How can you downgrade, I dont see an option in the web UI?

It's explained on the website. Go to manual install and just download older release and flash: https://espresense.com/install

This reflashes it, so you need to setup the device again (connect through wifi ect) after flashing.

DeekB1983, 2022-02-18:
> es it, so you need to setup the device again (connect through wifi ect) after flashing.

If using windows you can download espflasher tool from here - https://github.com/esphome/esphome-flasher/releases

And you can download all the different esp.bin versions for espresense from here - https://github.com/ESPresense/ESPresense/releases

Using the flash tool is straight forward, connect usb cable and esp32 board to usb port, open the tool and choose the relevant com port, then browse to the bin file you downloaded and press flash

siiiilky, 2022-02-18:
> Unfortunately that doesn't work for me. I get an permission error on the com port. I tried different cables, drivers and computers. No luck yet...

Try holding boot button on board as you flash.

Regards

wifispray, 2022-02-18:
Which is the best version to use for now? I cant seem to see any of my tile devices when I use the latest?

hellcry37, 2022-02-18:
2.0.20 is good, it is after this that storm begin

gooselk, 2022-02-18:
> Try holding boot button on board as you flash.

Not pressing 'view log' before flashing - as stated in the install instructions - did the trick for me. Thanks again!

mk-maddin, 2022-02-20:
having the same issue here - espresense device do not longer report any devices.
Even after updating to latest version 2.0.29 :(

Interestingly I have two modules (out of 13) where update stuck at 2.0.24 - these are still working fine...

djk314, 2022-02-20:
same here i don't see my tikwatch but some other not all, i've one with 2.0.20 and see all....

mk-maddin, 2022-02-20:
just wanted to link here, too - for everyone seeing the need to downgrade.
I was able to perform OTA downgrade using method as described here:
https://github.com/ESPresense/ESPresense/discussions/250#discussioncomment-2214888

hvorragend, 2022-02-20:
Working for me with v2.0.31

k3vyn32, 2022-02-20:
2nd that

closed 2022-02-21 by DTTerastar