ESPresense/ESPresense #305: Can't connect to MQTT, MQTT Explorer sw can with same credentials - [E][main.cpp:252] onMqttDisconnect(): Disconnected from MQTT; reason 0

mbariola, 2022-02-17:
**Describe the bug**

Can't connect to MQTT, MQTT Explorer sw can with same credentials - [E][main.cpp:252] onMqttDisconnect(): Disconnected from MQTT; reason 0

**Config**

Version:    v2.0.28
IP address:   192.168.0.9
DNS address:  192.168.0.1
Hostname:     espresense-living-room-couch
Room:         Living room couch
MQTT server:  mqtt://homeassistant.local:1883 **(tried without mqtt:// also with port 1884)**
Max Distance: 2.25
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:
Include:
Exclude:

user: mosquittouser (as set on Home Assistant)
password: (as set on Home Assistant)
Preferences: all ON except Arduino OTA Update

**Hardware**

[KeeYees Esp32 Mini](https://www.amazon.de/gp/product/B08TBPBJGV/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)

**Logs**

Connecting to WiFi SSID '[MYWIFINAME]'...........................192.168.0.9
Version:    v2.0.28
IP address:   192.168.0.9
DNS address:  192.168.0.1
Hostname:     espresense-living-room-couch
Room:         Living room couch
MQTT server:  mqtt://homeassistant.local:1883 (tried without mqtt:// also with port 1884)
Max Distance: 2.25
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:
Include:
Exclude:
[E][main.cpp:252] onMqttDisconnect(): Disconnected from MQTT; reason 0

0 Reconnect timer
0 Reconnecting to MQTT...
[E][main.cpp:252] onMqttDisconnect(): Disconnected from MQTT; reason 0

0 Reconnect timer
0 Reconnecting to MQTT...
[E][main.cpp:252] onMqttDisconnect(): Disconnected from MQTT; reason 0

0 Reconnect timer
0 Reconnecting to MQTT...
[E][main.cpp:252] onMqttDisconnect(): Disconnected from MQTT; reason 0

**Log from Mosquitto MQTT Broker on Home Assistant (note the MQTT Explorer connecting successfully)**:

1645122303: mosquitto version 1.6.12 starting
1645122303: |-- *** auth-plug: startup
[19:25:03] INFO: Successfully send discovery information to Home Assistant.
[19:25:04] INFO: Successfully send service information to the Supervisor.
1645122303: Config loaded from /etc/mosquitto/mosquitto.conf.
1645122303: Loading plugin: /usr/share/mosquitto/auth-plug.so
1645122303:  ├── Username/password checking enabled.
1645122303:  ├── TLS-PSK checking enabled.
1645122303:  └── Extended authentication not enabled.
1645122303: Opening ipv4 listen socket on port 1883.
1645122303: Opening ipv6 listen socket on port 1883.
1645122303: Opening websockets listen socket on port 1884.
1645122303: Warning: Mosquitto should not be run as root/administrator.
1645122303: mosquitto version 1.6.12 running
1645122303: New connection from 127.0.0.1 on port 1883.
1645122303: Socket error on client <unknown>, disconnecting.
1645122303: New connection from 172.30.32.1 on port 1883.
{"result": "ok", "data": {}}1645122303: New client connected from 172.30.32.1 as 6vRwFZNRkMdEG0iFDQGaW4 (p2, c1, k60, u'mosquittouser').
1645122326: New connection from 172.30.32.2 on port 1883.
1645122326: Socket error on client <unknown>, disconnecting.
1645122446: New connection from 172.30.32.2 on port 1883.
1645122446: Socket error on client <unknown>, disconnecting.
1645122548: New connection from 172.30.32.1 on port 1883.
1645122548: New client connected from 172.30.32.1 as mqtt-explorer-fbdae1a8 (p2, c1, k60, u'mosquittouser').
1645122566: New connection from 172.30.32.2 on port 1883.
1645122566: Socket error on client <unknown>, disconnecting.
1645122686: New connection from 172.30.32.2 on port 1883.
1645122686: Socket error on client <unknown>, disconnecting.
1645122806: New connection from 172.30.32.2 on port 1883.
1645122806: Socket error on client <unknown>, disconnecting.
1645122926: New connection from 172.30.32.2 on port 1883.
1645122926: Socket error on client <unknown>, disconnecting.
1645123046: New connection from 172.30.32.2 on port 1883.
1645123046: Socket error on client <unknown>, disconnecting.
1645123166: New connection from 172.30.32.2 on port 1883.
1645123166: Socket error on client <unknown>, disconnecting.
1645123286: New connection from 172.30.32.2 on port 1883.

**Screenshots**

If applicable, add screenshots to help explain your problem.

The error message does not change whether I turn off or reset the HA, nor whether I change the user password or anything. It appears as if the connection issue is on the board side. I am using the esp32 firmware flashed from your website.

mbariola, 2022-02-17:
Update: I tried to use the HA's IP address (192.168.0.10) - still cannot connect, but Reason changed from 0 to 5

DTTerastar, 2022-02-21:
You are connecting, but confusing the mqtt server.  The only thing it could be is a bad client credentials on connect.  Can you run tcp_dump port=1833 on your mqtt server?

mbariola, 2022-02-24:
It's a supervised Home Assistant instance with MQTT add-on so I don't have "real" shell access to it unless I can find again how to create and enable a "true" ssh shell on the pi. Maybe there's a simpler way that I am overlooking?

DTTerastar, 2022-03-22:
Did you ever figure this out?

mbariola, 2022-03-22:
Hi @DTTerastar , I tried for a while with no advancement. Since MQTT and HA were not the main reason for ESPresense in my setup (see more about it [here](https://github.com/ESPresense/ESPresense/issues/389)) , but rather the core realtime reporting of distance for use in a project of mine which runs other stuff on the same esp32, I am figuring out my own solution. I really like your fingerprinting feature (and how were you able to figure out the Measured Power / RSSI at 1 m for all those devices?) so I am taking inspiration from it and working with the barebones NimBLE library

closed 2022-05-28 by stale[bot]