ESPresense/ESPresense #306: Security when initializing

xeor, 2022-02-17:
**Is your feature request related to a problem? Please describe.**
There seams to be no way to securely setup a device. Without broadcasting over unencrypted wifi both the MQTT credentials, and the SSID/password you are connecting to.

**Describe the solution you'd like**
There are multiple solutions to this

* WPA2 encrypted wifi with a default password (that might be known). You can't sniff traffic easily on encrypted networks, even tho you know the password. The password can be `ESPresense123` and be documented.

* A documented way to flash using predefined settings. I'm not sure what happens if I flash after changing https://github.com/ESPresense/ESPresense/blob/master/src/defaults.h, but it will probably still start the captive portal..? This might be possible already, but I didn't find any documentation on how to do it..

**Describe alternatives you've considered**
Setting up a dedicated wifi with client separations and each dedicated mqtt users with policies on the mqtt server. Still not sure if this will be sufficient, as I still would feel like anyone could update which room I am in. (yes, I know, but you get the point. Also, I know you could more easily spoof bluetooth device mac's.. But still :))

**Additional context**
Thanks for what seams to be finally a good solution to room automation! Love this project!

gmanc2019, 2022-02-17:
Love the project, I would also add my support to improved security.
At the most basic level would suggest the mqtt password not be visible in plain text after a system reboot.
A possible user credential page to allow people access to the wifi or mqtt settings.
A switch exposed within the MQTT which disables the the wifi and mqtt setup page, this way in the future its easy to switch back on again if required.

Zipties, 2022-02-18:
I think a lot of obvious changes like that will need to be made at some point. Right now I am not sure if that is the most valuable place to spend time on though. Personally, I think making it an ESPHome project would save a lot of dev hours by not needing to have to recreate a bunch of wheels, and administration would be much easier.

closed 2022-04-27 by stale[bot]