ESPresense/ESPresense #307: mi band 6

Jos-77, 2022-02-17:
i have installed espresense correct, setup wifi and mqtt.

My mi band 6 is not showing up as a white mac but as a red mac with del in front of the mac adress.

1 Del   | MAC: c1070102204a, ID: mifit:c1070102204a 

Please help.

full terminal log

Connected 115200 8N1...
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (C8:C9:A3:D2:7B:2C)...
Connecting to WiFi SSID 'IoT'...............192.168.2.xx
Version:    v2.0.28
IP address:   192.168.2.xx
DNS address:  192.168.2.xx
Hostname:     espresense-buiten
Room:         Buiten
MQTT server:  192.168.2.xx:1883
Max Distance: 4.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      
Exclude:      
0 New   | MAC: 52d93cd4bad1, ID: sd:0xfe9f                                                    
0 New   | MAC: 6bef2d9964fb, ID: sd:0xfe9f                                                    
0 New   | MAC: 4775cc85ec1e, ID: sd:0xfe9f                                                    
0 New   | MAC: 6618e75fd539, ID: sd:0xfe9f                                                    
0 New   | MAC: 76990e840e25, ID: sd:0xfe9f                                                    
0 New   | MAC: 64b9535adf06, ID: sd:0xfe9f                                                    
0 New   | MAC: 6b6d4e31c178, ID: sd:0xfe9f                                                    
Not updating from (sc=302): https://github.com/ESPresense/ESPresense/releases/download/v2.0.28/esp32.bin
0 New   | MAC: 5a7eec6dbf8b, ID: sd:0xfe9f                                                    
1 Del   | MAC: 134038125d16, ID: exp:20                                                       b0d80ad11746684a79996052e313cd7c7a2073ce
0 New   | MAC: 697e7f616356, ID: apple:1005:9-8                                               2c1ca9415f
1 Del   | MAC: 76990e840e25, ID: sd:0xfe9f                                                    
1 Del   | MAC: 5a1751aeddb3, ID: sd:0xfe9f                                                    
1 Del   | MAC: c1070102204a, ID: mifit:c1070102204a                                           
1 Del   | MAC: 4775cc85ec1e, ID: sd:0xfe9f                                                    
Not updating from (sc=302): https://github.com/ESPresense/ESPresense/releases/download/v2.0.28/esp32.bin
Not updating from (sc=302): https://github.com/ESPresense/ESPresense/releases/download/v2.0.28/esp32.bin

DTTerastar, 2022-02-20:
Please try .29 and see if things are any better.

Jos-77, 2022-02-20:
Install worked better and a little faster, config as normal whit query set for mifit:
distance 4 m. led off.

if query is not set, no mifit devices found.

Terminal Output, something changed, but not visable in Home Assistant.

Connected 115200 8N1...
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (C8:C9:A3:D2:7B:2C)...
Connecting to WiFi SSID 'IoT'...............192.168.2.42
Version:    v2.0.29
IP address:   192.168.2.42
DNS address:  192.168.2.1
Hostname:     espresense-achtertuin
Room:         Achtertuin
MQTT server:  192.168.2.45:1883
Max Distance: 4.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        mifit:
Include:      
Exclude:      
1 New   | MAC: 72f2f2456681, ID: sd:0xfe9f                                                    
1 Query | MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -49
1 QryErr| MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -49, try 1, retry after 0ms
1 Query | MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -50
1 QryErr| MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -50, try 2, retry after 1000ms
1 Query | MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -71
1 QryErr| MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -71, try 1, retry after 5000ms
1 Query | MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -68
1 QryErr| MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -68, try 2, retry after 6000ms
1 Query | MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -49
1 QryErr| MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -49, try 3, retry after 5000ms
1 New   | MAC: 652525cb7f72, ID: sd:0xfe9f                                                    
1 Query | MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -72
1 QryErr| MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -72, try 3, retry after 10000ms
1 Query | MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -52
1 QryErr| MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -52, try 4, retry after 14000ms
1 New   | MAC: 410101c1ab5b, ID: sd:0xfe9f                                                    
1 New   | MAC: 7985e36f870b, ID: sd:0xfe9f                                                    
1 Query | MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -72
1 QryErr| MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -72, try 4, retry after 19000ms
1 Query | MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -52
1 QryErr| MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -52, try 5, retry after 30000ms
1 New   | MAC: 755a2dec27c3, ID: sd:0xfe9f                                                    
1 Query | MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -71
1 QryErr| MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -71, try 5, retry after 35000ms
1 Query | MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -51
1 QryErr| MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -51, try 6, retry after 30000ms
1 Query | MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -69
1 QryErr| MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -69, try 6, retry after 30000ms
1 Query | MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -50
1 QryErr| MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -50, try 7, retry after 30000ms
1 Query | MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -68
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -68, try 7, retry after 30000ms
1 Query | MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -49
1 QryErr| MAC: ce2c22b2a802, ID: mifit:ce2c22b2a802                                           rssi -49, try 8, retry after 30000ms
1 Query | MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -77
1 QryErr| MAC: c1070102204a, ID: mifit:c1070102204a                                           rssi -77, try 8, retry after 30000ms

HA Config

# ESPresense
  - platform: mqtt_room
    name: "Mi Band Jos"
    device_id: "mifit:c1070102204a"
    state_topic: "espresense/rooms"
    timeout: 5
    away_timeout: 300
  - platform: mqtt_room
    name: "Mi Band Petra"
    device_id: "mifit:ce2c22b2a802"
    state_topic: "espresense/rooms"
    timeout: 5
    away_timeout: 300

entitie: sensor.mi_band_jos
state: not_home
distance: null
friendly_name: Mi Band Jos

poweredgenl, 2022-02-20:
following this issue as well - im runing 2.0.30 but its not detecting my band in any way (yet)...

Jos-77, 2022-02-21:
v2.0.31 makes no diverence.

Jos-77, 2022-02-21:
same log output

Jos-77, 2022-02-21:
after reinstalling mqtt and reconfiguration data is flowing into Home Assistant. 

query error remains

Jos-77, 2022-02-21:
No distance updates. Keeps first distance

poweredgenl, 2022-02-21:
@Jos-77 im having a mi band 6 as well - and i -think- it works as expected here ... but im trying to figure out if i have unware of it, the same issue you have ... wil check it out ...

Jos-77, 2022-02-21:
I have the mi band in combo with a samsuns s21 5g. Maby this is an issue? The band is only connected with the mi app not to the phone.
If bl on Phone is off the band can be connected with espresense.

poweredgenl, 2022-02-21:
Funny enough I have an s21 as well...its connected to the mi fit app and to an another notify app which syncs to Google fit...still no issues....so our setup is exactly the same

hellcry37, 2022-02-22:
poweredgenl advise you well, enable discoverable mode in your mi app. It will work. I've also go a miband that works ok

Krzeslo, 2022-02-22:
> poweredgenl advise you well, enable discoverable mode in your mi app. It will work. I've also go a miband that works ok

Doesn't work in my case. I have MiBand 3 and it rarely shows up in the logs when its paired with my phone :/ I have setup band visibility in the app and tried turning it on and off multiple times but it shows fine and instantly when i unpair it or turn bt in my phone off. When i add Query filtering it keeps showing QryErr (it doesn't show those QryErr when i dont add Query filtering)

`[22:30:43]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:30:43]configsip: 0, SPIWP:0xee
[22:30:43]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:30:43]mode:DIO, clock div:2
[22:30:43]load:0x3fff0018,len:4
[22:30:43]load:0x3fff001c,len:1044
[22:30:43]load:0x40078000,len:8896
[22:30:43]load:0x40080400,len:5828
[22:30:43]entry 0x400806ac
[22:30:43]Connecting to WiFi (XXX)...
[22:30:46]Connecting to WiFi SSID 'XXX'...............192.168.1.10
[22:30:46]Version:    v2.0.31
[22:30:46]IP address:   192.168.1.10
[22:30:46]DNS address:  1.1.1.1
[22:30:46]Hostname:     espresense-maciek
[22:30:46]Room:         Maciek
[22:30:46]MQTT server:  XXX:1883
[22:30:46]Max Distance: 4.00
[22:30:46]Telemetry:    enabled
[22:30:46]Rooms:        enabled
[22:30:46]Devices:      enabled
[22:30:46]Discovery:    enabled
[22:30:46]PIR Sensor:   disabled
[22:30:46]Radar Sensor: disabled
[22:30:46]Query:        mifit:d7d2ea70e35b
[22:30:46]Include:      mifit:d7d2ea70e35b
[22:30:46]Exclude:      
[22:30:46]1 New   | MAC: 4372d450afb5, ID: msft:cdp:0902                                                57294f3a88bff322df4a98b57bb3c785892ff943
[22:30:46]1 New   | MAC: 14bb6e0973cc, ID: samsung:14bb6e0973cc                                         
[22:30:46]1 New   | MAC: 6104b67cf9a4, ID: md:00e0:8                                                    
[22:30:47]1 New   | MAC: 5448e6b7880a, ID: 5448e6b7880a                                                 
[22:30:47]1 New   | MAC: 5448e6bae733, ID: 5448e6bae733                                                 
[22:30:49]1 Close | MAC: 4372d450afb5, ID: msft:cdp:0902                                                (0.13m) -39dBm
[22:30:51]1 New   | MAC: 7028455f15b9, ID: 7028455f15b9                                                 
[22:30:53]1 New   | MAC: 3868a40adfd9, ID: samsung:3868a40adfd9                                         
[22:31:02]1 Close | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.00m) -39dBm
[22:31:02]1 New   | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                3c0e0abe72cbed8642ea265e45db5d22671deb4a
[22:31:20]1 New   | MAC: 70284561572b, ID: 70284561572b                                                 
[22:31:31]1 Left  | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.14m) -81dBm
[22:31:32]1 Close | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.14m) -39dBm
[22:32:53]1 Left  | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.15m) -103dBm
[22:32:53]1 Close | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.15m) -39dBm
[22:35:04]1 New   | MAC: d76f8a7efc10, ID: mifit:d76f8a7efc10                                           
[22:35:31]1 Left  | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.15m) -81dBm
[22:35:44]1 Close | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.15m) -39dBm
[22:35:47]Not updating from (sc=302): https://github.com/ESPresense/ESPresense/releases/download/v2.0.31/esp32.bin
[22:36:03]1 Del   | MAC: 4372d450afb5, ID: msft:cdp:0902                                                57294f3a88bff322df4a98b57bb3c785892ff943
[22:36:21]1 New   | MAC: 74d6673ffe45, ID: md:00e0:8                                                    
[22:36:28]1 New   | MAC: 6ac592d162ac, ID: md:00e0:8                                                    
[22:37:46]1 Left  | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.12m) -104dBm
[22:37:46]1 Close | MAC: 4d1f67bba1b4, ID: msft:cdp:0902                                                (0.12m) -39dBm
[22:39:53]1 New Q | MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           
[22:39:59]1 Query | MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           rssi -85
[22:40:03]1 QryErr| MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           rssi -85, try 1, retry after 5000ms
[22:40:07]1 Query | MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           rssi -79
[22:40:09]lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
[22:40:09]E NimBLEClient: "Connection failed; status=574 "
[22:40:09]1 QryErr| MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           rssi -79, try 2, retry after 6000ms
[22:40:18]1 Query | MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           rssi -76
[22:40:19]lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
[22:40:19]E NimBLEClient: "Connection failed; status=574 "
[22:40:19]1 QryErr| MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           rssi -76, try 3, retry after 10000ms
[... cut ...]
[22:41:02]1 New   | MAC: 4a5f810c41cd, ID: msft:cdp:0902                                                f8a14b1071ad422af495498f9d1f44c8396858c7
[22:41:16]1 Query | MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           rssi -75
[22:41:19]1 QryErr| MAC: d7d2ea70e35b, ID: mifit:d7d2ea70e35b                                           rssi -75, try 5, retry after 35000ms
[22:41:24]1 Del   | MAC: 6104b67cf9a4, ID: md:00e0:8                                                    
[22:41:29]1 Del   | MAC: 74d6673ffe45, ID: md:00e0:8                                                    
`

Jos-77, 2022-03-11:
Problem splved in latest fimware.

closed 2022-03-11 by Jos-77