ESPresense/ESPresense #313: Request MQTT password hidden - (password:******)

gmanc2019, 2022-02-18:
As a User of ESPresence, I [want to] have improved security associated with the MQTT password within the ESPresence web UI, [so that] the password is not left in plain text.

This prevents other having access to passwords/accounts being directly exposed to other users of the network.
Feels more secure. I know the password for wifi is hashed/blanked/starred out after submission.

[image]

ChirpyTurnip, 2022-03-08:
I concur.....+ sharing config is best done with a screenshot....which is not cool if there's a password there...  :-(

skotl, 2022-03-21:
Agreed, although I'd go further and suggest that there is no reason for the config page to expose the password at all. I'd suggest it just has static "******" content that, when overwritten, updates the password.

This wouldn't be as much of an issue if access to the espresence page was encrypted / authenticated, but having it just out there is a pretty easy hack for anyone getting onto the wifi.

cryptedx, 2022-04-23:
Please please hide the password from the WebUI! Thank you so much! :-)

DTTerastar, 2022-07-04:
Please test v2.2.1b5

DTTerastar, 2022-07-09:
This has been released.

closed 2022-07-09 by DTTerastar