ESPresense/ESPresense #314: Turn off Wifi hotspot - MQTT Switch

gmanc2019, 2022-02-18:
As a User of ESPresense, I [want to] have the ability to set a MQTT switch to turn off the ESP32 wifi hotspot (not the wifi connectivity) and the web UI (see screen shot below, highlighted red, similar to Active scan switch), [so that] it is only exposed during the initial configuration of the device, while also having the ability to set the MQTT switch to turn the interface back on in the future, this way locking down the capability while also keeping the option of switching back on in the future for additional configuration if required.

MQTT switch
[image]

Reference to MQTT switches: https://www.home-assistant.io/integrations/switch.mqtt/

ESPresence Web UI:
[image]

DTTerastar, 2022-02-20:
This sounds like a good idea to me

DTTerastar, 2022-04-24:
I still think we should do this.

julianrinaldi, 2023-09-06:
I would love to have this feature. The web server eats up a ton of memory, and it almost never works anyway for me, besides the first time it's set up.

DTTerastar, 2023-09-06:
It's not a bad idea.  PRs accepted!