ESPresense/ESPresense #324: Directions or Manual to assist with "Calibration" of multiple devices

moblsu, 2022-02-20:
**Is your feature request related to a problem? Please describe.**
I have 10 ESP32 devices and an equal number of BLE devices in my home. I have set several of them on a table next to each other and they all come back with different results for raw and distance. For example:
{"id":"iBeacon:xxx","idType":99,"rssi@1m":-69,"rssi":-93,"mac":"xxx","raw":4.85,"distance":3.93,"speed":0}
{"id":"itag:xxx","idType":17,"rssi@1m":-60,"rssi":-89,"mac":"xxx","raw":6.74,"distance":6.53,"speed":0}
measured distance: 3m

**Describe the solution you'd like**
Help understanding if, and how, I can calibrate each device (ESP or BLE device) so that I can calibrate/tune them to show more accurate distances. I've seen the statement on the Troubleshooting website: "Once you’ve found the device you’re interested in, you can use [Beacon Scope] to connect to and view information about your beacon, which can help in determining the correct setup." but, I do not know exactly what to do with this, other than prove my ignorance on the subject.

**Describe alternatives you've considered**
I would be willing to write the "step by step" documentation if I understand what is needed, but thus far, I've only been able to stumble through changing the "rssi@1m" through the HA companion app, but have no methods for other devices.

Is there a way to enter a "multiplier" into a "known devices" table so that I can adjust their distances? I've considered doing this in HA, but have not implemented it yet (distance x multiplier = actual distance):
   In above example: iBeacon:xxx "multiplier" = 0.763359 and itag:xxx "multiplier" = 0.459418

I don't know if that these multipliers would remain constant or change. I've also considered building a log table and calculating the average "distance" over a set period (10 min) to calculate the averages, rather than just taking the "first available reading"

**Additional context**
In this test, I was using an 2 x ESP32 (ASIN: B0718T232Z). 1 unit running v2.0.20 and 1 unit running v2.0.30. Neither have any filters. The unit running v2.0.30 was 1m from the two beacons and has only registered the iBeacon (not the itag) once (one ping) during the writing of this email {"id":"iBeacon:xxx","idType":99,"rssi@1m":-69,"rssi":-80,"mac":"xxx","raw":2.06,"distance":2.07,"speed":0} which would make the "multiplier" 0.483092... so that solution probably won't work universally, but might still work if the master table is contained in HA at the sensor level.

dashdrum, 2022-03-08:
I too would like to understand calibration a little better, and I am available to test any directions shared.

DTTerastar, 2022-03-10:
The first step is to place things 1m apart and get the calibrated RSSI@1m to match the RSSI you're seeing at 1m.  iBeacons and eddystone beacons actually broadcast an rssi@1m that is settable so you can tweak that to get the values to match.  If it's a device that doesn't broadcast rssi@1m you can change the global setting on the web portal of ESPresense. (yes a individual setting would be nice to have)

Once you have one meter down, you need to figure out the average absorption factor for your home.  It'll vary based on construction.  It'll also be lower if the path is free air, so you have to just figure out an average value.  I think putting the devices 5m away through a wall and then adjust the global absorption value (it's available in the newer firmwares) until the value shown is averaging 5m.

Remember that wireless signals follow the inverse square law so fundamentally the distance estimates will be worse and worse the farther you go.

One snafu is Eddystone actually broadcast RSSI at 0 m so ESPresense adds -41 (EDDYSTONE_ADD_1M) to the value to match iBeacons.  So if you're calibrating using Eddystone add 41 to it to get things to match.

I myself have not 100% calibrated everything to my liking so take the above with a grain of salt.  We're all in this together!

tylerwmarrs, 2022-03-14:
@DTTerastar I second the idea of having individual calibrations per device. It would also be nice to pull configuration from HA so that I don't have to set a filter of BLE devices on every node. I may try to pick up some C++ again and start contributing. The closest thing I've touched in the past 5 years is Cython. :)

closed 2022-05-20 by stale[bot]