ESPresense/ESPresense #326: Option to invert PIR signal

ahWorkShop, 2022-02-20:
I have Bosch PIR (model isc-bpr2-w12). In the Bosch environment when the circuit is opened, that signals movement was detected. Which seems to be the opposite of other PIR sensors. With the Bosch connected to ESPresense, when there is no movement the entity shows as movement detected and when there's movement its displayed as clear.

An option to configure how the PIR signal works in ESPresense or to change the MQTT payload would be helpful for these and other types of PIRs.

DTTerastar, 2022-04-24:
I was thinking we could allow for negative number input and when it's negative invert it.

rcmurphy, 2022-07-10:
Would love to see this implemented!

> I was thinking we could allow for negative number input and when it's negative invert it.

@DTTerastar as long as pin 0 isn't supported!

PR #663 (2022-09-09): Make PIR type configurable
Fixes #326
closed 2022-09-09 by DTTerastar