ESPresense/ESPresense #328: Raw and queried Devise shown in different rooms

ctschach, 2022-02-21:
**Describe the bug**

I still struggle with getting reliable data for my iPhone: My iPhone  with the raw „apple:1007-11-7“ is shown in one room, while the name after the query „IPhone12.5“ is shown in the other room.

the only „special“ thing is that „Wohnzimmer“ is where the phone currently is and „Schlafzimmer“ the room one floor above. If I look at the distance in the data from the sensor above, it’s 5.5m, while in the current room it shows as 2.8m (I’m sitting approx 1m away). 

Limiting the distance in the rooms does not really help, because you would probably need 5 sensor for a 40sqm room.

any idea why it’s showing up in different rooms? Any way to fix this in homeasistant?

**Config**

- Version: 2.0.31
- Active scan enabled: Yes
- Include filter: apple:1007:11-7
- Exclude filter: none 
- Query filter: none

**Hardware**

- Device (be specific):
ESP32 D1 Mini

**Screenshots**

 https://www.dropbox.com/s/iafojj0ce8nfb9r/IMG_1123.jpg?dl=0

DTTerastar, 2022-02-21:
This is because query requires the device to be nearby.  A feature I'm working on is to have the ESPs query amongst themselves *first* so that the further away nodes won't even try to connect to the device.

ctschach, 2022-02-22:
Okay...interesting...but makes sense. 

I know, it's a bit off-topic. But does HomeAssistant make use of the distance that is broadcasted? Or is it depending just on "which device received the last signal"?

ctschach, 2022-02-23:
Okay, now it start to get even more interesting. The raw device shows up as "not_home" while the "queried device" shows the exact room.

[image]

But unfortunately I've also seen the other way around which makes it a little unreliable and hard to work with.

Is there anything I can improve?

askthomas, 2022-02-26:
I have the same problem, apple watch jumps around in many romms, even if am longer away then set parameter,

I tried to shorten distans on my 5 esp set to 2 an all devices, sitt 1m from one, but still detect my watch in many rooms.

ctschach, 2022-02-26:
Interesting…I have similar problems. Even though I have the feeling that the Apple Watch is more reliable than the iPhone.

But it’s definitely interesting: because I sit at. my desk, the ESPresense is less than meter away and all in a Suggestion I‘m moving from this room into the room below me or I was detected at „not_home“…withou changing the position.

Not sure if this is a problem of ESPresence or of Home-Assistant…

askthomas, 2022-02-28:
it would be nice to turn up distans, and multipples get hits that watch is close, and has could pick the one with shortest distans. I dont know if that possible

closed 2022-05-07 by stale[bot]