ESPresense/ESPresense #334: New Tracking Beacon suggestion

gabroid, 2022-02-22:
**Is your feature request related to a problem? Please describe.**
Non directly. I would suggest a new device i've tested
[SODIAL GPS Mini Tag Smart Tracker Bluetooth]([url] -(TrackerPA)(https://www.amazon.it/gp/product/B07DXQKD9N/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1))
[image]

**Describe the solution you'd like**
add this device to the documentation

**Describe alternatives you've considered**
do not add this device to the documentation

**Additional context**
I tested this, it is not very faster but can be helpful for tracking object in the house.
Attention battery is not included.
I do not sell this device, i just want to contribute and help

_example from ha:_
[image]

_config example:_

```
  - platform: mqtt_room
    device_id: "$(macaddress)"
    name: "ESP-trackerPA-White"
    state_topic: "espresense/devices/$(macaddress)"
    timeout: 60
    away_timeout: 120
```

ozett, 2022-03-01:
i this tile permantenly sending announcements? 
so i can be discovered without pressing any button?

thanks in advance

gabroid, 2022-03-02:
> i this tile permantenly sending announcements? so i can be discovered without pressing any button?
> 
> thanks in advance

this is a good question, generally yes, i use this for guests, i'm still testing this device, i wuold say that it works fine, may be it's not let me say, "verbose" sending information. Seems that sometimes it go in somekind of sleep mode after longer time, it has a microswitch inside, if you push the center of the cover it wake up. This behavior (i dicovered after my first post) seems random and i'm not sure how it works. I'll will update when i've new information ! 

For the moment i have 1 ESP32-Room for each room, but at the moment i still not put them in the center of the room and i'm investigating about the better configuration of espresense and home assistant

ozett, 2022-03-03:
pressing a button for anouncement wont help me here... 😢 
i ordered some tiles from aliexpress . i will report how it goes with them after they arrive in some weeks...

closed 2022-05-09 by stale[bot]