ESPresense/ESPresense #337: Integration with domoticz

Epik77, 2022-02-23:
Hello,
I have installed your firmware on an ESP32 and I find it really functional.
I ask you if there is an integration with domoticz
or the possibility of varying the mqtt topic and the variables in the json so that we can build the  payload ourselves

thank you for your support

DTTerastar, 2022-03-10:
I do not know of anything for domoticz.  You should make one!

Epik77, 2022-03-11:
Hi,
if it can be useful I have created this small service
which finds macaddresses from espresence and writes them to a variable
domoticz.
The script works well I leave it to you in case you want to implement the possibility of inserting it directly into the code.
thank you :-) 

`#!/opt/node/bin/node
const toad = require('mqttletoad');
 
(async function () {
          const client = await toad.connect('mqtt://127.0.0.1:1883', {
                      username:'XXXXXX',
                      password:'XXXXXX',
                      encoder: 'json',
                      decoder: 'json'
                    });

 await client.subscribe('espresense/rooms/taverna', message => {
   var name=JSON.parse(message).name
   var distance=String(JSON.parse(message).distance)
   var mac=JSON.parse(message).mac
   var disc=JSON.parse(message).disc
   switch (mac) {
    case 'XXXXXXX': //  Tag Lorenzo
   var trad=`{"command": "setuservariable", "idx": 45, "value": "Taverna Tag ${distance} metri"}`;
     break;
    case 'XXXXXXX': //  MiBand Lorenzo
   var trad=`{"command": "setuservariable", "idx": 45, "value": "Taverna Miband ${distance} Metri"}`;
     break;
    case 'XXXXXXX': //  Tag Mary
   var trad=`{"command": "setuservariable", "idx": 46, "value": "Taverna Tag ${distance} metri"}`;
     break;
   };
        if (trad) {
        client.publish('domoticz/in',trad);  
 },);
        
  
}());
`

closed 2022-05-17 by stale[bot]