ESPresense/ESPresense #351: Samsung Smart Tag

wifispray, 2022-02-28:
Running the latest beta version.

Having some issues with Samsung smart tags. Im seeing the tags share the same ID but displaying different macs at different RSSIs. Is this expects?

wifispray, 2022-02-28:
1 Del   | MAC: 76eed12f0408, ID: sd:0xfd5a                                                    
1 Del   | MAC: 785496fa0038, ID: apple:1005:9-24                                              2718349730
1 Del   | MAC: 612422d0ed5e, ID: sd:0xfd5a                                                    
1 Del   | MAC: 4875a69e2e6e, ID: sd:0xfd5a                                                    
1 Del   | MAC: 4e887cec5444, ID: sd:0xfd5a                                                    
1 New   | MAC: c8e4e3ea156d, ID: c8e4e3ea156d

DTTerastar, 2022-03-02:
That fingerprint basically just means we can see it sending some data via a uuid (0xfd5a) but don't yet support it.

Can you install the verbose firmware and send the log of that?  Can you install and scan it w/ nRF Connect (send screenshots)?  We have to figure out which part of the data is consistent to the particular tag.  By default BLE uses random macs, which makes them useless for telling stuff apart.

wifispray, 2022-03-04:
Connected 115200 8N1...
Verbose | exp:20                                                    AD: 0xfd6f
Verbose | exp:20                                Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000
Verbose | apple:1005:9-26                                           MD: 4c0010052f981a1bd3
Verbose | exp:20                                                    AD: 0xfd6f
Verbose | exp:20                                                    SD: 0xfd6f/185eef9fbbe573c93bd6ddb12d5e4dbda3a09ba3
Verbose | 88c626bd3c57                                              AD: 72daa6c3-29c2-6283-0c4a-2818e4d37e75
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | apple:1006:10-1                                           MD: 4c0010060f1ded5b3168
Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | apple:1006:10-12                                          MD: 4c001006131ea6cd03c8
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | apple:1006:10-12                                          MD: 4c001006131ea6cd03c8
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000
Verbose | samsung:bc7e8ba93db5                                      MD: 7500420401016fbc7e8ba93db5be7e8ba93db424b40000000000
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | apple:1006:10-1                                           MD: 4c0010060f1ded5b3168
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | exp:20                                                    AD: 0xfd6f
Verbose | exp:20                                                    SD: 0xfd6f/185eef9fbbe573c93bd6ddb12d5e4dbda3a09ba3
Verbose | apple:1005:9-26                                           MD: 4c0010052f981a1bd3
Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | apple:1006:10-12                                          MD: 4c001006131ea6cd03c8
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | exp:20                                                    AD: 0xfd6f
Verbose | exp:20                                                    SD: 0xfd6f/185eef9fbbe573c93bd6ddb12d5e4dbda3a09ba3
Verbose | apple:1005:9-26                                           MD: 4c0010052f981a1bd3
Verbose | apple:1006:10-12                                          MD: 4c001006131ea6cd03c8
Verbose | apple:1007:11-7                                           MD: 4c0010073d1f18c1b95158
Verbose | apple:1006:10-1                                           MD: 4c0010060f1ded5b3168
Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000
Verbose | apple:1006:10-12                                          MD: 4c001006131ea6cd03c8
Verbose | 88c626bd3c57                                              AD: 72daa6c3-29c2-6283-0c4a-2818e4d37e75
Verbose | exp:20                                                    AD: 0xfd6f
Verbose | exp:20                                                    SD: 0xfd6f/6fa36ae7e1f815740819bc9592956629460c373c
Verbose | apple:1007:11-7                                           MD: 4c0010073d1f18c1b95158
Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000
Verbose | msft:cdp:0902                                             MD: 060001092002bdb1b23922681907e6d9104a1f1d923281d6a6e9896470
Verbose | apple:1006:10-1                                           MD: 4c0010060f1ded5b3168
Verbose | 88c626bd3c57                                              AD: 72daa6c3-29c2-6283-0c4a-2818e4d37e75
Verbose | samsung:bc7e8ba93db5                                      MD: 7500420401016fbc7e8ba93db5be7e8ba93db424b40000000000
Verbose | exp:20                                                    AD: 0xfd6f
Verbose | exp:20                                                    SD: 0xfd6f/185eef9fbbe573c93bd6ddb12d5e4dbda3a09ba3
Verbose | sd:0xfd5a                                                 AD: 0xfd5a
Verbose | sd:0xfd5a                                                 SD: 0xfd5a/13f3e4004619b535b9a3b595b30000007cda3362
Verbose | apple:1007:11-7                                           MD: 4c0010073d1f18c1b95158
Verbose | 88c626bd3c57                                              AD: 72daa6c3-29c2-6283-0c4a-2818e4d37e75
[... cut ...]
Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000
Verbose | apple:1006:10-1                                           MD: 4c0010060f1ded5b3168
Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000
Verbose | apple:1007:11-7                                           MD: 4c0010073d1f18c1b95158
Verbose | samsung:d49dc03f66ad                                      MD: 75004204018060d49dc03f66add69dc03f66ac01000000000000

wifispray, 2022-03-04:
[image]
[image]

DTTerastar, 2022-03-04:
Well the data in FD5A seems fairly constant.  I'll try adding that all to the id and you can try it

wifispray, 2022-03-04:
Perfect will do. I have 3 others in the same network. Will they need to be
programmed?

wifispray, 2022-03-07:
Is this part of release 36?

DTTerastar, 2022-03-09:
Sorry, not yet, swamped.

DTTerastar, 2022-03-12:
2.0.38b4 will have it

wifispray, 2022-03-12:
Amazing. Will this work ok with the m5stack atom?

wifispray, 2022-03-23:
This is now working, but showing up as multiple smart tags, I only have 3 but have loads of entires....

DTTerastar, 2022-03-24:
Is any part of the id constant?

wifispray, 2022-03-28:
Example of one....

espresense/devices/smarttag:13e4ed001bb8e2bd620a47e1b30000007af75e56
espresense/devices/smarttag:13e3ed001bb8e2bd620a47e1b30000001c216711
espresense/devices/smarttag:13e4ed001bb8e2bd620a47e1b30000007af75e56

DTTerastar, 2022-03-28:
are those 3 separate tags, or the same tag?

wifispray, 2022-03-30:
This is the same device
espresense/devices/smarttag:13beee000dd117aa72469412b3000000dfbadba2/ohg_office
espresense/devices/smarttag:13bdee000dd117aa72469412b3000000b26860a4/ohg_office

DTTerastar, 2022-03-31:
So if we snip off the last 8 characters do you thing it'll work?

wifispray, 2022-03-31:
Not sure, been monitoring for a while now with two devices... They show as below.

smarttag:1300ef000dd117aa72469412b300000065dbe618
smarttag:1300ef00e7ce5be3f5b15fe6b300000020a5d48d

smarttag:13ffee00e7ce5be3f5b15fe6b30000000edc7ca4
smarttag:13ffee000dd117aa72469412b30000003befc952

smarttag:1301ef00e7ce5be3f5b15fe6b30000001430110e
smarttag:1301ef000dd117aa72469412b3000000af5168ba

wifispray, 2022-03-31:
Same two devices....

espresense/devices/smarttag:1302ef00e7ce5be3f5b15fe6b3000000e2d7adca/ohg_office
espresense/devices/smarttag:1302ef000dd117aa72469412b30000005e96b3aa/ohg_office
espresense/devices/smarttag:1303ef00e7ce5be3f5b15fe6b3000000d1243432/ohg_office
espresense/devices/smarttag:1303ef000dd117aa72469412b3000000e0dc80f0/ohg_office
espresense/devices/smarttag:1304ef00e7ce5be3f5b15fe6b30000006da38328/ohg_office
espresense/devices/smarttag:1304ef000dd117aa72469412b30000006a1eb962/ohg_office
espresense/devices/smarttag:1305ef00e7ce5be3f5b15fe6b30000000ea59bf7/ohg_office
espresense/devices/smarttag:1305ef000dd117aa72469412b300000039e60749/ohg_office
espresense/devices/smarttag:1306ef00e7ce5be3f5b15fe6b3000000db5c080d/ohg_office
espresense/devices/smarttag:1306ef000dd117aa72469412b3000000fa8424aa/ohg_office
espresense/devices/smarttag:1307ef00e7ce5be3f5b15fe6b3000000328abdf0/ohg_office
espresense/devices/smarttag:1307ef000dd117aa72469412b300000027a1a93f/ohg_office

DTTerastar, 2022-04-02:
Have you had a chance to check it out?

wifispray, 2022-04-02:
Hey sorry.... 
This is what im seeing now...

[image]

wifispray, 2022-04-02:
[image]

wifispray, 2022-04-02:
think auto update was on...... working i think
over 250 msg no change
[image]

ARowbot, 2022-04-17:
I have 1 smart tag and 2 smart tag+ and I am noticing that they are randomly generating the IDs and MACs unless I have implemented something incorrectly. I am using  v2.1.1b1 on 5 Atoms and every 15 minutes or so the IDs change to something completely different. Here are the results of approx 20 minutes:

[image]

DTTerastar, 2022-04-24:
I don't have any of those tags.  What I did was just based on the above.  It's probably just impossible to use them like apple airtags.

DTTerastar, 2022-05-14:
OK, well I think I should just add this to the list of stuff that doesn't work :(

Zixim, 2022-05-14:
unfortunately...this is the way.

DTTerastar, 2022-05-23:
Added.

rumbu13, 2022-06-19:
Is it possible to publish on mqtt more information about the smarttag than the mac address? Eventually as an option or in verbose version.

I am currently using 4 smartttags (obviously all identified as smarttag:20) and I succesfully managed in 90% of cases to keep track of their MAC changes. Currently I am keeping track of their last position and when one of the four MACs is changing I calculate the best candidate. Empirically it is always the one which was not updated any room topic in the last 5 seconds. 

But there are some corner cases which I cannot manage if I don't have more info:
- when at least 2 of them are changing their MAC during 5 seconds. In this case I use the geometric mean of their position deltas, but this will not always work.
- when at least 2 of them leave the house and one come back later. In this case I pick randomly one which was not seen in the last 30 seconds.

I feel that if I would have more info in the mqtt message I can distinguish better between them.

DTTerastar, 2022-06-19:
That's what the discriminator field is useful for.  What portion of the broadcast do you want put into discriminator? Have you tried scanning the tags with RF connect when the Mac address changes?

rumbu13, 2022-06-20:
> What portion of the broadcast do you want put into discriminator?

Currently I have only one mqtt sensor in ha mapped to espresence/devices/smarttag:20
If you put something else in discriminator, it will not report in ha.

Behind the scenes, I am using NetDaemon (I feel more confortable in C# than in python) to monitor that mqtt sensor an do all the logic I described. I keep a dictionary of mac addresses and I update the distances only for the reported mac address.  When I encounter more than 4 addreses, it is clear that one of the smarttags had changed its mac (or a neighbour bought a smarttag, but let's not complicate things). Consequently, I map the fifth mac to the best candidate among the existing four. Usually, the "best" candidate is the one which not reported in the last 5 seconds to any room (I have 6 mcu's), otherwise I  choose to replace the one with the closest position to the new one. 

>Have you tried scanning the tags with RF connect when the Mac address changes?

I tried, but every test I made was inconclusive, that's why I would prefer to have that data in the mqtt message to process and log it for further analysis.

closed 2022-09-07 by stale[bot]