ESPresense/ESPresense #354: Automatically query for Room Assistant iOS beacon id using manufacturer data

ronaldheft, 2022-02-28:
**Is your feature request related to a problem? Please describe.**
Tracking iPhones has been a struggle, and even with querying for the Room Assistant beacon id by placing Apple identifiers in the query box, that doesn't seem to always work (Room Assistant will see my phone but ESPresense will not). I'm not sure if my Apple identifier is moving constantly, or it's due to the large number of iPhones in my area with colliding device ids (I live in an apartment).

**Describe the solution you'd like**
Thankfully, there appears to be a solution. When the Room Assistant Companion is running on an iPhone, there is extra manufacturer data in the BLE advertisement that can be used to identifier if the device has a Room Assistant identifier. You can see an example in the Room Assistant source code here: https://github.com/mKeRix/room-assistant/blob/main/src/integrations/bluetooth-low-energy/bluetooth-low-energy.service.ts#L482

Room Assistant is looking for Apple manufacturer ids that have the extra bits before deciding to query for the extra service characteristics. It looks like the [4th bit of the 3rd byte of the manufacturer data should be a 1](https://github.com/mKeRix/room-assistant/blob/1a7f579cddee3bddb74cc8c0ac73e05ba0969879/src/integrations/bluetooth-low-energy/bluetooth-low-energy.service.ts#L596) on devices with the Room Assistant companion. [There is also a function that is used to extract the extra manufacturer data](https://github.com/mKeRix/room-assistant/blob/1a7f579cddee3bddb74cc8c0ac73e05ba0969879/src/integrations/bluetooth-low-energy/bluetooth-low-energy.service.ts#L578). (More info on how this works: http://www.davidgyoungtech.com/2020/05/07/hacking-the-overflow-area)

My thinking is instead of placing a device id in the query box to have ESPresense scan for the Room Assistant identifier, it could automatically do it if it sees this special bit in the BLE advertisement. This would cut down on the number of devices queried when there are lots of iPhones with colliding identifiers and it can be determined with the passive BLE advertisement.

**Describe alternatives you've considered**
Not everyone may want ESPresense automatically querying iPhones if this identifier is present, so maybe it's put behind a configuration option. Another option could be placing the identifier "roomassistant" in the query box to allow ESPresense to query devices with this extra manufacturer data.

DTTerastar, 2022-03-01:
I know it's not documented well but we actually already do this.  If you install the app you should see the ID change to something that starts with roomAssistant:

ronaldheft, 2022-03-01:
@DTTerastar Without placing anything in the "Query device ids for characteristics" field? I have not observed that behavior. I've had to track down the device id and add it to that list for the `roomAssistant` identifier to appear.

ronaldheft, 2022-03-01:
Hmm, actually scratch that. It seemed the Room Assistant app had stopped broadcasting the identifier. I restarted it and I think it may have worked without adding anything in query field. I'll report back in an hour or so to confirm it's held stable.

DTTerastar, 2022-03-01:
> Hmm, actually scratch that. It seemed the Room Assistant app had stopped broadcasting the identifier. I restarted it and I think it may have worked without adding anything in query field. I'll report back in an hour or so to confirm it's held stable.

Yeah I believe that iOS when it backgrounds the app it stops broadcasting the meta-data as often. I do honored if I see it. In my own config I've added it to query so that it works 100% of the time.

ronaldheft, 2022-03-01:
So it seemed to work for a bit without anything in the query box, but as soon as my phone's mac address changed, it stopped. Room Assistant is still chugging along just fine. I tried restarting my ESPs (I have three) and none of them have picked up the `roomAssistant` identifier yet. One of my ESP is right next to one of my Pis and I am currently 5 feet from it.

Here's the last three hours:
[image]

If I look at my MQTT server, I can see my phone with a generic identifier, but nothing is coming back with roomAssistant. I also tried playing with turning active scanning on and off.

Any ideas on things to try?

DTTerastar, 2022-03-01:
Can you tell via a BLE scanner if room assistant is still advertising?

ronaldheft, 2022-03-01:
Yes, here is the Room Assistant service characteristic on my phone (using Bluetility).
[image]

Here is the manufacturer id:
[image]

DTTerastar, 2022-03-01:
Very weird, I have ble device that can hook to wireshark, i'll try that and see if it stops or if it's my code that's broke.

ronaldheft, 2022-03-01:
Thanks for taking a looking. I'm reading over the source code, and it looks like roomAssistant [may only be recognized if the iPhone is advertising its services](https://github.com/ESPresense/ESPresense/blob/fdd38947e656dd4081d6e7ca04cf5868e79b334d/lib/BleFingerprint/BleFingerprint.cpp#L95). I don't think the iPhone is always advertising services on a passive advertisement and only does so when queried, hence why Room Assistant is using the manufacturer string workaround.

DTTerastar, 2022-03-01:
Oh have you tried Active?  That might be the difference.

DTTerastar, 2022-03-01:
We could make active more dynamic and only turn on for a bit after seeing a new device advertise.

ronaldheft, 2022-03-02:
> Oh have you tried Active?  That might be the difference.

Turned on active. No difference. It is occasionally finding my phone again, but Room Assistant has remained stable on finding my phone the whole time.

Maybe it's getting hung up on a previous MAC and is struggling to switch?

[image]

DTTerastar, 2022-03-02:
Room assistant you need to provide the hardware mac, right?  Doesn't it just query that every time?  That mac doesn't change as well.  That's cool, but it goes against the passive listening... But query does go against it as well, so maybe that's a better way.  At least that way doesn't have to requery every 20 minutes.

ronaldheft, 2022-03-02:
Just loaded up the pre-release. No luck in finding the roomAssistant id without placing the device id in the query list.

I added my iPhone to the query list and I saw this message a few times, and then it eventually found the roomAssistant id.
```
[10:42:08]1 Query | MAC: 5e65e73f20cc, ID: apple:1007:11-7                                              rssi -61
[10:42:09]lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
[10:42:09]E NimBLEClient: "Connection failed; status=574 "
```

I will report back a little later the stability of putting the device id in the query box.

ronaldheft, 2022-03-02:
Looks like I'm also getting some crashes with the prerelease:
```
[10:47:16]abort() was called at PC 0x401aa6eb on core 1
[10:47:16]
[10:47:16]ELF file SHA256: 0000000000000000
[10:47:16]
[10:47:16]Backtrace: 0x4008f670:0x3ffe6590 0x4008f8ed:0x3ffe65b0 0x401aa6eb:0x3ffe65d0 0x401aa732:0x3ffe65f0 0x401a9e0b:0x3ffe6610 0x401aa142:0x3ffe6630 0x401af33d:0x3ffe6650 0x401af4b2:0x3ffe6670 0x401af5dd:0x3ffe66b0 0x400f7999:0x3ffe66d0 0x400fa13b:0x3ffe66f0 0x400f5c7e:0x3ffe6730 0x400f5e63:0x3ffe67c0 0x400d2476:0x3ffe67e0 0x400e0d11:0x3ffe6800 0x400e41ca:0x3ffe6840 0x400e90a5:0x3ffe68b0 0x400e9151:0x3ffe68f0 0x400e939d:0x3ffe6910 0x400e7c55:0x3ffe6930 0x400eedc6:0x3ffe6950 0x400e024b:0x3ffe6970 0x400908fe:0x3ffe6990
[10:47:16]
[10:47:16]Rebooting...
[10:47:16]ets Jun  8 2016 00:22:57
[10:47:16]
[10:47:16]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:47:16]configsip: 0, SPIWP:0xee
```

ronaldheft, 2022-03-02:
Besides the crashing every 5-15 minutes, the new beta has done well tracking me throughout the day.

I currently have:
* Active scanning disabled
* My device id entered into the query box (roomAssistant was not showing up otherwise)

[image]

ronaldheft, 2022-03-03:
So last 12 hours have been less reliable. Same configuration as above, but every time the MAC rotates the ESPresense instances lose the Room Assistant beacon for a while.

[image]

Looking at the logs, the instances are now running for about 60-90 minutes before crashing, so I suspect the constant crashing before was actually contributing to finding the Room Assistant beacon more often.

DTTerastar, 2022-03-06:
I think I may have found in the ble specification the thing we're not doing, you can make a "scan request" to get more uuids if they don't fit in the main advertisement.  I bet the iphone sorts them such that they make it on the main advertisment if the app is up but fall back to the scan request/response if not.

hvorragend, 2022-03-06:
Now I have a problem, 

I have several Gigaset G-Tags and now the device name doesn't not contains the mac address. 
The topic looks like this: `espresense/devices/name:gigaset-g-tag/floor`

And inside this topic I have:

```
{
  "id": "name:gigaset-g-tag",
  "name": "Gigaset G-tag",
  "idType": 35,
  "rssi@1m": -71,
  "rssi": -83,
  "raw": 2.2,
  "distance": 2.23,
  "speed": 0,
  "mac": "7cxx80xx91xx"
}
```

Sadly now I cannot use the mqtt_room integration in Home Assistant, because there are many devices inside the same topic.

DTTerastar, 2022-03-06:
Ok, i have a feature in mind for that use case

mww012, 2022-03-07:
I'm seeing some of the same behavior with the Room Assistant app on my iPhone.  Happy to be another tester if that would be helpful.

DTTerastar, 2022-03-07:
Try turning on active scan.  That should automatically ask the iPhone for more info that may include roomAssistant info.

mww012, 2022-03-07:
I do have active scan enabled.  Since I'm not running on battery I just enable it by default.  Also, I'm running v2.0.36 and my filter is empty in case that matters.

One interesting thing I noticed is that when ESPresense boots I see the Room Assistant beacon and then the query.  The phone responds with the proper beacon and everything works as expected.  After a while I stop seeing room assistant beacon (with the app in the background).  In order to see the beacon again I have to pull the app back to the foreground AND restart my ESP32.  As soon as it restarts I see the beacons again.

gadric, 2022-03-13:
@mww012: I have the same issue, with and without active scan enabled.

DTTerastar, 2022-03-13:
I don't think this is an issue I can do something about in the code.  If I open the room assistant app the code instantly finds the advertisement and works great.  But after some amount of idle it stops advertising.  This is actually a GOOD thing because on something like my iPad i leave it in a room and the id is more of an "actively using" tracker.  But for something like my phone which I always want to track, I just add the apple:100x:x id to query and it works 100% of the time now (in the pre-lease).

gadric, 2022-03-13:
@DTTerastar I would like to test the pre-lease to provide feedback. I guess I can install the pre-release marking the install pre-lease right? 

How can I down-grade if it does not work?

DTTerastar, 2022-03-13:
Correct.  Uncheck the box :)

gadric, 2022-03-13:
@DTTerastar: Great work! Installed the pre-release 2 hours ago. Since then, the tracking of my iPhone works really good. I can report tomorrow if this worked over the night.

gadric, 2022-03-13:
Update: I am seeing 13 minutes intervals were my iPhone is recognized as not_home.

gadric, 2022-03-14:
It seems that iOS still has hick ups. This is the the situation from the last 12 hours. As you can see, it reports often as `not_home`. My phone was not moving in the night. 

iPhone 12: 
[image]

In comparison to MiFit over ESP32 
[image]

gadric, 2022-03-20:
I think I figured out the issue. 

first of all, I have brought more ESP32 and place them in all rooms. And since 20 hours, it works without any issue as long as I deactivate room assistant on my iPhone. It seems that it causes problems if both are running at the same time.

mww012, 2022-03-20:
Awesome!  I'll give that a try as well. I hadn't considered Room Assistant causing a conflict. I'll let you guys know what the results are. 

Also, I enabled pre-release version and all my esp32s are running 2.1.0 but I believe that's the general release version. Is that what I should be seeing?

gadric, 2022-03-27:
Hmm, after days of testing, it does not to work for me ... 

I tried many combinations: with active scanning, with query, with only room-assistant, without only apple id. 

it seems that it goes offline after a couple of minutes. I dont know, may it related to that I have an iPad and AppleTV around here? Too many collisions?

bachoo786, 2022-04-05:
hi guys I am struggling to find my ID for my iphone 12. Can anyone help please?

DTTerastar, 2022-04-06:
Are you using the serial monitor and putting them on top of each other?

bachoo786, 2022-04-06:
> Are you using the serial monitor and putting them on top of each other?

I have 4 base stations and under devices in my mqtt explorer I have lots of apple devices showing and other various devices. I can't really tell which one is which.

DTTerastar, 2022-04-09:
The serial output shows CLOSE when the device is touching the esp32.  It's very easy that way to get the id.

closed 2022-06-15 by stale[bot]