ESPresense/ESPresense #360: Paired bluetooth devices not broadcasting MQTT while in range of phone

moblsu, 2022-03-02:
**Describe the bug**

I have 3 devices paired to my Samsung Note Ultra Plus, a Fitbit, Galaxy Watch, and LG headset. I can only find them through MQTT Explorer if my phone is "away"/"off"/"not available". IE: If I leave the devices at home or if I turn off my bluetooth on my phone, they  broadcast and can be seen on MQTT. While I'm at home, they do not.

Has anyone else experienced this or found a work around?

**Config**

- Version: v2.0.20 through v2.0.34
- Active scan enabled: yes and no
- Include filter: none
- Exclude filter: none
- Query filter: none

**Hardware**

- Firmware flavor: 
- Device (be specific):
Synology DS920+ running as MQTT host, ESP32 WROOM 32D

fredamn76, 2022-03-03:
I have the same issue with my Fitbit.

DTTerastar, 2022-03-03:
I need some proof the device is actually broadcasting during that time.  Otherwise it's not a bug, it's just a device that can't be used.

moblsu, 2022-03-03:
I believe that it is not a bug.

Since, and after some thought, I believe that the bluetooth devices are broadcasting until the point at which they are connected (paired) to a device. If they get disconnected, they then will start broadcasting so that they can be found by another device.

I think that these devices are not able to be utilized as both their intended uses and as a beacon. You're better off buying a Blue Charm beacon and hooking it to your house shoes (yes, that's what I did).

Closing the stream

closed 2022-03-03 by moblsu

DTTerastar, 2022-03-03:
I have one on my dog.  He's always at my feet :)