ESPresense/ESPresense #365: Include Filter isn't working

begunfx, 2022-03-04:
**Describe the bug**

Adding devices to the filter list seems to just prevent all traffic from being received by the sensors.  The Query filter doesn't seem to pick up the devices listed in the filter.

**Config**

- Version: 2.0.34
- Active scan enabled:
- Include filter: iBeacon.1fb27f22-e037-4d7a-8644-ed01b4b99ab8-19-73 iBeacon.5e8fbd77-f32d-460b-8ee1-9be5497a1e26-19-81 apple.iphone13-2
- Exclude filter:
- Query filter:

**Hardware**
Atom LITE

- Firmware flavor: Release version
- Device (be specific): Atom LITE

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.

DTTerastar, 2022-03-04:
Replace the dots with colons (:) :)

begunfx, 2022-03-04:
Awesome!  Thanks!  It was driving me crazy that it didn't seem to work.  Can you maybe tweak the usage so it's a little clearer that you need to follow that format?  The ids listed in the beacons use periods, so logically it seemed that we just copy and paste them.

DTTerastar, 2022-03-04:
The ids are always colon separated.  Not sure where you've seen dots.

begunfx, 2022-03-04:
Weird.  I had copied the IDs from MQTT Explorer a while back and they were using dots.  I guess I hadn't noticed that the format changed?  I was just re-pasting the values that were there before (me being lazy).

begunfx, 2022-03-04:
Either way.  I guess I can close this then.  The filter queries were a big deal for me to move forward using this, since I was getting way too much traffic to my MQTT broker.

closed 2022-03-04 by begunfx

simonyel, 2022-03-08:
And what if one of the devices listed in the include filter is a MAC address? Without the : in the ID?

DTTerastar, 2022-03-11:
The logic is, take all the values find the spaces and make a list of prefixes.  If any of the prefixes match the id then include it.  So an id without a colon should work fine.

simonyel, 2022-03-11:
I can't copy/paste my filter currently, but I have a filter with two (hass app) iBeacon entries and Mac from a watch. The watch is never located. Without filter, the watch is reported along with 30 other devices. Might be worth double checking if column also plays part in parsing the filter.

DTTerastar, 2022-03-11:
What kind of watch?

simonyel, 2022-03-11:
Garmin Vivo active, I believe.

- Simon Yelfimov