ESPresense/ESPresense #367: [Support:] UUID sending?

ozett, 2022-03-05:
**Is your feature request related to a problem? Please describe.**
i want to clearly identify all seen BT devices in my environment.
seems that the UUID is a fixed advertisment, but not send over MQTT

i want to identify most of the seen BT devices clearly.
fingerprinting is good, [but still changes above devices](https://github.com/ESPresense/ESPresense/issues/260)
UUID could be solution for BT5 (BT4?) https://github.com/ESPresense/ESPresense/issues/82

**Describe the solution you'd like**
send the UUID with MQTT

**Describe alternatives you've considered**
looked through issues but did not find solution https://github.com/ESPresense/ESPresense/issues?q=UUID
maybe i dont have all BT details for creating UUIDs and they change also?
ID-Type is made by espresence and no BT standard https://github.com/ESPresense/ESPresense/issues/193
maybe already working, but not over MQTT? https://github.com/ESPresense/ESPresense/issues/41

**Additional context**
https://www.argenox.com/library/bluetooth-low-energy/ble-advertising-primer/#a-quick-look-into-uuids

[image]

ozett, 2022-03-13:
some suggestion about the UUID format
https://github.com/shmuelzon/esp32-ble2mqtt/issues/12#issuecomment-456804828

ozett, 2022-03-19:
Please consider sending the alway preset apple nearby information in RAW format, too.

https://github.com/furiousMAC/continuity/blob/master/messages/nearby_info.md

DTTerastar, 2022-03-20:
It's in the disc field already

ozett, 2022-03-21:
> It's in the disc field already

what is the disc-field?
is not only pure apples nearby information. why not?
is always something else per device/manufacturer?  is it a mixed-type field?

https://github.com/ESPresense/ESPresense/blob/3a28f8c869f525437a3b3d9596c9b0fbecfd9d52/lib/BleFingerprint/BleFingerprint.cpp#L292

https://github.com/ESPresense/ESPresense/blob/3a28f8c869f525437a3b3d9596c9b0fbecfd9d52/lib/BleFingerprint/BleFingerprint.cpp#L322

still looks to me promising for seondary fingerprinting to send the pure UUID via MQTT

ozett, 2022-03-21:
while observing the disc-values, it seems to me, that its not always starts with the apple-nearby identifyer.
could this be the case?

[image]

DTTerastar, 2022-03-22:
The first 2 bytes are not in it, because those are the last 2 bytes of id.

DTTerastar, 2022-03-22:
> > It's in the disc field already
> 
> what is the disc-field? is not only pure apples nearby information. why not? is always something else per device/manufacturer? is it a mixed-type field?

It's discriminator.  It's basically the portion of the advertisement that we excluded from the id.  My thought was it would be useful at a higher level app to detect mac rollovers.  If the mac changed but the disc didn't we'd know it was still the same device.  It is currently unused.

closed 2022-05-28 by stale[bot]