ESPresense/ESPresense #369: DHT22 Humidity percentage units "%" not making it through to home assistant

NickC4, 2022-03-05:
Hi
Running a DHT22 on pin 16 of a a D1 mini32. Espresense v2.034 /w sensors.
All looks good but, I think the humidity comes over MQTT with just a number without units.
Temperature plots look good, but humidity has no "unit of measurement" included.
Thanks
Nick.

tr-v-r, 2022-03-06:
Hey Nick,

I was just looking at this too. I'm not sure that this is the right way, but I found that by adding the unit_of_measurement to either my sensor settings or homeassistant customize section resolved this for me.

```yaml
homeassistant:
  customize:
    sensor.espresense_humidity:
        unit_of_measurement: "%"
```

Here are a few links talking about this; https://community.home-assistant.io/t/adding-units-of-measurement-in-2021-12/384972/6 and https://www.home-assistant.io/integrations/sensor.mqtt/#temperature-and-humidity-sensors

Would be great if there is a way to do this without the work around, but I hope it helps in the meantime.

Cheers,

T.

NickC4, 2022-03-06:
Yes your workaround sorts it. In my case I put the last 2 lines in customize.yaml and it gives the sensor the right units so you can plot alongside other humidity meters etc.
Still hope this can be sorted in code too.
Thanks T.
Nick.

closed 2022-05-12 by stale[bot]