ESPresense/ESPresense #370: Add a option for a temperature offset with DHT11 & DHT22

tr-v-r, 2022-03-06:
I have seen several posts in other forums talking about the challenge of including a temperature sensor in the same enclosure as an ESP32 (or any other microcontroller} due to excess heat creating inaccurate reporting. That got me wondering if creating a user configurable static offset that could be applied to the temperature reading before sending the temperature to MQTT would be consistent enough to address this.

Does this seam like a reasonable approach, and would it be worth including?

I wasn't comfortable with creating a pull request without raising this for discussion, so here is a possible solution:

https://github.com/ESPresense/ESPresense/blob/b0798e5baf2ead8f4de60f90bb1456f7aa42e030/src/main.h#L110-L112
```c
// Add
float dhtTempOffset;
```

https://github.com/ESPresense/ESPresense/blob/b0798e5baf2ead8f4de60f90bb1456f7aa42e030/src/main.cpp#L152-L153
```c
// ADD
dhtTempOffset = WiFiSettings.floating("dhtTemp_offset", -40, 125, 0.0, "DHT temperature offset");
```
https://github.com/ESPresense/ESPresense/blob/b0798e5baf2ead8f4de60f90bb1456f7aa42e030/src/main.cpp#L215-L218
```c
// ADD
Serial.print("DHT Temp Offset: ");
Serial.println(dhtTempOffset ? "enabled" : "disabled");
```
https://github.com/ESPresense/ESPresense/blob/b0798e5baf2ead8f4de60f90bb1456f7aa42e030/src/main.cpp#L702
```c
// Replace with
float temperature = dhtSensorData.temperature + dhtTempOffset;
```

I also noticed that the temp and humidity are being sent with 2 decimal places which is more precision that the sensors offer and is inconstant with other devices I have seen. The following change would bring this down to a single decimal place.

https://github.com/ESPresense/ESPresense/blob/b0798e5baf2ead8f4de60f90bb1456f7aa42e030/src/main.cpp#L703-L706
```c
// Replace with
Serial.println("Temp: " + String(temperature, 1) + "'C Humidity: " + String(humidity, 1) + "%");

mqttClient.publish((roomsTopic + "/humidity").c_str(), 0, 1, String(humidity, 1).c_str());
mqttClient.publish((roomsTopic + "/temperature").c_str(), 0, 1, String(temperature, 1).c_str());
```

Cheers,

T.

DTTerastar, 2022-03-11:
I'm fine w/ a PR for this!

tr-v-r, 2022-03-11:
A pull request has been summited https://github.com/ESPresense/ESPresense/pull/385

closed 2022-05-17 by stale[bot]