ESPresense/ESPresense #374: Is 2.0.36 broken?

skycryer, 2022-03-07:
I See on my devices with the latest version running on esp32 that after the configuration the devices go into endless reboot loop. On device on older versions like 2.0.12 this does not happen. The log shows no errors, terminal shows just connecting to Matt and wifi then rebooting.

medains, 2022-03-07:
Occurring for me too - I'm wondering if there's something in BLE range that's triggering a bad code path.  All of my espresence devices are doing it, and they can all see a google home mini (which always looks like md:00e0:8)

entegral, 2022-03-07:
im getting endless reboots as well, the following log repeats:

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (XX:XX:XX:XX:AB:F0)...
Connecting to WiFi SSID '....'.................192.168.XXX.XXX
Version:    v2.0.36
IP address:   192.168.XXX.XXX
DNS address:  192.168.XXX.XXX
Hostname:     espresense-kitchen
Room:         Kitchen
MQTT server:  192.168.XXX.XXX:1883
Max Distance: 16.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      
Exclude:      
1 New   | MAC: d8714d82a519, ID: d8714d82a519                                                 
1 New   | MAC: 70b950d966be, ID: 70b950d966be                                                 
abort() was called at PC 0x401aab0f on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f670:0x3ffe5d10 0x4008f8ed:0x3ffe5d30 0x401aab0f:0x3ffe5d50 0x401aab56:0x3ffe5d70 0x401aa22f:0x3ffe5d90 0x401aa566:0x3ffe5db0 0x401af761:0x3ffe5dd0 0x401af8d6:0x3ffe5df0 0x401afa01:0x3ffe5e30 0x400f7dbd:0x3ffe5e50 0x400fa55f:0x3ffe5e70 0x400f6092:0x3ffe5eb0 0x400f6277:0x3ffe5f40 0x400d2492:0x3ffe5f60 0x400e1125:0x3ffe5f80 0x400e45de:0x3ffe5fc0 0x400e94b9:0x3ffe6030 0x400e9565:0x3ffe6070 0x400e97b1:0x3ffe6090 0x400e8069:0x3ffe60b0 0x400ef1da:0x3ffe60d0 0x400e065f:0x3ffe60f0 0x400908fe:0x3ffe6110

Rebooting...
```

ChirpyTurnip, 2022-03-08:
Same here. It boots, then boot loops. In Home Assistant the uptime counter is frozen on 15 seconds....after a while it just completely craps out and then it stays offline until I power cycle it.  On previous versions I don't think I'd seen the counter go higher than 7000 so it was probably rebooting every 2 hours or so anyway....

Also I should mention that the ESP32, which is *only* doing ESPresence BLE work at present, is also running pretty darn hot.....in contrast I have another ESP32 which is running ESPHome with a custom BLE detection module and it is cool to the touch. So the ESPresence code is making it work really, really hard....but doing what?

medains, 2022-03-08:
I got more of a message on the crash:

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble)

medains, 2022-03-08:
Turning off active-scan seems to have improved matters.

theuzzy, 2022-03-08:
Same problem here, active scanning is off by default. 

Regards

maromaro70, 2022-03-09:
Definitely an issue. Interestingly, some esp32 boards I have at home are not affected, the one affected the most is M5Stack atom lite, this particular one started having trouble with 2.0.35, and it continues today with 2.0.36.

theuzzy, 2022-03-09:
I went back to version 2.0.30 and it works again.

andreasgily, 2022-03-09:
I can confirm this - 7/8 ESP32 NodeMCU Modules hang after some hours since the auto update to 2.0.36.

NickC4, 2022-03-09:
Also failing are my 3x D1 mini 32 ESP32s.
3 out of 3 are offline. Immediate bootloops since ~ 2.0.34 ish. 
I can plug in and load half the web page if I'm quick enough with the F5 key.
Terminal flashing 2.0.36 without active scanning enabled does not help.
Nick.

iCSpotRun, 2022-03-09:
Can confirm that M5Stack Atom Lite is consistently having this problem, even on 2.0.37

Terminal:

abort() was called at PC 0x401aab1f on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f670:0x3ffe5d10 0x4008f8ed:0x3ffe5d30 0x401aab1f:0x3ffe5d50 0x401aab66:0x3ffe5d70 0x401aa23f:0x3ffe5d90 0x401aa576:0x3ffe5db0 0x401af771:0x3ffe5dd0 0x401af8e6:0x3ffe5df0 0x401afa11:0x3ffe5e30 0x400f7dcd:0x3ffe5e50 0x400fa56f:0x3ffe5e70 0x400f60a2:0x3ffe5eb0 0x400f6287:0x3ffe5f40 0x400d2492:0x3ffe5f60 0x400e1121:0x3ffe5f80 0x400e45da:0x3ffe5fc0 0x400e94b5:0x3ffe6030 0x400e9561:0x3ffe6070 0x400e97ad:0x3ffe6090 0x400e8065:0x3ffe60b0 0x400ef1d6:0x3ffe60d0 0x400e065b:0x3ffe60f0 0x400908fe:0x3ffe6110

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

DTTerastar, 2022-03-10:
Can you try running the verbose version and give me the log line before the crash I do think it is a device that is causing us trouble.

medains, 2022-03-10:
Here's a few:

```
Verbose | md:00e0:8                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
Verbose | md:00e0:8                                                 MD: e00001b2ca62e240
CORRUPT HEAP: multi_heap.c:744 detected at 0x3ffe4350
abort() was called at PC 0x4009430a on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f670:0x3ffdf4b0 0x4008f8ed:0x3ffdf4d0 0x4009430a:0x3ffdf4f0 0x400945a1:0x3ffdf510 0x4014af1e:0x3ffdf530 0x4014af7d:0x3ffdf570 0x40102ae2:0x3ffdf5b0 0x400d8ca3:0x3ffdf5d0 0x400d6805:0x3ffdf600 0x400d784f:0x3ffdf670 0x400908fe:0x3ffdf6f0
```

```
Verbose | md:00e0:8                                                 MD: e0000175ca8321bb
CORRUPT HEAP: multi_heap.c:744 detected at 0x3ffe4350
abort() was called at PC 0x4009430a on core 1

ELF file SHA256: 0000000000000000
```

```
Verbose | itag:ffffa009570c                                         MD: 0501ffffa009570c6602010300
1 New   | MAC: ffffa009570c, ID: itag:ffffa009570c                                            
Verbose | msft:cdp:0902                                             MD: 0600010920028ae42e93426fb8ffab9a74b5e61b5bd08d387cec285df0
CORRUPT HEAP: multi_heap.c:744 detected at 0x3ffe4350
abort() was called at PC 0x4009430a on core 1
```

```
Verbose | name:v-voactive3                                          MD: 87000a8c
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble) 
Core 1 register dump:
PC      : 0x400f9d5e  PS      : 0x00060e36  A0      : 0x800fa0ac  A1      : 0x3ffe65f0  
A2      : 0x3ffe71f0  A3      : 0x00000001  A4      : 0x00000001  A5      : 0xaae16f5c  
A6      : 0x3ffca124  A7      : 0x3ffba524  A8      : 0x3ffba9e0  A9      : 0x3ffe6860  
A10     : 0x3ffe6540  A11     : 0x00000001  A12     : 0x3ffe65a8  A13     : 0x00000001  
A14     : 0x3ffca124  A15     : 0x3ffe6544  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000
```

In my environment turning active scan off means that it can run for hours rather than minutes.

DTTerastar, 2022-03-10:
OK thank you those are some good hints as to what part of the code!  Can you try running the pre-release I did already make some changes to that part.

PhillipRt, 2022-03-10:
I have the same issue on the newest prerelease, also if I move with my laptop into a different room, it stops without me even unplugging the esp32.
I beleve it is defenetly a device causing a buffer overflow or similar.

```
[13:42:29]Verbose | dbf7536c9772                                              MD: 5900dbf7536c9772
[13:42:29]1 New   | MAC: dbf7536c9772, ID: md:0059:8                                                    
[13:42:29]Verbose | 5448e6dc0ea6                                              MD: 8f030a10560500a40edce6485481
[13:42:29]1 New   | MAC: 5448e6dc0ea6, ID: 5448e6dc0ea6                                                 
[13:42:29]Verbose | 44c4c45e5393                                              AD: 0xfe9f
[13:42:29]Verbose | ad:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
[13:42:29]1 New   | MAC: 44c4c45e5393, ID: sd:0xfe9f                                                    
[13:42:29]Verbose | 789cab2e5399                                              AD: 0xfe9f
[13:42:29]Verbose | ad:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
[13:42:29]1 New   | MAC: 789cab2e5399, ID: sd:0xfe9f                                                    
[13:42:29]Verbose | md:0059:8                                                 MD: 5900dbf7536c9772
[13:42:29]abort() was called at PC 0x401a98a3 on core 1
[13:42:29]
[13:42:29]ELF file SHA256: 0000000000000000
[13:42:29]
[13:42:29]Backtrace: 0x4008f670:0x3ffe7dc0 0x4008f8ed:0x3ffe7de0 0x401a98a3:0x3ffe7e00 0x401a98ea:0x3ffe7e20 0x401a8fc3:0x3ffe7e40 0x401a92fa:0x3ffe7e60 0x401ae4dd:0x3ffe7e80 0x401ae652:0x3ffe7ea0 0x401ae77d:0x3ffe7ee0 0x400f67bd:0x3ffe7f00 0x400f9057:0x3ffe7f20 0x400f4871:0x3ffe7f60 0x400f4a83:0x3ffe8000 0x400d2406:0x3ffe8020 0x400e182d:0x3ffe8040 0x400e3f1e:0x3ffe8080 0x400e8501:0x3ffe80f0 0x400e85ad:0x3ffe8130 0x400e87f9:0x3ffe8150 0x400e756d:0x3ffe8170 0x400edfee:0x3ffe8190 0x400e0def:0x3ffe81b0 0x400908fe:0x3ffe81d0
[13:42:29]
[13:42:29]Rebooting...
[13:42:29]ets Jun  8 2016 00:22:57
```

```
[13:42:20]Verbose | ed7fbdc97230                                              MD: 5900ed7fbdc97230
[13:42:20]1 New   | MAC: ed7fbdc97230, ID: md:0059:8                                                    
[13:42:20]Verbose | f8c731c4469d                                              MD: 5900f8c731c4469d
[13:42:20]1 New   | MAC: f8c731c4469d, ID: md:0059:8                                                    
[13:42:21]Verbose | dbf7536c9772                                              MD: 5900dbf7536c9772
[13:42:21]1 New   | MAC: dbf7536c9772, ID: md:0059:8                                                    
[13:42:21]Verbose | 3e90325bf5f1                                              MD: 060001092002c2f37ced814d1e64645a0ec9374179e47f5be9ef0cb136
[13:42:21]1 New   | MAC: 3e90325bf5f1, ID: msft:cdp:0902                                                c2f37ced814d1e64645a0ec9374179e47f5be9ef
[13:42:21]Verbose | 789cab2e5399                                              AD: 0xfe9f
[13:42:21]Verbose | ad:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
[13:42:21]1 New   | MAC: 789cab2e5399, ID: sd:0xfe9f                                                    
[13:42:21]Verbose | 69ab40426d26                                              AD: 0xfe9f
[13:42:21]Verbose | ad:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
[13:42:21]1 New   | MAC: 69ab40426d26, ID: sd:0xfe9f                                                    
[13:42:21]Verbose | md:0059:8                                                 MD: 5900dbf7536c9772
[13:42:21]Verbose | 5448e6dc0ea6                                              MD: 8f030a10560500a40edce6485481
[13:42:21]1 New   | MAC: 5448e6dc0ea6, ID: 5448e6dc0ea6                                                 
[13:42:21]abort() was called at PC 0x401a98a3 on core 1
[13:42:21]
[13:42:21]ELF file SHA256: 0000000000000000
[13:42:21]
[13:42:21]Backtrace: 0x4008f670:0x3ffe7710 0x4008f8ed:0x3ffe7730 0x401a98a3:0x3ffe7750 0x401a98ea:0x3ffe7770 0x401a8fc3:0x3ffe7790 0x401a92fa:0x3ffe77b0 0x401ae4dd:0x3ffe77d0 0x401ae652:0x3ffe77f0 0x401ae77d:0x3ffe7830 0x400f67bd:0x3ffe7850 0x400f9057:0x3ffe7870 0x400f4871:0x3ffe78b0 0x400f4a83:0x3ffe7950 0x400d2406:0x3ffe7970 0x400e182d:0x3ffe7990 0x400e3f1e:0x3ffe79d0 0x400e8501:0x3ffe7a40 0x400e85ad:0x3ffe7a80 0x400e87f9:0x3ffe7aa0 0x400e756d:0x3ffe7ac0 0x400edfee:0x3ffe7ae0 0x400e0def:0x3ffe7b00 0x400908fe:0x3ffe7b20
[13:42:21]
[13:42:21]Rebooting...
[13:42:21]ets Jun  8 2016 00:22:57
[13:42:21]
[13:42:21]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

```

DTTerastar, 2022-03-10:
It's a Harman Kardon or JBL ble device :)

theuzzy, 2022-03-11:
Ok, I've updated to pre release version: ESPresense Version: v2.0.38b1 on a ESP32devkitv1.
Its not crashing right away. I'll keep an eye on it and report back later.

regards,

medains, 2022-03-11:
2.0.38b1 with active scanning on is still crashing for me:

```
Verbose | b20cd108efe1                                              SD: 0xfdf7/01acc285e0a23b5d8c0c5ec2a59401f7470000000003
Verbose | b20cd108efe1                                              MD: 650001d205
Verbose | msft:cdp:0902                                             MD: 06000109200245de5590b20bc2d7b31ab4425fa2e32d52a5704fa265b6
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble) 
Core 1 register dump:
PC      : 0x400f83f2  PS      : 0x00060636  A0      : 0x800f8740  A1      : 0x3ffe6b80  
A2      : 0x3ffe7780  A3      : 0x00000001  A4      : 0x00000001  A5      : 0xaed3281a  
A6      : 0x3ffc9aec  A7      : 0x3ffba524  A8      : 0x3ffba9e0  A9      : 0x3ffe6df0  
A10     : 0x3ffe8f64  A11     : 0x00000001  A12     : 0x3ffe8fcc  A13     : 0x00000001  
A14     : 0x3ffc9aec  A15     : 0x3ffe8f68  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
```

DTTerastar, 2022-03-11:
This is driving me nuts!  C++ is a pain in the ass!  I don't know how I can fix this without it happening for me.

maromaro70, 2022-03-11:
> This is driving me nuts! C++ is a pain in the ass! I don't know how I can fix this without it happening for me.

For me on M5Stack Atom lite, all broke with the 2.035, went back to 2.034 and all is well. Not sure how easy it is to troubleshoot by looking at changes starting with 2.035. interestingly, 2.036 and 2.037 work just fine on other esp32 flavors I have. Thank you you for the great work, by the way, tremendous effort! This is grabbing a lot of interest.

iCSpotRun, 2022-03-11:
I echo @maromaro70 observations with the Atom Lite as well as the recognition given. Thank you devs.

DTTerastar, 2022-03-11:
I have a m5stickc that also has terrible uptime.  My m5atoms are fine tho.  The m5stickc has a blown serial port so i can't see what's up inside.  Maybe it's time to take it apart and hook up an ftdi cable.  I have it stuck to the bottom of my monitor tho, so i really don't want to mess with it.... Oh well...

DTTerastar, 2022-03-11:
[image]

DTTerastar, 2022-03-11:
When is the last time you serial flashed it instead of OTA?  Maybe that is the key to it.  This one with a broken serial port has ONLY been ota flashed for a long time now.

iCSpotRun, 2022-03-11:
I have never done OTA. Only serial.

DTTerastar, 2022-03-11:
Well taking apart a m5stick-c didn't help, no rx/tx on top, couldn't figure out how to get to the bottom without breaking it.

maromaro70, 2022-03-11:
> When is the last time you serial flashed it instead of OTA? Maybe that is the key to it. This one with a broken serial port has ONLY been ota flashed for a long time now.

All my updates after the first installation have been OTA

scottgerhardt, 2022-03-14:
I feel like I might have tried a bad time to get into this.  7 units, 6 of them seem to get borked shortly after boot.

Also, unrelated, you should have some kinda guide on how to fine tune settings so you can get the right things connecting in the right room.  I see settings, but to be honest I'm not familiar enough in this world to know how which ones to adjust and how,  I love this project and think it would be amazing, but it's kinda driving me batty at the moment....lol.

mlava, 2022-03-15:
Just a note to follow up on my linked closed issue #395.
I now have four ESPresense devices running 2.0.34 and things are working well.
Cheers!

NickC4, 2022-03-15:
> Also failing are my 3x D1 mini 32 ESP32s. 3 out of 3 are offline. Immediate bootloops since ~ 2.0.34 ish. I can plug in and load half the web page if I'm quick enough with the F5 key. Terminal flashing 2.0.36 without active scanning enabled does not help. Nick.

After a couple of hours of uptime now, no bootloops with v2.1.0 flashed in.
3x D1 mini32 (sensor version (PIR, DHT22) and regular version). 
Nick.

ChirpyTurnip, 2022-03-16:
So far an uptime of 26,593s....I don't think I've ever had such a 'high
score' before, on any version....it always seemed to cap out around 7-8000s
before a soft boot. :-)

medains, 2022-03-19:
v2.1.0 - still had bootloops with active scanning, but I'm up to over 2 days now (261877s) without it; so something has gone right.

ChirpyTurnip, 2022-03-21:
Finally died! Last reported uptime: 491,090s

IP: 192.168.20.200
Uptime: 491,090
Firm: esp32-verbose-sensors
Rssi: -46
Ver: v2.1.0
Adverts: 12,900,085
Seen: 12,790,853
Reported: 1,273,100
FreeHeap: 118,424
MaxAllocHeap: 80,728
MemFrag: 31.83
ResetReason: PowerOn

And since then it has become unstable. Home Assistant reports:

Was disconnected
08:10:52 - 15 seconds ago
Was connected
08:10:33 - 34 seconds ago
Was disconnected
08:08:35 - 3 minutes ago
Was connected
08:07:25 - 4 minutes ago
Was disconnected
08:07:24 - 4 minutes ago
Was connected
08:05:08 - 6 minutes ago

<Power cycled>
Was disconnected
00:24:32 - 8 hours ago
Was connected
00:21:31 - 8 hours ago
Was disconnected
00:21:21 - 8 hours ago

The attributes no longer update either...

Connecting directly I can see it has reverted to 2.0.37.

:-(

Back to boot looping....

CP

DTTerastar, 2022-03-21:
What?  It's updated to an old firmware????

ChirpyTurnip, 2022-03-21:
Yup....looks like that's exactly what it did.....

Configuration
Language: English
ESPresense Version: v2.0.37
Room: Lounge

Sorry.....I know it is a major pain when everything goes sideways.....

DTTerastar, 2022-03-21:
Is pre-release on?

ChirpyTurnip, 2022-03-21:
That's odd.....it was running just fine....a reboot didn't get it to come back.....I will re-flash it....and might in fact flash it twice to get around that problem....?

closed 2022-05-27 by stale[bot]