ESPresense/ESPresense #377: Adding phtoresistor / LDR support

freaked1234, 2022-03-09:
I´d imagine that a lot of us use motion or presence detection to control our lights but setting sun angle or time is not a optimal condition to determine if lights are needed in the first place.
Since we already have a device in each room that sports a few ADCs we could install photoresistors on our espresense devices as a check if lights are needed.

Adding another device just to check light levels would be a waste of resources, so I would love a LDR implementation!

DTTerastar, 2022-03-10:
Others have suggested a lux sensor. Is a photocell really viable? Why do lux sensors even exist?

freaked1234, 2022-03-10:
I am using a photoresistor together with a PIR sensor to activate lights in 2 rooms on a ESP8266 running ESPhome and it works great. 
It is simply reading the resistance via ADC and I checked the value for "too dark" via trial and error. Great thing is that these things just cost a few Cents and do the job.
Would be great to be able to use the PIR + LDR on the same ESP that detects presence. 

How else do people reliably decide its dark enough to turn on the lights? 

...... while typing this I realized that a lux sensor probably is just a photoresistor on a PCB with the 10k resistor onboard ^^
So yea, if you could add an option to read the resistance on an ADC pin where we can connect a LDR it would be great.

closed 2022-05-16 by stale[bot]