ESPresense/ESPresense #378: New 2.0.37 dont have all files, failed to install from www, 404 errpr

askthomas, 2022-03-09:
**Describe the bug**

A clear and concise description of what the bug is.  If it relates to fingerprinting or presense please include the id you expect/monitor.

**Config**

- Version:
- Active scan enabled:
- Include filter:
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor:
- Device (be specific):

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.
[image]

DTTerastar, 2022-03-10:
Yes this happens every time I make a release for 20 minutes because github it's still building the releases.

askthomas, 2022-03-10:
Cant it be compiled first, then repoint latest to it when done. It feels a bot strange to have to point live linke with lates to en empy folder.

closed 2022-05-28 by stale[bot]