ESPresense/ESPresense #379: LED for PIR detection.

askthomas, 2022-03-09:
**Is your feature request related to a problem? Please describe.**
Option to add LED for PIR detetion

**Describe the solution you'd like**
When PIR detection is made, it would be nice to have an LED to light up,

simonyel, 2022-04-11:
I would like to have an option to add a RGB cathode LED to the mix and port it as an entity to HA. That way I can show the status inside all the ESPresense enabled rooms.  Ex: White as a night light, RED as alarm set, GREEN flash as device detected, BLUE flash as BLE beacon detected, etc. would be a very powerful add-on. I do this with all my ESPhome devices, but since BLE isn't a component there I'm looking to migrate.

DTTerastar, 2022-04-11:
Yeah, I agree.  It would be nice to have the led controllable via an entity.

DTTerastar, 2022-08-30:
This is implemented in v3.  Set the pins on led 1, 2, and 3 to the pins of the led on pwm and it should work!

closed 2022-09-15 by DTTerastar