ESPresense/ESPresense #380: Giga-set G -tagID not unique anymore

FlingersDH7, 2022-03-09:
After update 20.0.36 and also with 20.0.37, the ID of gigaset g-tag is not unique anymore. All my ESP32's show the same ID's. In earlier versions of ESPresense the ID  always was the MAC address of the tags. This makes tracking of multiple tags impossible.

Home assistant also shows the tags as not_home.
This is my config which worked before.

- platform: mqtt_room
  device_id: "7c2f80c3a1c6"
  name: 'ESPresense Presence tag 1'
  state_topic: 'espresense/rooms'
  timeout: 60
  away_timeout: 120

Screenshot of the ID's running on version 20.0.37

[image]

DTTerastar, 2022-03-10:
Can you try the pre-release and see if that fixes it?

FlingersDH7, 2022-03-10:
It looks like it's working again. Thank You very much.

closed 2022-03-11 by DTTerastar