ESPresense/ESPresense #382: Atom lite stops detecting devices after some time

gmathur, 2022-03-10:
**Describe the bug**

BT scanning stops working after some when using atom lite as beacon. The presence detection works great for some time but after a while each beacon stops reporting any BT devices detected. 

Each beacon seems alive as observed within home assistant - it’s config page is accessible and I verified it’s uploading the heartbeat to Mqtt correctly.

To fix the problem just unplug and plug the atom lite back in and it starts working right away.

**Config**

- Version: 2.0.36
- Active scan enabled: yes - trying to scan for just one iPhone fingerprint
- Include filter: yes tracking one device
- Exclude filter: 
- Query filter: no

**Hardware**

- Firmware flavor: 2.0.36
- Device (be specific): atom lite (m5 stack)

DTTerastar, 2022-03-10:
I believe this is corrected in the pre-release, can you check?

gmathur, 2022-03-10:
Ill give it a whirl today.

gmathur, 2022-03-12:
No dice. Updates still not coming in from the devices.

DTTerastar, 2022-03-12:
Ok, I figured it out, github did something odd with b2 and compiled master, so no wonder you saw no change. b3 is going out now.

gmathur, 2022-03-12:
Ok I see b3 now. Ill give it a whirl

gmathur, 2022-03-13:
B4 is looking so much better. Tracking seems to be working quite reliably. I’ll keep giving it a whirl today and update

gmathur, 2022-03-14:
Overall this is a BIG upgrade. I think there is still a memory leak issue -- see the free memory graph below
[image]

DTTerastar, 2022-03-22:
I've seen my debug WROVER run for 5 days straight.  So if there is a memory leak it is very small.  Can this be closed?

closed 2022-05-28 by stale[bot]