ESPresense/ESPresense #384: Reboot for some unknown reason

hellcry37, 2022-03-11:
I just wanted to report that v2.0.36 and v2.0.37 have a weird bug that reboots all my nodes (5 nodes). I do not know exactly what is going on but all my nodes reboot after a period of time ~ 1 day or maybe a bit less. After reboot they work ok again for a period of time ~ 1 day then reboot again.

- all 5 nodes reboot
- not in the same time (not a power issue)
- they work ok after reboot for a while

All this make me believe this is a memory leak of some kind. If you want to investigate, thanks in advance!

Note: v2.0.35 does not have this issue for me I think. I noticed this first in v2.0.36

DTTerastar, 2022-03-11:
The prerelease has a fix for the memory leak.

hellcry37, 2022-03-11:
perfect waiting for the release. i'll close this as soon as it's done

mehdic, 2022-03-12:
I am on the 38b2, and 4 of my devices keep rebooting constantly then drop off the wifi completly. Is this bein worked on? Can i help with some logs ?

mehdic, 2022-03-12:
Woke us this morning and all my devices are completly down.

DTTerastar, 2022-03-12:
> I am on the 38b2, and 4 of my devices keep rebooting constantly then drop off the wifi completly. Is this bein worked on? Can i help with some logs ?

Yes,Please open a separate issue w/ all your details and logs.  I can't figure out the pattern quite yet.

hellcry37, 2022-03-16:
this is solved i think in 2.1.0

closed 2022-03-16 by hellcry37

mehdic, 2022-03-16:
yes it is

DTTerastar, 2022-03-16:
Awesome!  So glad to have squashed that bug.

hellcry37, 2022-03-16:
thanks very much