ESPresense/ESPresense #386: MiWatch would be great

DuploDom, 2022-03-11:
Dear ESPresense Team,
Your tool works great with my phone. I would love to use it with my MiWatch color.
I have MAC and ID but HomeAssistant shows just "distance: null".
Is there a way to make it work?

best regards and
Thanks anyway for Your great job
Dominik

DTTerastar, 2022-03-12:
It sounds more like a configuration issue.  The mi stuff generally just works.

DuploDom, 2022-03-13:
Hi there,
It for sure a issue in my side and not with the tool.
This morning I shorty tought it works, because it showed a distance. But it didnt change at all.
When I restart the Watch it gets an update. Is there a trick or workaround?
Also the Terminal shows the MAC and ID only once when the Watch restarts.

best regards
Dominik

DTTerastar, 2022-03-13:
[Look i](https://github.com/ESPresense/ESPresense/issues/184#issuecomment-1066086925) has info on how to turn on discoverable mode.

closed 2022-05-19 by stale[bot]