ESPresense/ESPresense #387: Devices being detected despite they left home 2 days ago

blacknell, 2022-03-12:
**Describe the bug**

Despite leaving the property 2 days ago, devices are still be temporarily reported as being present. I first noticed this when out for a day trip, but it is definitely happening regularly.

**Config**

- Version: 2.0.37
- Active scan enabled: Y
- Include filter: N
- Exclude filter: N
- Query filter: N

```
  - platform: mqtt_room
    device_id: "apple:1006:10-7"
    name: "Paul's iPhone (ESP)"
    state_topic: 'espresense/rooms'
    timeout: 60
    away_timeout: 180
```

**Hardware**

- Firmware flavor: ESP32
- Device (be specific): ESP32-D0WDQ6

**Logs**

Don't know how to do this

**Screenshots**
[image]

[image]

blacknell, 2022-03-15:
Further update. For the last several days ESPresense has been reporting the phones as present for short periods of time. None of the phones are in the property as it is empty for several days. Last night I forced a reboot of the ESP and now only one device is occasionally being reported as being present. These durations are for just a few minutes but I'm at a loss as to how this can be possible if they are not in the property.

I'm monitoring MQTT to see events generated by the module and will report if I see something

@DTTerastar  I would appreciate it if someone could acknowledge this problem. I'm keen to help

blacknell, 2022-03-15:
I just captured this on mqtt (this device is my phone) under the topic espresense/devices. There were 4 messages sent between 08:32:02 and 08:32:13 this morning.

```
{
   "id":"apple:1006:10-7",
   "disc":"431eac2f9f65",
   "idType":150,
   "rssi@1m":-65,
   "rssi":-88,
   "raw":4.54,
   "distance":5.57,
   "speed":-0.06,
   "mac":"73f695217c8c",
   "interval":1328
}
```

blacknell, 2022-03-15:
I've since seen 5 apple devices reported. `apple:1006:10-7` is mine, `apple:1006:10-12` is my wife's.

Also see are `apple:1005:9-8`, `apple:1005:9-26` & `apple:1007:11-1`. These are unknown to me but could be neighbour's phones.

My think is that the unique id derived is not so unique?

Zipties, 2022-03-18:
The apple device IDs are not fully unique, another device that is the same model and build number will report as the same device

blacknell, 2022-03-29:
> 

When you say enable query for those IDs may help, can you elaborate please

DTTerastar, 2022-04-02:
The best solution is to run the room-assistant app so you get a unique id.  Make the Query "apple:1006:10-7 apple:1006:10-12" and then ESPresense will ask every phone with those ids if they have a room assistant uuid.  Only your phone or your wife's will answer with that uuid, it'll "upgrade" the id from those to ones that start with roomAssistant: once it gets the result.

DTTerastar, 2022-04-06:
<https://github.com/mKeRix/room-assistant-companion-ios/issues/5> is it this?

51av0sh, 2022-07-11:
I encountered the same issue today. I've been using ESPresense for a few months with my two android phones and working very well. I decided to add my wife's devices (iPhone and Apple Watch) to the mix last night. Everything was working well until this morning when she left for work and her iPhone was still reporting in our master bedroom and actually updating the distance every few seconds for about 30min. Hers is the only iPhone in our house and the distance was reporting less than 1m from the ESP32. Any thouhgts?

DTTerastar, 2022-07-12:
What kind of ID were you using?

51av0sh, 2022-07-12:
I'm using the apple:10* id. Query didn't return anything different.

DTTerastar, 2022-07-12:
Query should return a model number which is more reliable. I'm currently working on a way I finally defeating the random iOS Mac addresses so soon me may not even have to worry about this.

51av0sh, 2022-07-12:
Yeah, which is odd. The Apple Watch does return a different ID which includes the model number. The iPhone is not responding I guess. I don't get a new for that one.

blacknell, 2022-07-12:
> I'm currently working on a way of finally defeating the random iOS Mac addresses so soon me may not even have to worry about this.

Let me know if I can help test beta code - this fix would be important to me

DTTerastar, 2022-08-12:
The new irk functionality in the beta is way better that query check out #580

closed 2022-08-12 by DTTerastar