ESPresense/ESPresense #394: Reboot loop

bobbanks, 2022-03-13:
**Describe the bug**

Reboot loop every few minutes.

**Config**

- Version: 2.0.38b4 - 2.0.38b6
- Active scan enabled: Y
- Include filter: N
- Exclude filter: N
- Query filter: N

**Hardware**

- Firmware flavor: verbose 
- Device (be specific): Teyleten Robot ESP32S ESP-WROOM-32 

**Logs**

`Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble) 
Core 1 register dump:
PC      : 0x40080080  PS      : 0x00060436  A0      : 0x800f8b48  A1      : 0x3ffea364  
A2      : 0x3ffbaa00  A3      : 0x3ffc9b04  A4      : 0x3ffea365  A5      : 0x800f8971  
A6      : 0x3ffe7a60  A7      : 0x3ffe85d0  A8      : 0x00000001  A9      : 0x3ffea3c8  
A10     : 0x00000001  A11     : 0x3ffc9b04  A12     : 0x3ffea365  A13     : 0x3ffbaa20  
A14     : 0x3ffc9b04  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40080080:0x3ffea364

Rebooting...
`
`
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble) 
Core 1 register dump:
PC      : 0x40080080  PS      : 0x00060e36  A0      : 0x800f8b48  A1      : 0x3ffebff4  
A2      : 0x3ffbaa00  A3      : 0x3ffc9b04  A4      : 0x3ffebff5  A5      : 0x800f8971  
A6      : 0x3ffe7a60  A7      : 0x3ffe85d0  A8      : 0x00000001  A9      : 0x3ffec058  
A10     : 0x00000001  A11     : 0x3ffc9b04  A12     : 0x3ffebff5  A13     : 0x3ffbaa20  
A14     : 0x3ffc9b04  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40080080:0x3ffebff4

Rebooting...
`
`
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble) 
Core 1 register dump:
PC      : 0x40080080  PS      : 0x00060e36  A0      : 0x800f8b48  A1      : 0x3ffe2178  
A2      : 0x3ffbaa00  A3      : 0x3ffc9b04  A4      : 0x3ffe2179  A5      : 0x800f8971  
A6      : 0x3ffe7a60  A7      : 0x3ffe85d0  A8      : 0x00000001  A9      : 0x3ffe21dc  
A10     : 0x00000001  A11     : 0x3ffc9b04  A12     : 0x3ffe2179  A13     : 0x3ffbaa20  
A14     : 0x3ffc9b04  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40080080:0x3ffe2178

Rebooting...
`

**Screenshots**

If applicable, add screenshots to help explain your problem.

bobbanks, 2022-03-13:
Doesn't happen on v2.0.34

DTTerastar, 2022-03-13:
I need more info about what device is causing it.  If you're already running verbose you should be able to see it in the logs right before the Guru.  I understand people don't want their non random macs in github issue but any info you can give will help track it down. We're using a newer version of NimBLE that seems to like to be recursive forever.

ChirpyTurnip, 2022-03-13:
Hi,

I have just installed the latest ESPresence + Sensors (Verbose) via USB cable to an official Espressif ESP32-WROOM-32U DevKit C. System reboots instantly and continuously. There are *NO* sensors attached....it's just the board + USB power supply

Log snippet:

```

I (2793) wifi:AP's beacon interval = 102400 us, DTIM period = 1
192.168.20.200
Version:    v2.0.37
IP address:   192.168.20.200
DNS address:  192.168.1.6
Hostname:     espresense-lounge
Room:         Lounge
MQTT server:  192.168.1.10:1883
Max Distance: 5.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
DHT11 Sensor: disabled
DHT22 Sensor: disabled
BH1750_I2c Sensor:  on bus 1
BME280_I2c Sensor:  on bus 1
TSL2561_I2c Sensor:  on bus 1

Query:        
Include:      
Exclude:      

Initialized I2C Bus 1
1 New   | MAC: b03829c14a42, ID: name:hb-00046189                                             
Verbose | d8e0e12eb941                                              MD: 75004204018040d8e0e12eb941dae0e12eb94001010000000000
1 New   | MAC: d8e0e12eb941, ID: samsung:d8e0e12eb941                                         
Verbose | 7a005ee5caa8                                              MD: 7500021821a1553e34f8f4a9a9321c3b0dde64ad
1 New   | MAC: 7a005ee5caa8, ID: samsung:7a005ee5caa8                                         
Verbose | samsung:d8e0e12eb941                                      MD: 75004204018040d8e0e12eb941dae0e12eb94001010000000000
Verbose | 66082b376474                                              MD: 060001092002d2c0f5630682a3d0bbedb7aca266389f18b9bce651d4e7
1 New   | MAC: 66082b376474, ID: msft:cdp:0902                                                d2c0f5630682a3d0bbedb7aca266389f18b9bce6
abort() was called at PC 0x401b071b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ff00:0x3ffe7350 0x4009017d:0x3ffe7370 0x401b071b:0x3ffe7390 0x401b0762:0x3ffe73b0 0x401afe3b:0x3ffe73d0 0x401b0172:0x3ffe73f0 0x401b536d:0x3ffe7410 0x401b54e2:0x3ffe7430 0x401b560d:0x3ffe7470 0x400fc951:0x3ffe7490 0x400ff0f3:0x3ffe74b0 0x400fac2a:0x3ffe74f0 0x400fae0f:0x3ffe7580 0x400d2aae:0x3ffe75a0 0x400e4019:0x3ffe75c0 0x400e74de:0x3ffe7600 0x400ec3b9:0x3ffe7670 0x400ec465:0x3ffe76b0 0x400ec6b1:0x3ffe76d0 0x400eaf69:0x3ffe76f0 0x400f20da:0x3ffe7710 0x400e353f:0x3ffe7730 0x4009118e:0x3ffe7750

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (94:3C:C6:6D:2C:D4)...
I (95) wifi:wifi driver task: 3ffd6458, prio:23, stack:3584, core=0
I (430) wifi:wifi firmware version: dc30037
I (430) wifi:config NVS flash: enabled
I (430) wifi:config nano formating: disabled
I (431) wifi:Init data frame dynamic rx buffer num: 32
I (435) wifi:Init management frame dynamic rx buffer num: 32
I (441) wifi:Init management short buffer num: 32
I (445) wifi:Init dynamic tx buffer num: 32
I (449) wifi:Init static rx buffer size: 1600
I (453) wifi:Init static rx buffer num: 16
I (457) wifi:Init dynamic rx buffer num: 32
I (548) wifi:mode : sta (94:3c:c6:6d:2c:d4)
Connecting to WiFi SSID 'Slytherin'..............I (2618) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2619) wifi:state: init -> auth (b0)
I (2630) wifi:state: auth -> assoc (0)
I (2639) wifi:state: assoc -> run (10)
.I (2666) wifi:connected with Slytherin, aid = 4, channel 11, BW20, bssid = d6:21:f9:5c:56:ee
I (2667) wifi:security type: 3, phy: bgn, rssi: -36
I (2675) wifi:pm start, type: 1

I (2679) wifi:AP's beacon interval = 102400 us, DTIM period = 1
192.168.20.200
Version:    v2.0.37
IP address:   192.168.20.200
DNS address:  192.168.1.6
Hostname:     espresense-lounge
Room:         Lounge
MQTT server:  192.168.1.10:1883
Max Distance: 5.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
DHT11 Sensor: disabled
DHT22 Sensor: disabled
BH1750_I2c Sensor:  on bus 1
BME280_I2c Sensor:  on bus 1
TSL2561_I2c Sensor:  on bus 1

Query:        
Include:      
Exclude:      

Initialized I2C Bus 1
abort() was called at PC 0x401b071b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ff00:0x3ffe6c40 0x4009017d:0x3ffe6c60 0x401b071b:0x3ffe6c80 0x401b0762:0x3ffe6ca0 0x401afe3b:0x3ffe6cc0 0x401b0172:0x3ffe6ce0 0x401b536d:0x3ffe6d00 0x401b54e2:0x3ffe6d20 0x401b560d:0x3ffe6d60 0x400fc951:0x3ffe6d80 0x400ff0f3:0x3ffe6da0 0x400fac2a:0x3ffe6de0 0x400fae0f:0x3ffe6e70 0x400d2aae:0x3ffe6e90 0x400e4019:0x3ffe6eb0 0x400e74de:0x3ffe6ef0 0x400ec3b9:0x3ffe6f60 0x400ec465:0x3ffe6fa0 0x400ec6b1:0x3ffe6fc0 0x400eaf69:0x3ffe6fe0 0x400f20da:0x3ffe7000 0x400e353f:0x3ffe7020 0x4009118e:0x3ffe7040

Rebooting...
```

Hope this helps!

ChirpyTurnip, 2022-03-13:
Just tried with the non-sensor version of the build. Same result:

```

I (2793) wifi:AP's beacon interval = 102400 us, DTIM period = 1
192.168.20.200
Version:    v2.0.37
IP address:   192.168.20.200
DNS address:  192.168.1.6
Hostname:     espresense-lounge
Room:         Lounge
MQTT server:  192.168.1.10:1883
Max Distance: 5.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
DHT11 Sensor: disabled
DHT22 Sensor: disabled
BH1750_I2c Sensor:  on bus 1
BME280_I2c Sensor:  on bus 1
TSL2561_I2c Sensor:  on bus 1

Query:        
Include:      
Exclude:      

Initialized I2C Bus 1
1 New   | MAC: b03829c14a42, ID: name:hb-00046189                                             
Verbose | d8e0e12eb941                                              MD: 75004204018040d8e0e12eb941dae0e12eb94001010000000000
1 New   | MAC: d8e0e12eb941, ID: samsung:d8e0e12eb941                                         
Verbose | 7a005ee5caa8                                              MD: 7500021821a1553e34f8f4a9a9321c3b0dde64ad
1 New   | MAC: 7a005ee5caa8, ID: samsung:7a005ee5caa8                                         
Verbose | samsung:d8e0e12eb941                                      MD: 75004204018040d8e0e12eb941dae0e12eb94001010000000000
Verbose | 66082b376474                                              MD: 060001092002d2c0f5630682a3d0bbedb7aca266389f18b9bce651d4e7
1 New   | MAC: 66082b376474, ID: msft:cdp:0902                                                d2c0f5630682a3d0bbedb7aca266389f18b9bce6
abort() was called at PC 0x401b071b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ff00:0x3ffe7350 0x4009017d:0x3ffe7370 0x401b071b:0x3ffe7390 0x401b0762:0x3ffe73b0 0x401afe3b:0x3ffe73d0 0x401b0172:0x3ffe73f0 0x401b536d:0x3ffe7410 0x401b54e2:0x3ffe7430 0x401b560d:0x3ffe7470 0x400fc951:0x3ffe7490 0x400ff0f3:0x3ffe74b0 0x400fac2a:0x3ffe74f0 0x400fae0f:0x3ffe7580 0x400d2aae:0x3ffe75a0 0x400e4019:0x3ffe75c0 0x400e74de:0x3ffe7600 0x400ec3b9:0x3ffe7670 0x400ec465:0x3ffe76b0 0x400ec6b1:0x3ffe76d0 0x400eaf69:0x3ffe76f0 0x400f20da:0x3ffe7710 0x400e353f:0x3ffe7730 0x4009118e:0x3ffe7750

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (94:3C:C6:6D:2C:D4)...
I (95) wifi:wifi driver task: 3ffd6458, prio:23, stack:3584, core=0
I (430) wifi:wifi firmware version: dc30037
I (430) wifi:config NVS flash: enabled
I (430) wifi:config nano formating: disabled
I (431) wifi:Init data frame dynamic rx buffer num: 32
I (435) wifi:Init management frame dynamic rx buffer num: 32
I (441) wifi:Init management short buffer num: 32
I (445) wifi:Init dynamic tx buffer num: 32
I (449) wifi:Init static rx buffer size: 1600
I (453) wifi:Init static rx buffer num: 16
I (457) wifi:Init dynamic rx buffer num: 32
I (548) wifi:mode : sta (94:3c:c6:6d:2c:d4)
Connecting to WiFi SSID 'Slytherin'..............I (2618) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2619) wifi:state: init -> auth (b0)
I (2630) wifi:state: auth -> assoc (0)
I (2639) wifi:state: assoc -> run (10)
.I (2666) wifi:connected with Slytherin, aid = 4, channel 11, BW20, bssid = d6:21:f9:5c:56:ee
I (2667) wifi:security type: 3, phy: bgn, rssi: -36
I (2675) wifi:pm start, type: 1

I (2679) wifi:AP's beacon interval = 102400 us, DTIM period = 1
192.168.20.200
Version:    v2.0.37
IP address:   192.168.20.200
DNS address:  192.168.1.6
Hostname:     espresense-lounge
Room:         Lounge
MQTT server:  192.168.1.10:1883
Max Distance: 5.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
DHT11 Sensor: disabled
DHT22 Sensor: disabled
BH1750_I2c Sensor:  on bus 1
BME280_I2c Sensor:  on bus 1
TSL2561_I2c Sensor:  on bus 1

Query:        
Include:      
Exclude:      

Initialized I2C Bus 1
abort() was called at PC 0x401b071b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ff00:0x3ffe6c40 0x4009017d:0x3ffe6c60 0x401b071b:0x3ffe6c80 0x401b0762:0x3ffe6ca0 0x401afe3b:0x3ffe6cc0 0x401b0172:0x3ffe6ce0 0x401b536d:0x3ffe6d00 0x401b54e2:0x3ffe6d20 0x401b560d:0x3ffe6d60 0x400fc951:0x3ffe6d80 0x400ff0f3:0x3ffe6da0 0x400fac2a:0x3ffe6de0 0x400fae0f:0x3ffe6e70 0x400d2aae:0x3ffe6e90 0x400e4019:0x3ffe6eb0 0x400e74de:0x3ffe6ef0 0x400ec3b9:0x3ffe6f60 0x400ec465:0x3ffe6fa0 0x400ec6b1:0x3ffe6fc0 0x400eaf69:0x3ffe6fe0 0x400f20da:0x3ffe7000 0x400e353f:0x3ffe7020 0x4009118e:0x3ffe7040

Rebooting...
```

:-(

bobbanks, 2022-03-15:
Can't explain but I can't replicate now.  Tried a few different versions.  Happen to be using a clean ESP32.  Not sure if it's related. I'll keep an eye on it.

bobbanks, 2022-03-15:
Spoke too soon.  Here's 3 crashes.  Happened after unplugged from USB port and powered via block.  

```
Using 'COM3' as serial port.
Showing logs:
[23:28:23]Connecting to WiFi (C8:C9:A3:C7:AE:70)...
[23:28:26]Connecting to WiFi SSID 'taconet'................10.100.1.197
[23:28:26]Firmware:     esp32
[23:28:26]Version:      v2.1.0
[23:28:26]IP address:   10.100.1.197
[23:28:26]DNS address:  10.100.1.1
[23:28:26]Hostname:     espresense-living-room
[23:28:26]Room:         Living Room
[23:28:26]MQTT server:  10.100.1.143:1883
[23:28:26]Max Distance: 16.00
[23:28:26]PIR Sensor:   disabled
[23:28:26]Radar Sensor: disabled
[23:28:26]Query:        
[23:28:26]Include:      
[23:28:26]Exclude:      
[23:28:26]Known Macs:   
[23:28:26]Count Ids:    
[23:28:26]1 New   | MAC: b03829c0f7b8, ID: b03829c0f7b8                                                 
[23:28:26]1 New   | MAC: 6bb9ff30c6e2, ID: apple:1005:9-26                                              0f18be4a11
[23:28:26]1 New   | MAC: 231a9cc4c1df, ID: exp:20                                                       778ea58ce282109298aa24a3f2532872b91d1343
[23:28:26]1 New   | MAC: 74f61c0220c0, ID: 74f61c0220c0                                                 
[23:28:27]1 New   | MAC: 775380d9b739, ID: exp:20                                                       bed7b0b2f31ca37d56a6649402e371896bcf7dd6
[23:28:27]1 New   | MAC: 4f043667afe0, ID: iBeacon:52137873-d970-4b2c-b86e-fcffbbe176c0-100-1           
[23:28:27]1 New   | MAC: 7d5744a294b6, ID: msft:cdp:0902                                                4dfa4cf009edd19273694e99202bd4c282b3ea1f
[23:28:27]1 New   | MAC: 59baff3e3442, ID: apple:1006:10-12                                             171af5289b00
[23:28:27]1 New   | MAC: 74f61c0220d2, ID: 74f61c0220d2                                                 
[23:28:27]1 New   | MAC: 6683f439045c, ID: samsung:6683f439045c                                         
[23:28:27]1 New   | MAC: 9893d8992f6f, ID: iBeacon:74278bda-b644-4520-8f0c-720eaf059935-0-47251         
[23:28:27]1 New   | MAC: 7fa50608a429, ID: msft:cdp:0902                                                8cfb74e4be2f8e4409a269fbaee75ec05dafc97d
[23:28:27]1 New   | MAC: 5da3465884e2, ID: apple:1005:9-12                                              051cf352bd
[23:28:27]1 New   | MAC: 192efd7449d9, ID: msft:cdp:0902                                                7bf342aa16ee7f5723cb4b2fbc7c465fffdef413
[23:28:27]1 New   | MAC: 79852fbb1952, ID: apple:1006:10-8                                              3e1e0b97a944
[23:28:27]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[23:28:27]Debug exception reason: Stack canary watchpoint triggered (ble) 
[23:28:27]Core 1 register dump:
[23:28:27]PC      : 0x40080080  PS      : 0x00060636  A0      : 0x800f8858  A1      : 0x3ffe9cb8  
[23:28:27]A2      : 0x3ffbaa00  A3      : 0x3ffc9a9c  A4      : 0x3ffe9cb9  A5      : 0x800f8681  
[23:28:28]A6      : 0x3ffe7a60  A7      : 0x3ffe85d0  A8      : 0x00000001  A9      : 0x3ffe9d1c  
[23:28:28]A10     : 0x00000001  A11     : 0x3ffc9a9c  A12     : 0x3ffe9cb9  A13     : 0x3ffbaa20  
[23:28:28]A14     : 0x3ffc9a9c  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
[23:28:28]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[23:28:28]
[23:28:28]ELF file SHA256: 0000000000000000
[23:28:28]
[23:28:28]Backtrace: 0x40080080:0x3ffe9cb8
[23:28:28]
[23:28:28]Rebooting...
[23:28:28]ets Jun  8 2016 00:22:57
[23:28:28]
[23:28:28]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:28:28]configsip: 0, SPIWP:0xee
[23:28:28]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:28:28]mode:DIO, clock div:2
[23:28:28]load:0x3fff0018,len:4
[23:28:28]load:0x3fff001c,len:1044
[23:28:28]load:0x40078000,len:8896
[23:28:28]load:0x40080400,len:5828
[23:28:28]entry 0x400806ac
[23:28:28]Connecting to WiFi (C8:C9:A3:C7:AE:70)...
[23:28:30]Connecting to WiFi SSID 'taconet'...............10.100.1.197
[23:28:30]Firmware:     esp32
[23:28:30]Version:      v2.1.0
[23:28:30]IP address:   10.100.1.197
[23:28:30]DNS address:  10.100.1.1
[23:28:30]Hostname:     espresense-living-room
[... 88 lines cut ...]
[23:28:37]PC      : 0x40080080  PS      : 0x00060036  A0      : 0x800f8858  A1      : 0x3ffea978  
[23:28:37]A2      : 0x3ffbaa00  A3      : 0x3ffc9a9c  A4      : 0x3ffea979  A5      : 0x800f8681  
[23:28:37]A6      : 0x3ffe71e0  A7      : 0x3ffe7d50  A8      : 0x00000001  A9      : 0x3ffea9dc  
[23:28:37]A10     : 0x00000001  A11     : 0x3ffc9a9c  A12     : 0x3ffea979  A13     : 0x3ffbaa20  
[23:28:37]A14     : 0x3ffc9a9c  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
[23:28:37]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[23:28:37]
[23:28:37]ELF file SHA256: 0000000000000000
[23:28:37]
[23:28:37]Backtrace: 0x40080080:0x3ffea978
[23:28:37]
[23:28:37]Rebooting...
[23:28:37]ets Jun  8 2016 00:22:57
[23:28:37]
[23:28:37]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:28:37]configsip: 0, SPIWP:0xee
[23:28:37]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:28:37]mode:DIO, clock div:2
[23:28:37]load:0x3fff0018,len:4
[23:28:37]load:0x3fff001c,len:1044
[23:28:37]load:0x40078000,len:8896
[23:28:37]load:0x40080400,len:5828
[23:28:37]entry 0x400806ac
[23:28:37]Connecting to WiFi (C8:C9:A3:C7:AE:70)...
```

mehdic, 2022-03-15:
I was having this problem since 2.0.34, i had tried all the versions since with the same problem.

That is until the latest memory fix (released now in the 2.1.0) which seems to have fixed the problem. I will keep an eye on it though.

ChirpyTurnip, 2022-03-15:
Same here....just flashed to 2.1.0 (ESP32 + Sensors (Verbose)) on a genuine
Espressif ESP32-WROOM-32U and it stayed up for more than 30 seconds for the
first time in a few weeks! So far been running for 5 minutes and looking
good.

Here's hoping we have a fix!  :-)

needs-coffee, 2022-03-17:
I am experiencing the same issue on 2.1.0 with the same errors as OP with an ESP32 DEVKIT DOIT v1 
Standard 2.1.0 with a PIR sensor attached, however the errors and rates are unchanged if i disconnect the sensor. 
29 reboots / 24 hours.
will install the verbose version and supply logs.

ChirpyTurnip, 2022-03-17:
Dang......I'm still going strong here:

```
**Attributes**
Uptime - 173,068
Firm - esp32-verbose-sensors
Rssi - -51
Ver - v2.1.0
Adverts - 4,686,755
Seen - 4,628,192
Reported - 464,448
FreeHeap - 121,712
MaxAllocHeap - 79,104
MemFrag - 35.01
ResetReason - PowerOn
```

In my case though I was previously boot looping with about 15 seconds between boot and reboot until it just locked up completely. There might be something else going on with your one as it is running for almost an hour before it dies....

gastonMM, 2022-03-19:
I had the same failure until I used these connection pins, with any other GND it failed
[image]

DTTerastar, 2022-03-22:
If you are still getting this, please follow this guide: https://github.com/ESPresense/ESPresense/blob/master/BUILDING.md 
While still connected to that computer monitor and catch it in the act.  It will decode the stack frames so we can see where exactly in the code the issue is occurring.  Without that I'm not sure how to proceed.

needs-coffee, 2022-04-02:
Have just pulled at 2gb verbose log for one of the sensors - 182 reboots over 48 hours. 
could not identify  a specific device/devices that were commonly in range before each reboot, seems to be different every time.
Backtrace is different every time (though not surprising)
with the log entry ```abort() was called at PC 0x401aa143 on core 1```
there was a common frequency with the PC but i am not sure how to interpret that.

-  0x401aa143 : 100
-  0x40086ca7 : 13
-  0x40086ca7 : 13
-  0x4018c43d : 7
-  0x40086cf5 : 7
-  0x4011fceb : 1
 
will build and run attached to vscode as suggested above

needs-coffee, 2022-04-02:
Stack trace for generic ESP32 with ESP32-Verbose-Sensors build from current master branch (27c6239). Running with PIR sensor on pin 27.

```
stack:
0x4008ff00: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x4009017d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x401affe7: esp_brownout_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/brownout.c:61
0x401b002e: esp_aes_setkey_hardware at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/hwcrypto/aes.c:433
0x401af707: esp_intr_alloc_intrstatus at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c:660
0x401afa3e: esp_int_wdt_cpu_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/int_wdt.c:121
0x401af7a5: esp_intr_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c:727
0x401cb68a: tcpip_adapter_get_ip6_linklocal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1199
0x40103b48: ble_sm_sc_dhkey_check_rx at D:\ESPresense/.pio\libdeps\esp32-verbose-sensors\NimBLE-Arduino\src\nimble\nimble\host\src/ble_sm_sc.c:907
0x400dc984: ArduinoJson6193_F1::MemoryPool::canAlloc(unsigned int) const at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Memory/MemoryPool.hpp:113
  \-> inlined by: ArduinoJson6193_F1::MemoryPool::allocRight(unsigned int) at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Memory/MemoryPool.hpp:199
  \-> inlined by: ArduinoJson6193_F1::VariantSlot* ArduinoJson6193_F1::MemoryPool::allocRight<ArduinoJson6193_F1::VariantSlot>() at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Memory/MemoryPool.hpp:195
  \-> inlined by: ArduinoJson6193_F1::MemoryPool::allocVariant() at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Memory/MemoryPool.hpp:59
  \-> inlined by: ArduinoJson6193_F1::CollectionData::addSlot(ArduinoJson6193_F1::MemoryPool*) at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:19
  \-> inlined by: ArduinoJson6193_F1::VariantData* ArduinoJson6193_F1::CollectionData::addMember<ArduinoJson6193_F1::ZeroTerminatedRamString, ArduinoJson6193_F1::LinkStringStoragePolicy>(ArduinoJson6193_F1::ZeroTerminatedRamString, ArduinoJson6193_F1::MemoryPool*, ArduinoJson6193_F1::LinkStringStoragePolicy) at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:43
  \-> inlined by: ArduinoJson6193_F1::VariantData* ArduinoJson6193_F1::CollectionData::getOrAddMember<ArduinoJson6193_F1::ZeroTerminatedRamString, ArduinoJson6193_F1::LinkStringStoragePolicy>(ArduinoJson6193_F1::ZeroTerminatedRamString, ArduinoJson6193_F1::MemoryPool*, ArduinoJson6193_F1::LinkStringStoragePolicy) at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:155
  \-> inlined by: ArduinoJson6193_F1::VariantData* ArduinoJson6193_F1::VariantData::getOrAddMember<ArduinoJson6193_F1::ZeroTerminatedRamString, ArduinoJson6193_F1::LinkStringStoragePolicy>(ArduinoJson6193_F1::ZeroTerminatedRamString, ArduinoJson6193_F1::MemoryPool*, ArduinoJson6193_F1::LinkStringStoragePolicy) at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:288
  \-> inlined by: ArduinoJson6193_F1::VariantRef ArduinoJson6193_F1::JsonDocument::getOrAddMember<char const>(char const*) at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:250
  \-> inlined by: ArduinoJson6193_F1::MemberProxy<ArduinoJson6193_F1::JsonDocument&, char const*>::getOrAddUpstreamMember() const at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:187
  \-> inlined by: ArduinoJson6193_F1::MemberProxy<ArduinoJson6193_F1::JsonDocument&, char const*>& ArduinoJson6193_F1::MemberProxy<ArduinoJson6193_F1::JsonDocument&, char const*>::operator=<char const>(char const*) at D:\ESPresense/.pio/libdeps/esp32-verbose-sensors/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:58
  \-> inlined by: Motion::SendDiscovery(ArduinoJson6193_F1::BasicJsonDocument<ArduinoJson6193_F1::DefaultAllocator>&) at D:\ESPresense/src/MotionSensors.cpp:101
0x4010a1f4: uECC_shared_secret at D:\ESPresense/.pio\libdeps\esp32-verbose-sensors\NimBLE-Arduino\src\nimble\ext\tinycrypt\src/ecc_dh.c:160
0x4009118e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

DTTerastar, 2022-04-03:
It looks like it blew up on sending discovery info.  But some of that stack trace looks suspect. tcpip_adapter_get_ip6_linklocal, huh?

needs-coffee, 2022-04-03:
I have looked at the backtrace from running this bare board without any sensors connected but config unchanged - 16 crash events, backtrace same as above.

I am currently running it again with a memory trace running to see if the crash is due to insufficient memory so will have a look at the memory allocations after its crashed a few times - however i see the main loop checks free memory regularly and there were no log messages about it so seems less likely. 

Checked my wifi config and there is only a ipv4 address assigned for this device, there is no ipv6 assigned or advertised.  
tcpip_adapter_get_ip6_linklocal  is defined in the framework-arduinoespressif32 but not currently found where/why its getting called but seems to be part of the ESP TCP/IP stack anyway.

Just to check i am reading the stacktrace right its innermost at the bottom right?

It crashed on Motion::SendDiscovery so seems to be the PIR code that its crashing on so i will compile and test it again after the memory trace has finished with the MotionSensors code commented out and see if this crashes as well.

needs-coffee, 2022-04-08:
An update on further debugging:
- After commenting out the Motion::SendDiscovery the crashes continued - However the stacktrace was no longer consistent and wildly different with each crash.
- Free heap and max alloc heap were never below 50k on any build
- using the stack high watermark to debug some of the FreeRTOS loops (scantask and reporttask) showed a high mark of ~5k with 7k allocated for each stack, so these loops dont appear to be overflowing.
- Stack Canary watchpoints were identified on several of the backtraces suggesting a stack overflow/corruption of 2 of the loops - ble and Tmr Svc 
- Other exception reasons were LoadProhibited - the memory addresses that were trying to be loaded were out of scope of any address on the esp32, also suggest overflow or corruption
- I reflashed with a JLINK instead of a serial programmer with a new ESP32 from the current master branch and it is now staying up for ~24 hours without crashing, will see if these stack traces provide any further insight.

Unfortunately i barbequed the original ESP32 miswiring the debug probe so that one is dead.
Not clear if there was a problem with the original ESP or if there is an ongoing memory issue with the current builds but will see if further stack traces provide insight.

@DTTerastar are you aware of any tool to specifically debug FreeRTOS on the ESP32, there is a plugin for eclipse from NXP that allows realtime monitoring of the stacks for each RTOS task but i have not been able to get this to work so far. Am essentially resorting to print level debugging for memory utilisation

DTTerastar, 2022-04-08:
WOW! Nice work!  I am stuck using print debugging most of the time.  Even the JTAG hasn't really been all that helpful. 
The WROVER on the Espressif board never seems to have the same issues as the other cheaper WROOM boards...

needs-coffee, 2022-04-12:
[image]

Managed to get the FreeRTOS plugin in eclipse by NXP working to look at the stack utilisation (will add documentation shortly) and it looks like we have a winner.

Left it running overnight and found it stuck on a watchpoint with the BLE stack overflowing. 

This is thankfully configurable with a macro ```CONFIG_BT_NIMBLE_TASK_STACK_SIZE```

I have doubled the stack size and have left it running to see if this stack is now large enough. I may be that at times of high bluetooth traffic the stack fills and overflows, however will leave it running to see if it is a leak.

DTTerastar, 2022-04-12:
wow!  very nice!!!!

DTTerastar, 2022-04-13:
Noting that Tasmota doesn't set it: https://github.com/arendst/Tasmota/search?q=CONFIG_BT_NIMBLE_TASK_STACK_SIZE

needs-coffee, 2022-04-13:
In theory the ble stack usage here would be expected to be higher than tasmota but time will tell with the debugging. 

Its now running at 8 hrs uptime (would have crashed by now) with the double stack but just checked the stacks and its at 4.5k for the ble stack usage, so there may be a leak. Will leave it running to see if this climbs or stabilises around there. 

The other stack that caused crashes in previous testing was the Tmr Svc with a default stack of 2k, however there is no easy way to increase this as it is part of the Arduino core - which is included as precompiled libraries so changing this within ```sdkconfig.h ``` or ```FreeRTOSConfig.h``` makes no difference as the libraries would need rebuilt with ESP-IDF.

DTTerastar, 2022-04-14:
> In theory the ble stack usage here would be expected to be higher than tasmota but time will tell with the debugging.

I'm not sure. We don't send much, it seems like just passively listening causes the issue, so wouldn't tasmota also see it?

Can't argue w/ what you're seeing tho, I think it might be "Extended" advertising devices perhaps?  My devices just don't see this particular issue so it's got to be some device I don't have...

commit 35323817 (2022-04-13): Increase ble task stack size (fixes #394) (#443)
closed 2022-06-20 by stale[bot]