ESPresense/ESPresense #396: Garmin

rbjoensen, 2022-03-13:
Hi
After auto update on March 6 it can not see my Garmin watch (Fenix 6) is there anything I can do to fix it? 
Sincerely. Rasmus

ozett, 2022-03-13:
more invisible garmin https://github.com/ESPresense/ESPresense/issues/365#issuecomment-1065109434

rbjoensen, 2022-03-13:
Hi previously my watch reported an id which was my mac address and when the watch was not connected to my mobile phone, my watch reported its id. 
Now my watch report an id md: 0087: 4 
My wife's garmin watch gets the same id 
When id was the mac address it worked. 
Sincerely. Rasmus

DTTerastar, 2022-03-13:
That id is enough info that I can make it use the id "garmin:{mac}".  Will be in next beta.

rbjoensen, 2022-03-13:
#DTTerastar 
Nice if you notify me when it's ready, I'll be happy to test it