ESPresense/ESPresense #398: Encrypted Connection to MQTT Broker does not work

gadric, 2022-03-13:
First of all, I would like to thank all contributers of this project. Currently, I am using room-assistant, but find it time-consuming to maintain so many pis with its full linux stacks. This project could save a lot of time and therefore, I am thinking of switching to ESPresense. I just brought an ESP32 for testing. As of now, my iPhone does not work so good, but my MiBand 6 seems to be pretty good. 

**Describe the bug**
I would like to connect to the MQTT server in encrypted my over the port 8883. However, connection to it results in an error. Without encryption it works fine: 

ESP log
```
[12:16:03]0 Reconnect timer
[12:16:03]0 Reconnecting to MQTT...
[12:16:03][E][main.cpp:257] onMqttDisconnect(): Disconnected from MQTT; reason 0
```

Home Assistant Log
```
1647170188: New connection from 192.168.xxx.31 on port 8883.
1647170188: OpenSSL Error[0]: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
1647170188: Socket error on client <unknown>, disconnecting.
```
**Config**

- Version: 2.0.37
- Active scan enabled: yes
- Include filter: yes
- Exclude filter: no
- Query filter: no

**Hardware**

- Firmware flavor: ESP32
- Device (be specific): AZDelivery ESP32 (ESP32-WROOM-32), CP2102

**Logs**

[12:16:03]0 Reconnect timer
[12:16:03]0 Reconnecting to MQTT...
[12:16:03][E][main.cpp:257] onMqttDisconnect(): Disconnected from MQTT; reason 0

DTTerastar, 2022-03-13:
yes, ssl takes 40-80k of memory, it is disabled for space. If I was going to enable it it would only be on WROVER modules because they have more memory.  Since all these packets are flying unencrypted through the air anyways, is ssl actually needed?  Probably the only secret item is your mqtt password so just don't use one :)

gadric, 2022-03-13:
Well, the WIFI connection is encrypted, so it does not really fly around unencrypted. But I understand if it takes more memory. I am not really aware of the current space usage. Does it make sense to enable it for the devices with more memory?

DTTerastar, 2022-03-13:
The bluetooth packets we are sending are in plaintext is what I mean. What WROVERs are you going to test with?  They do cost $5-10 more than the standard.  Make sure not to get a WROVER-S2 because those are a single core.

gadric, 2022-03-13:
I am using this one: https://www.az-delivery.de/products/esp-32-dev-kit-c-v4

DTTerastar, 2022-03-13:
That's a ROOM not a ROVER.  Doesn't have the extra 8mb of ram.

gadric, 2022-03-13:
Hm, ok. Then I guess I currently do not have the hardware for testing ...

closed 2022-05-19 by stale[bot]