ESPresense/ESPresense #400: HA dont create sensor

pidiman, 2022-03-13:
**Describe the bug**

Hi guys, I’m coming here for advice. I installed espresense on esp32. In HA I see that the sensor is working properly. He shows me the room - workroom. Then I config my Iphone 12 to sensor's configuration file:

```
  - platform: mqtt_room
    name: "FajnyIphone12"
    device_id: "apple:iphone13-2"
    state_topic: "espresense/rooms"
    timeout: 60
    away_time: 120
```
But HA dont create this Sensor.. I see these infos in mqtt explorer:

[image]

Any advice ? Thanks a lot!

**Config**

- Version: 2.0.26
- Active scan enabled: yes
- Include filter:
- Exclude filter:
- Query filter: 1

**Hardware**

- Firmware flavor: 
- Device (be specific): ESP32-D0WDQ6

**Logs**
```
1647196491: New client connected from 192.168.0.127 as esp32-486f06a4ae30 (p2, c1, k15, u'mqttuser').
1647196499: New connection from 192.168.0.127 on port 1883.
1647196499: Client esp32-486f06a4ae30 already connected, closing old connection.
1647196499: New client connected from 192.168.0.127 as esp32-486f06a4ae30 (p2, c1, k15, u'mqttuser').
1647196507: New connection from 192.168.0.127 on port 1883.
1647196507: Client esp32-486f06a4ae30 already connected, closing old connection.
1647196507: New client connected from 192.168.0.127 as esp32-486f06a4ae30 (p2, c1, k15, u'mqttuser').
1647196515: New connection from 172.30.32.2 on port 1883.
```

mehdic, 2022-03-19:
Check the entities page. You will find it there.

mehdic, 2022-03-19:
And update to 2.1.0. It is the most stable version now, as far as I know.

pidiman, 2022-03-21:
I updated esp to 2.1.0, everything is like before.. 

I see info in MQTT explorer..
I see info in espresense terminal..

But my HA dont create any entities.. 

```
  - platform: mqtt_room
    device_id: "apple:1005:9-26"
    name: 'Apple1005'
    state_topic: 'espresense/rooms'
    timeout: 60
    away_timeout: 120

  - platform: mqtt_room
    device_id: "apple:1007:11-8"
    name: 'Apple1007'
    state_topic: 'espresense/rooms'
    timeout: 60
    away_timeout: 120
```