ESPresense/ESPresense #401: PIR Motion sensors very bad concictency detecting motion

freaked1234, 2022-03-14:
I am filing this as bug report here because I have 2 different PIR motion sensors (5pcs each) around and when using them on my esp82 boards with ESP home I can get them to a point where I almost get any wrong detections. OK there are some modules that are just crap and dont work at all but most do.
When I am using known working PIR detectors on my espresense devices I get overwhelmed with false positives and I dont have any idea what is going on....
For example I have one that is 2meters away from the ESP32 module and faces a whole room from a top corner that is pretty unreliable on espresense and I have one in a very small room 20cm away from the module that keeps detecting false positives even tho there is nothing inside the room. 
Any ideas?

freaked1234, 2022-03-14:
Update: I even just flashed my living room ESP32 that was running espresense with ESPHome without any other changes and the false positives are gone...

Rappoman, 2022-03-14:
I am experiencing the same issue.  Using the same boards and PIRs with ESPHome the false positives are non-existent.

freaked1234, 2022-03-18:
I did not know that since I am also new here but I am interested in any solution since I like the core functionality and would love to keep using it.
I´d hope that the project creator chimes in some time soon but I understand that he is busy.

Photo64, 2022-03-18:
Unfortunately it doesn't look like the sensors bin has made a difference. Over a 15 minute period I received 8 false triggers so there still appears to be an issue. I like that I could add additional sensors with this bin but motion is the one I cold use the most. Hopefully it can get resolved.

Rappoman, 2022-03-18:
@Photo64 I did not realise that either I will have a look at these later and report back as well.  Thanks for pointing this out.

Rappoman, 2022-03-18:
@Photo64 out of interest do you have your esp32 and PIR in a case or is there some distance between the two?  Because I think so of my issues may relate to the proximity of the pir to the board.  A Google search suggests this is a known issue.

Photo64, 2022-03-18:
@Rappoman - It is in a case. I can try moving it apart but that sort of defeats the purpose of keep it all in a nice little package. I have other boards with the same motion sensor in a case that have no problems.

Rappoman, 2022-03-18:
@Photo64 I agree.  But I have taken the one I was having the most false positives out of its case and that seems to have resulted in no false readings for the last hour.  Whereas when it was in the case they were very frequent.

Photo64, 2022-03-18:
@Rappoman - Turns out you may be on to something. I removed it from the case and extended it as far as I could which was only about 3 inches and I haven't had a false positive since I did it, over an hour ago.  Let me know if you see a case that can separate them but still be housed together somehow.

freaked1234, 2022-03-19:
@Rappoman @Photo64 sadly this is not the case for me. I have one sensor 2m apart from the ESP32 and one of them is 20cm away from the ESP32. The only way to get rid of the false positives for me is to flash ESPHome one the same units in the same place.

DTTerastar, 2022-03-20:
I use a PIR HAT on my M5StickC plus and it works great.  I don't see how this issue has anything to do with the firmware.  We just parrot the value of the pin.

freaked1234, 2022-03-20:
@DTTerastar the only thing I can tell you is that I have 2 ESP32, one in my living room with 2m of cable from the ESP to the PIR and the other in my bathroom with 20cm of wire. On ESPresense I get constant false positives, when flashing those 2 ESPs with ESPHome, without changing anything else, I get no false positives. So I dont know what else causes this and it seems that its not only me.
If there is any information I can provide to sort this out I am happy to help!

Photo64, 2022-03-20:
I had asked in some other places and had this as a response -> "Those sensors need a debounce added. Not sure if that code is capable but I’d try to add a second or so before the actual input is triggered. A debounce is a way to keep the input from chattering on and off rapidly."  
A little out of my realm but if it makes any sense maybe it could be added??

btw - I had separated mine but again only about a 2 or 3 inch separate and while the false positives have significantly dropped I still do get a few so maybe the "debouce" would help(?)

DTTerastar, 2022-03-28:
Please test v2.1.1b0 and see if this improves things

Photo64, 2022-03-29:
I have been testing v2.1.1b0 and it has been way more stable with fewer false positives. When I first reconnected after updating to v2.1.1b0 I was getting what appeared to be some false positives every couple of minutes but after letting it settle a bit it seems to have stopped sending the false positives. I have the esp32 board and motion sensor in a 3D printed box together and they are separated by less than 2 inches. I'll continue to monitor the results however it does seem like a tremendous improvement from previously.

Rappoman, 2022-03-29:
@Photo64 care to share the stl for your case as I am getting loads of false positives when I attempt to use my case.  Even when using v2.1.1b0.

DTTerastar, 2022-04-02:
I have 2 of these sensors: https://ewelinkstore.com/product/sonoff-zigbee-motion-sensor-znzb-03/?v=fa868488740a in my bathroom near a heater vent.  The heat from the vent causes one of them to false trigger occasionally.  What i've done is this: ```state: >-
      {{ (is_state('binary_sensor.bathroom_darrell_motion', 'on') and is_state('binary_sensor.bathroom_heather_motion', 'on'))``` on my template sensors.  This requires both to activate, this reduces my false positives to zero, while still being super responsive.  You might want to try something like that?

Photo64, 2022-04-02:
> I have 2 of these sensors: https://ewelinkstore.com/product/sonoff-zigbee-motion-sensor-znzb-03/?v=fa868488740a in my bathroom near a heater vent. The heat from the vent causes one of them to false [...]

I am not sure how this prevents the esp32 from false triggering. When the attached motion sensor is too close to the esp32 or within the same box as shown above the false triggers occur every few seconds. They are not randomly occurring. Once they are separated as I did in the photo above the false positives almost do not occur. They unfortunately can not be within the same unit or the false positives happen. I have other sensors in units running tasmota and I receive no false positives. Currently the latest update that was made did allow the configuration of boxes shown above to mostly work and I currently have a few setup this way.

DTTerastar, 2022-04-03:
Electrically you must need a resistor to pull it up or down so it doesn't float.  That's my best guess. My change was to request the esp32 to enable the internal pullup (meaning pull the pin to +3.3).  I was using the input with a relay and without that the input just flapped up and down.

DTTerastar, 2022-04-10:
Closing because I do believe this is an electronics thing and not really about ESPresense.

closed 2022-04-10 by DTTerastar