ESPresense/ESPresense #413: Distance update delay in HA

harleyzhao, 2022-03-17:
**Describe the bug**
the distance in the home assistant is updated very slow and delayed. When the phone is already in the room and close to the esp, it still shows it is far. It get updated every 3 seconds and the distance get closer and closer and settle down to the correct  number. It seemed to be delayed quite a lot. 

I tracked the mqtt messages and it showed the esp send messages every aournd 3 seconds. 

Please help.

DTTerastar, 2022-03-22:
What's the interval for the sensor?  It's a property of the json now.  The current settings are tuned for around 500ms.

capandcamera, 2022-03-22:
The intervals in the JSON range around 350-371

capandcamera, 2022-03-22:
[image]

DTTerastar, 2022-03-22:
That looks good to me, so it's more on the end of Home Assistant updating mqtt_room?

capandcamera, 2022-03-22:
I believe so, the waveforms seem to show that the messages are being received in good increments, so it must be HA / mqtt_room thats causing the delay, Where is best to post for support on mqtt_room? (apologies if not here)

capandcamera, 2022-03-22:
I'm not 100% sure, I cleared the chart in MQTT Explorer to watch the data as it processes and I can see that the chart only updates every 5 seconds. Is this usual behaviour?

DTTerastar, 2022-03-22:
From https://espresense.com/configuration/settings#filtering 

> Skip reporting if not seen (in milliseconds) / report early if beacon has moved more than this distance. The basic logic is if i’ve reported this recently check out how long it’s been and how far we’ve moved. If it’s been > than the distance here it will report. If its over the skip reporting ms then we’ll send the report with whatever the current distance is.

If those are set default it'll update every 5000ms or 0.5m whichever happens first.  Try lowering 0.5 to 0.25

capandcamera, 2022-03-22:
The issue still persists, I've rebuilt the mqtt container, aswell as setting up a completely isolated mqtt container to ensure it wasn't delays due to zigbee2mqtt but that does not seem to be an issue.

I've tried to capture the delay:
`https://imgur.com/a/qky5H3e`

st is typed when i start moving from the living room, i go via the office and fn is typed when i reach the kitchen.

The rooms are quite small but i'm not sure that would pose a problem as i'd assume (possibly incorrectly) that the signal would be stronger and track easier.

onuralpp, 2022-04-21:
I have the same issue. Did anyone fix the problem?

capandcamera, 2022-04-22:
> I have the same issue. Did anyone fix the problem?

Nah, unfortunately I gave up, it was too unreliable.

closed 2022-07-11 by stale[bot]

ZerefDeLaValier, 2022-11-18:
I have the same issue right now. Did anyone fix this after so much time?

BlueBenson, 2023-10-26:
I think the problem is mqtt_room sensor, I use the node-red's mqtt-in node get the topic from mqtt server and make a new node-red sensor get the msg.payload.rssi infomation. All problems sloved. Very sensitive.