ESPresense/ESPresense #415: Would you consider making it compatible with D1 Mini's running the ESP 8266 Chip

soundslikeluke, 2022-03-17:
Hello,

I love esp presence and want to start putting them in every room, I was just wondering if adding support for D1 Mini's using the esp 8266 chip was in the table? I'm sorry I'm not even sure if it possible I just thought I would ask before buying more esp32s

needs-coffee, 2022-03-17:
Not possible - ESP8266 does not have any Bluetooth capabilities so this would be limited to ESP32 or future iterations.

soundslikeluke, 2022-03-17:
I feel silly thankyou so much

ozett, 2022-03-26:
i desperately tried to bring espresence to the esp32,
but failed.

than i discovered that the esp32-S2 variants also lack bluetooth. 
maybe worth mentioning on the install-page?

https://www.espressif.com/en/news/espressif-announces-%E2%80%A8esp32-s2-secure-wi-fi-mcu