ESPresense/ESPresense #416: Add support for M5 Stock Core 320x240 LCD Screen

kungfuice, 2022-03-18:
**Is your feature request related to a problem? Please describe.**
Add support for M5Stack CORE 2.0 320x240 IPS panel. The current M5Stick LCD support doesn't work out of the box with the M5 Core so you're forced to just use the ESP32 base firmware, which doesn't support the screen at all.

**Describe the solution you'd like**
I've created a branch with changes that support this LCD with a larger scrolling area than the M5Stick previously supported. The contents of the serial output can be presented to the user with information about devices coming and going.

**Describe alternatives you've considered**
I looked at just using the M5 StickC support that currently existed but this is a very small screen vs the M5 Core and it didn't really work all that well. At least the firmware loaded, but the overall experience was lacking so that's why I came up with my own solution.

**Additional context**

kungfuice, 2022-03-18:
I have a created a feature branch m5core-support with the relevant changes.

DTTerastar, 2022-04-10:
Cool I have one and can test a PR if you'd like

closed 2022-11-08 by stale[bot]