ESPresense/ESPresense #418: Multiple Apple Watch same ID and Error

dave-parsons, 2022-03-19:
### Discussed in https://github.com/ESPresense/ESPresense/discussions/417

<div type='discussions-op-text'>

<sup>Originally posted by **dave-parsons** March 19, 2022</sup>
My wife and I have identical iphone12s and identical Apple Watch series 7.   I'm using ESPresense version  v2.1.0.  
Id's for our phones work great, but I get the same id for our watches  "apple:1005:9-26"
I added apple:1005:9-26 to the query and get the following log.  
I don't know what the error means or if there's anything here to help me differentiate between apple watch devices.   
Can anyone share some advice?

```
` New   | MAC: 77825781f1c5, ID: apple:1007:11-7                                              781f9a50bad548
1 Query | MAC: 57a59b8246f4, ID: apple:1005:9-26                                              3ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
1 QryErr| MAC: 57a59b8246f4, ID: apple:1005:9-26                                              rssi -84, try 7, retry after 30000ms
1 Query | MAC: 67ed1a4329f1, ID: apple:1005:9-26                                              1ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
1 QryErr| MAC: 67ed1a4329f1, ID: apple:1005:9-26                                              rssi -63, try 19, retry after 30000ms
1 New   | MAC: e7e7e2ae584c, ID: sonos:e7e7e2ae584c 
````
</div>
This log entry after installing verbose firmware

[13:26:23]Verbose | apple:1005:9-26                                           MD: 4c0010052a181635ca
[13:26:23]Verbose | 445ce9dab56b                                              MD: 75004204018066445ce9dab56b465ce9dab56a01000000000000
[13:26:23]1 Query | MAC: 587f7a79dc54, ID: apple:1005:9-26                                              9ms
[13:26:29]E NimBLEClient: Connection was made but no response from MTU exchange
```
`

DTTerastar, 2022-03-19:
If they're both the exact same model query won't help.  That error is an open issue: https://github.com/h2zero/NimBLE-Arduino/issues/366

You can use the new counting feature to work around the lack of unique id for a watch: <https://espresense.com/configuration/settings#counting>

dave-parsons, 2022-03-20:
Thank you @DTTerastar.   I have the counting working but I don't really see a way to make that useful to decide which watch is where.   It appears the distances reported as some combination of the two watches.  Below example - iphone locations are fairly accurate and are co-located with the respective watches.  

[image]

DTTerastar, 2022-03-20:
The counting doesn't update the sensors.  It updates this: 
[image]

DTTerastar, 2022-04-10:
Unfortunately counting is the best we can do if the watches are exactly the same.

closed 2022-04-10 by DTTerastar