ESPresense/ESPresense #423: Query ids don't stick

nickelnine, 2022-03-23:
**Describe the bug**

I have 3 ESP32 nodes of which 1 doesn't honor the device ids I enter in the Query field. When I open the web page for the node, it will show either ON or OFF in the Query box instead of the device ids I had entered. No reboot, no changes just suddenly when an automation for that room stops working it's because the device id vanished and replaced with the above.

**Config**

- Version: 2.1
- Active scan enabled: Yes, Request scan results is enabled
- Include filter: none
- Exclude filter: none
- Query filter: apple:1005:9-12 apple:1005:9-8

**Hardware**

- Firmware flavor: the default OTA flavor
- Device (be specific): MH-ET LIVE D1 mini ESP32

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.
[image]

nickelnine, 2022-03-24:
I find it offers more reliability

nickelnine, 2022-05-24:
Still an issue. Query ids entered into the Query box disappear and replaced by literal string "ON".

DTTerastar, 2022-06-04:
Hmmm, I think there is something sending mqtt messages on your network that are conflicting.  Can you use mqtt explorer and look for a message that looks like espresense/rooms/*/query/set w/ the payload ON?

nickelnine, 2022-06-10:
Yes, currently I see a payload of ON in espresense/rooms/bedroom/query. I don't have a /set segment. My livingroom node is chugging along just fine with the query id I set but my bedroom node refuses to stick. I changed it manually yesterday but overnight it went to ON again.

nickelnine, 2022-08-03:
Not an issue since using IRK

closed 2022-08-03 by nickelnine