ESPresense/ESPresense #428: ESPresense not detecting my phone's bluetooth

pickonedev, 2022-03-30:
**Describe the bug**

I have a Huawei P30 Pro, with bluetooth activated and visible for other devices, but the ESPresense does not detect my phone. I'm at 0.5 - 1 meter distance, nothing. I found my phone's bluetooth mac address, but I can't get the ID, because the device it is not showing that the mac has been detected. I have even added the mac to the Known BLE, still no detection...

**Config**

- Version: v2.1.0
- Active scan enabled:
- Include filter:
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor: 
- Device (be specific): Huawei P30 Pro Phone

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.

DTTerastar, 2022-04-01:
Androids need an app to be detected: https://espresense.com/beacons/apps

pickonedev, 2022-04-01:
Yes, in the end, I used the Beacon from Home Assistant app... Thank you!

closed 2022-04-01 by pickonedev