ESPresense/ESPresense #433: esp32 crashing shortly after boot, requires manual restart

chrisron95, 2022-04-01:
**Describe the bug**

I've been having trouble for several versions now. I followed a couple similar issues but when they were resolved in an update, my problem persisted. When I plug in my esp32 loaded with ESPresense, it'll load up fine and start reporting to Home Assistant instantly. However, within anywhere from a couple minutes to an hour (no more) the device becomes unavailable in Home Assistant. I also can't reach the devicde through the web anymore and the blue onboard status light stops flashing. I've tried with several different versions of ESPresense, even tried beta (2.1.1b1). I've manually flashed them all multiple times through serial using ESPHome-Flasher. I have 5 of these identical devices and they're all doing the same thing. It seems to work perfectly fine, but then out of nowhere something halts it. From looking at the logs I don't see anything causing it, but I may be missing something. I attached logs below.

**Config**

- Version: [2.1.1b1](https://github.com/ESPresense/ESPresense/releases/tag/v2.1.1b1) (also tried with 2.1.0 and below)
- Active scan enabled: No
- Include filter: n/a
- Exclude filter: n/a
- Query filter: n/a

**Hardware**

- Firmware flavor: esp32-sensors
- Device (be specific): [esp32 - 30 pin](https://smile.amazon.com/gp/product/B08246MCL5/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1)

**Logs**
```
[20:19:00]Temp: 22.4'C Humidity: 48.0%
[20:19:06]PIR MOTION DETECTED!!!
[20:19:09]Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
[20:19:09]Debug exception reason: BREAK instr 
[20:19:09]Core 0Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
[20:19:09]Debug exception reason: Stack canary watchpoint triggered (K) 
[20:19:09]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
...last line repeated 120x...
[20:19:10]Core 0Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[20:19:11]Core 0ets Jun  8 2016 00:22:57
[20:19:11]
[20:19:11]rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:19:11]configsip: 0, SPIWP:0xee
[20:19:11]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:19:11]mode:DIO, clock div:2
[20:19:11]load:0x3fff0018,len:4
[20:19:11]load:0x3fff001c,len:1044
[20:19:11]load:0x40078000,len:8896
[20:19:11]load:0x40080400,len:5828
[20:19:11]entry 0x400806ac
[20:19:11]Connecting to WiFi (40:91:51:9A:DE:CC)...
```
(then nothing else until a manual reboot)

**Screenshots**

If applicable, add screenshots to help explain your problem.

chrisron95, 2022-04-01:
Re-flashed them all with v2.1.1b2, used ESPHome-flasher and erased the flash for each. They seemed to last a bit longer but still all went unavailable within an hour or so. I caught one while it was still connected to the computer, it seemed to be going through a reboot cycle this time but I didn't catch the logs to paste here.

I also tried with v2.1.1b3 on all, same result.

DTTerastar, 2022-04-02:
I figured it out, the ESP32S is a variant with one core.  We use both cores extensively.  The debug is because it's noticing we're trying to run something on the 2nd core which doesn't exist.

chrisron95, 2022-04-02:
@DTTerastar are you sure that's the case here? According to the [Amazon listing for these](https://smile.amazon.com/gp/product/B08246MCL5/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1), it says dual-core. Unless I'm looking at the wrong thing, which is possible.

DTTerastar, 2022-04-02:
The beginning of the listing says: Teyleten Robot ESP32S.  S=Single Core D=Double Core. <http://elektronikagratis.blogspot.com/2019/03/esp32-modules-and-boards-esp32-esp32d.html>

DTTerastar, 2022-04-02:
Since the stuff that works it completely on core 0, and the stuff that causes the issues is on core 1 is also a hint.

chrisron95, 2022-04-02:
Gotcha, thanks for the clarification. Well guess I'm off to Amazon for some new ones!

chrisron95, 2022-04-04:
@DTTerastar So I went with [these ESP32 Mini boards](https://smile.amazon.com/gp/product/B08MFCC4SR/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1) which are the same as [what's linked in the docs](https://amzn.to/3tlkK8D) (that listing wasn't available for me so I found ones that matched). The listing specifies `CPU: Dual core Tensilica LX6 (32 bit) Clock` so I'm pretty confident I got the correct ones. They arrived yesterday, I wired them up and flashed them, but I'm still getting the same problem. Here's the logs from the crash I caught while it was plugged in:

```
[11:26:17]BH1750 (452 ms): 46 lx
[11:26:19]Temp: 24.2'C Humidity: 55.5%
[11:26:20]BH1750 (453 ms): 45 lx
[11:26:21]BH1750 (452 ms): 46 lx
[11:26:23]Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
[11:26:23]Debug exception reason: BREAK instr 
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
[11:26:23]Debug exception reason: Stack canary watchpoint triggered (K) 
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:23]Core 0Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[... cut ...]
[11:26:24]Core 1Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:24]Core 1Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:24]Core 1Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:24]Core 1Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:24]Core 1Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
[11:26:24]Core 1Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
```

DTTerastar, 2022-04-04:
That one mentions Core 1, so it's definitely not the exact same.   But yes, it is similar.   Can I see your configuration?  Is there a very long room name, or some special characters?  Can you try flashing w/ an erase and just fill in wifi and mqtt?

chrisron95, 2022-04-05:
Here's my config, it's basically the same on all devices (simple names: Bedroom, Kitchen, Living Room, etc.). Some have a DHT22 instead of DHT11, but other than that all identical. Kept the MQTT creds off for security but that connection is good
[image]
[image]
[image]
[image]

I also tried loading it with the verbose logging version, this was the crash with verbose, surprisingly less crash info than before:
```
Verbose | apple:1007:11-8                                           MD: 4c001007351f0f9e018748
Verbose | 8c79f5f5f9a1                                              MD: 750042040180608c79f5f5f9a18e79f5f5f9a001000000000000
Verbose | sonos:4b042d50f2a0                                        AD: 0xfe07
Verbose | sonos:4b042d50f2a0                                        MD: a705051011000000000000021f00ca
Verbose | 245a4c5474e1                                              AD: d83b80e3-e81f-4f66-879b-779be5f6a693
Verbose | 245a4c5474e1                                              SD: 0x252a/245a4c5474da
Verbose | apple:1006:10-8                                           MD: 4c0010063a1a19abd422
Verbose | apple:1005:9-12                                           MD: 4c00100501144a8c09
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr 
```

And I also did ESP32 - sensors (non-verbose) but didn't fill out any sensors, this one was a little different:
```
[23:27:18]1 New   | MAC: 5138e0564847, ID: apple:1006:10                                                01190c527d38
[23:27:20]1 New   | MAC: 7d05d79e0c5d, ID: apple:1005:9-4                                               031c3d601e
[23:27:34]1 Del   | MAC: 5e4ef4e27008, ID: apple:1005:9-12                                              0618617807
[23:27:39]1 Del   | MAC: 42fa011927c9, ID: apple:1006:10-8                                              361a30cb008f
[23:28:02]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[23:28:02]Debug exception reason: Stack canary watchpoint triggered (ble) 
[23:28:02]Core 1 register dump:
[23:28:02]PC      : 0x40080080  PS      : 0x00060c36  A0      : 0x800fd294  A1      : 0x3ffee62c  
[23:28:02]A2      : 0x3ffbaa00  A3      : 0x3ffc9f44  A4      : 0x3ffee62d  A5      : 0x800fd0bd  
[23:28:02]A6      : 0x3ffe8820  A7      : 0x3ffe9390  A8      : 0x00000001  A9      : 0x3ffee7e4  
[23:28:02]A10     : 0x00000001  A11     : 0x3ffc9f44  A12     : 0x3ffee62d  A13     : 0x3ffbaa20  
[23:28:02]A14     : 0x3ffc9f44  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
[23:28:02]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[23:28:02]
[23:28:02]ELF file SHA256: 0000000000000000
[23:28:02]
[23:28:02]Backtrace: 0x40080080:0x3ffee62c
[23:28:02]
[23:28:02]Rebooting...
[23:28:02]ets Jul 29 2019 12:21:46
```

It then rebooted itself and is working alright, but that was only a little while ago so it could happen again. I'll keep an eye on it and update if it crashes again.

needs-coffee, 2022-04-13:
```Stack canary watchpoint triggered (ble)```  on your last log seems to be related to #394

DTTerastar, 2022-04-14:
Yes, the latest prerelease has a fix for `Stack canary watchpoint triggered (ble)`

DTTerastar, 2022-04-17:
The latest release has this improvement, has it made a difference for you @chrisron95 ?

chrisron95, 2022-04-21:
@DTTerastar sorry for the delay, haven't had any free time to try it out. Well, I flashed it this morning and so far so good. It's been up for a couple hours now and no crashes yet. Will let it run for a while more and comment here again later today if I notice anything but at this point looks like you may have solved it! Thanks for your hard work!

DTTerastar, 2022-04-24:
Great!  Close the issue when you know it's good!

closed 2022-05-02 by chrisron95