ESPresense/ESPresense #438: m5 stack atom (support for m5stack atom tail)

wifispray, 2022-04-04:
Maybe a bit of a long shot to post here.... Im using the m5stack, which is great. But having cables in each room is a bit of a nightmare. I hoped the atom tail might be able to be used with atom (and be able to view in HA). Is this something supported or planned to be supported?

DTTerastar, 2022-04-04:
The tail just power, right?  Not sure what we'd need to support.  We already support all the m5stack atoms.  I misnamed it -matrix, but it works on all m5atoms.

wifispray, 2022-04-06:
It’s is yes. More to monitor the usage in HA and what the battery charge
is. Seems to drain rather quickly, not sure it’s the best option

How do you power yours?

iCSpotRun, 2022-04-07:
> The tail just power, right? Not sure what we'd need to support. We already support all the m5stack atoms. I misnamed it -matrix, but it works on all m5atoms.

I wish I knew this earlier. I was flashing my non-Matrix Atom's with the Generic ESP32 and now not only does the LED work again, but it seems more stable. Suggest the web UI get's updated to avoid confusion and issues.

iCSpotRun, 2022-06-04:
I got ahead of myself there. I have several M5 Atoms that constantly halt and never come back, or reboot. I've flashed with both Generic and M5 Stack firmwares. Recent 2.1.2 update does not improve this.

DTTerastar, 2022-06-18:
I have seen one atom that is flakey, but not sure how to figure out what's its issue.  I have 5 other atoms that work great.

closed 2022-09-01 by DTTerastar