ESPresense/ESPresense #440: Constant reboot loop

ilium007, 2022-04-07:
**Describe the bug**

Adafruit Huzzah32 in constant reboot loop. I had managed to get this exact device to connect to MQTT yesterday, came back today and in constant reboot loop. Tried different power supply, its not a brownout reboot.

**Config**

- Version: 2.1.10
- Active scan enabled:
- Include filter:
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor: Huzzah32
- Device (be specific): Adafruit Huzzah32

**Logs**

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FST_FLSH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (24:6F:28:3:2:00)...
                                       Connecting to WiFi SSID 'LIBERTY2'...............192.168.10.66
Firmware:     adafruit-huzzah32
Version:      v2.1.0
IP address:   192.168.10.66
DNS address:  192.168.10.1
Hostname:     espresense-office
Room:         office
MQTT server:  192.168.10.45:1883
                                Max Distance: 16.00
                                                   PIR Sensor:   disabled
Radar Sensor: disabled
Query:
Include:
Exclude:
Known Macs:
Count Ids:
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble)
Core 1 register dump:
PC      : 0x400f850a  PS      : 0x00060236  0      : 0x800f8858  1      : 0x3ffe7a00
2      : 0x3ffe8600  3      : 0x00000001  4      : 0x00000001  5      : 0x9da557ff
6      : 0x3ffc9a9c  7      : 0x3ffba524  8      : 0x3ffba9e0  9      : 0x3ffe7c70
10     : 0x3ffe8777  11     : 0x00000001  12     : 0x3ffe2314  13     : 0x00000001
14     : 0x3ffc9a9c  15     : 0x3ffe877a  SR     : 0x00000020  EXCCUSE: 0x00000001
EXCVDDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SH256: 0000000000000000

Backtrace: 0x400f850a:0x3ffe7a00 0x400f8855:0x3ffe7a70 0x400f853a:0x3ffe7a90 0x400f867e:0x3ffe7b00 0x400f8855:0x3ffe7b70 0x400f853a:0x3ffe7b90 0x400f867e:0x3ffe7c00 0x400f8855:0x3ffe7c70 0x400f853a:0x3ffe7c90 0x400f867e:0x3ffe7d00 0x400f8855:0x3ffe7d70 0x400f853a:0x3ffe7d90 0x400f867e:0x3ffe7e00 0x400f8855:0x3ffe7e70 0x400f853a:0x3ffe7e90 0x400f867e:0x3ffe7f00 0x400f8855:0x3ffe7f70 0x400f853a:0x3ffe7f90 0x400f867e:0x3ffe8000 0x400f8855:0x3ffe8070 0x400f853a:0x3ffe8090 0x400f867e:0x3ffe8100 0x400f8855:0x3ffe8170 0x400f853a:0x3ffe8190 0x400f867e:0x3ffe8200 0x400f8855:0x3ffe8270 0x400f853a:0x3ffe8290 0x400f867e:0x3ffe8300 0x400f8855:0x3ffe8370 0x400f853a:0x3ffe8390 0x400f867e:0x3ffe8400 0x400f8855:0x3ffe8470 0x400f853a:0x3ffe8490 0x400f867e:0x3ffe8500 0x400f857d:0x3ffe8570 0x400f8879:0x3ffe85e0 0x400f89c5:0x3ffe8600 0x400f8ebf:0x3ffe8660 0x400f9049:0x3ffe86d0 0x400f9167:0x3ffe86f0 0x400f487d:0x3ffe8730 0x400f4a8f:0x3ffe87d0 0x400d309a:0x3ffe87f0 0x400e1a99:0x3ffe8810 0x400e4132:0x3ffe8850 0x400e8715:0x3ffe88c0 0x400e87c1:0x3ffe8900 0x400e8a0d:0x3ffe8920 0x400e7781:0x3ffe8940 0x400ee202:0x3ffe8960 0x400e1103:0x3ffe8980

Rebooting...
ets Jun  8 2016 00:22:57
```

**Screenshots**

If applicable, add screenshots to help explain your problem.

ilium007, 2022-04-07:
Tried both HUZZAH32 sensor and non-sensor installs. Installed via Chrome. It obviously connects to wifi and gets IP address as seen in the logs.

Same issue as here: https://github.com/ESPresense/ESPresense/issues/394

needs-coffee, 2022-04-08:
Are you able to build this on your computer in VSCode and install and then collect the backtrace if that installed build crashes? (see the BUILDING document in the repo). Using the build (firmware.elf) and the backtrace will help figure out what the cause of the crash is with this https://github.com/janLo/EspArduinoExceptionDecoder

ilium007, 2022-04-13:
This isn't a reboot look every few mins as per the other issue description. This does not get past first boot.

needs-coffee, 2022-04-13:
are you able to build with VSCode and provide a log with that build with the backtrace, if you are unable to decode the backtrace you can provide the ```firmware.elf``` file here from the .pio/build directory and it and i can decode the trace

do/did you get any uptime at all?

ilium007, 2022-04-13:
Ok - will do. I know I have had issue sin the past (esphome) trying to decode a backtrace. I get zero uptime. Instant reboot with that error.

needs-coffee, 2022-04-13:
I still think the errors are related with the ble stack overflowing but interesting that yours wont boot at all, if you try and build with the latest branch from source with VSCode the ble stack has now been increased to 8k from 4k so you might have better luck if this is the underlying issue for you

ilium007, 2022-04-13:
What is interesting and I forgot to mention, I deployed ESPrensense to this board the day before and it was all working. I decided to deploy to it again the next day but without sensors and this happened. Switching back was also unsuccessful. I have not tried to do a full chip erase (if even possible)

needs-coffee, 2022-04-13:
interesting - try erasing the chip before installing - it can be done in several ways but the easiest is with [esptool](https://github.com/espressif/esptool) which can be installed from pip.
The erase command would be ``` esptool.py --chip esp32 p com5 erase_flash```
replace com5 with the serial port/serial device of your esp32

ilium007, 2022-04-13:
yeah, I had assumed the Chrome install would have erased. Will try.

ilium007, 2022-04-13:
Ok... so the version I built and uploaded using platformio worked... no erase.

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FST_FLSH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Setup network
Connecting to WiFi SSID 'xxxxxx'...............192.168.10.66
Firmware:     adafruit-huzzah32
IP address:   192.168.10.66
DNS address:  192.168.10.1
Hostname:     espresense-office
Room:         office
MQTT server:  192.168.10.45:1883
                                Max Distance: 16.00
                                                   PIR Sensor:   disabled
Radar Sensor: disabled
Query:
Include:
Exclude:
Known Macs:
Count Ids:
1 New   | MC: xxxxxxx, ID: xxxxxxx
```

ilium007, 2022-04-13:
What is the esptool command to write the .bin file to flash? I have done an erase and want to try loading the 2.1.0 release from GitHub. Just don' want to put it in the wrong memory location

ilium007, 2022-04-13:
When I checkout the 2.1.0 branch and build it it now goes into endless brownout detector reboot loop - mentions canary issue

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FST_FLSH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Connecting to WiFi (24:6F:28:3:2:00)...

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FST_FLSH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Connecting to WiFi (24:6F:28:3:2:00)...
                                       Connecting to WiFi SSID 'xxxx'...............192.168.10.66
Firmware:     adafruit-huzzah32
IP address:   192.168.10.66
DNS address:  192.168.10.1
Hostname:     espresense-office
Room:         office
MQTT server:  192.168.10.45:1883
                                Max Distance: 16.00
                                                   PIR Sensor:   disabled
Radar Sensor: disabled
Query:
Include:
Exclude:
Known Macs:
Count Ids:
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble)
Core 1 register dump:
PC      : 0x400f82ba  PS      : 0x00060236  0      : 0x800f8608  1      : 0x3ffe7a00
2      : 0x3ffe8600  3      : 0x00000001  4      : 0x00000001  5      : 0x292f7f40
6      : 0x3ffc9a9c  7      : 0x3ffba524  8      : 0x3ffba9e0  9      : 0x3ffe7c70
10     : 0x3ffe8777  11     : 0x00000001  12     : 0x3ffe8e98  13     : 0x00000001
14     : 0x3ffc9a9c  15     : 0x3ffe877a  SR     : 0x00000020  EXCCUSE: 0x00000001
EXCVDDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SH256: 0000000000000000

Backtrace: 0x400f82ba:0x3ffe7a00 0x400f8605:0x3ffe7a70 0x400f82ea:0x3ffe7a90 0x400f842e:0x3ffe7b00 0x400f8605:0x3ffe7b70 0x400f82ea:0x3ffe7b90 0x400f842e:0x3ffe7c00 0x400f8605:0x3ffe7c70 0x400f82ea:0x3ffe7c90 0x400f842e:0x3ffe7d00 0x400f8605:0x3ffe7d70 0x400f82ea:0x3ffe7d90 0x400f842e:0x3ffe7e00 0x400f8605:0x3ffe7e70 0x400f82ea:0x3ffe7e90 0x400f842e:0x3ffe7f00 0x400f8605:0x3ffe7f70 0x400f82ea:0x3ffe7f90 0x400f842e:0x3ffe8000 0x400f8605:0x3ffe8070 0x400f82ea:0x3ffe8090 0x400f842e:0x3ffe8100 0x400f8605:0x3ffe8170 0x400f82ea:0x3ffe8190 0x400f842e:0x3ffe8200 0x400f8605:0x3ffe8270 0x400f82ea:0x3ffe8290 0x400f842e:0x3ffe8300 0x400f8605:0x3ffe8370 0x400f82ea:0x3ffe8390 0x400f842e:0x3ffe8400 0x400f8605:0x3ffe8470 0x400f82ea:0x3ffe8490 0x400f842e:0x3ffe8500 0x400f832d:0x3ffe8570 0x400f8629:0x3ffe85e0 0x400f8775:0x3ffe8600 0x400f8c6f:0x3ffe8660 0x400f8df9:0x3ffe86d0 0x400f8f17:0x3ffe86f0 0x400f462d:0x3ffe8730 0x400f483f:0x3ffe87d0 0x400d306a:0x3ffe87f0 0x400e1849:0x3ffe8810 0x400e3ee2:0x3ffe8850 0x400e84c5:0x3ffe88c0 0x400e8571:0x3ffe8900 0x400e87bd:0x3ffe8920 0x400e7531:0x3ffe8940 0x400edfb2:0x3ffe8960 0x400e0eb3:0x3ffe8980

Rebooting...
```

ilium007, 2022-04-13:
[firmware.elf.zip](https://github.com/ESPresense/ESPresense/files/8480728/firmware.elf.zip)

Backtrace above

ilium007, 2022-04-13:
Built and uploaded master again and it works fine.

needs-coffee, 2022-04-13:
```
PC: 0x400f82ba

stacktrace:
0x400f82ba: std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_M_dfs(std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_Match_mode, long) at //.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/regex_executor.tcc:197
0x400f8605: std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_M_rep_once_more(std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_Match_mode, long) at //.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/regex_executor.tcc:181
0x400f82ea: std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_M_dfs(std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_Match_mode, long) at //.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/regex_executor.tcc:217
0x400f842e: std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_M_dfs(std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_Match_mode, long) at //.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/regex_executor.tcc:294
0x400f8605: std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_M_rep_once_more(std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_Match_mode, long) at //.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/regex_executor.tcc:181
0x400f82ea: std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_M_dfs(std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::__cxx11::regex_traits<char>, true>::_Match_mode, long) at //.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/regex_executor.tcc:217
0x400f842e: std::__detail::_Executor<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_s [... cut ...]
  \-> inlined by: nimble_port_run at //projects/ESPresense/.pio/libdeps/adafruit-huzzah32/NimBLE-Arduino/src/nimble/porting/nimble/src/nimble_port.c:79
0x400e0eb3: NimBLEDevice::host_task(void*) at //projects/ESPresense/.pio/libdeps/adafruit-huzzah32/NimBLE-Arduino/src/NimBLEDevice.cpp:791
```
For reference decoded stacktrace from @ilium007 above.
FYI - @ilium007, your pc username is revealed in the firmware.elf file as it shows the full path to the source files,  i have redacted from the stacktrace above but you may want to remove the firmware file now if you want too keep that private

ilium007, 2022-04-13:
Thanks and done

needs-coffee, 2022-04-13:
> Built and uploaded master again and it works fine.

If you stay on the current master and change the config in platformio.ini to change the BLE stack size back to 4k do you still have the same issues as before?
change ```CONFIG_BT_NIMBLE_TASK_STACK_SIZE=8192``` to ```CONFIG_BT_NIMBLE_TASK_STACK_SIZE=4096``` and rebuild.

ilium007, 2022-04-13:
Same rebooting lop when setting master to `CONFIG_BT_NIMBLE_TASK_STACK_SIZE=8192`

ilium007, 2022-04-13:
Back on master with `CONFIG_BT_NIMBLE_TASK_STACK_SIZE=8192` and working fine again

ilium007, 2022-04-13:
All seems to be working but no data populating mqtt, topics all appeared but no data.

DTTerastar, 2022-04-17:
Based on the SUPER nice stack trace above I think the stack being increased will just let the madness the regex is attempting to do last a bit longer before ultimately getting the same fate.  I think some BLE device is somehow confusing NimBLE to buffer overflow a huge string, which my code tries to "slugify" with bad results.

ChirpyTurnip, 2022-05-03:
Hi,

Just for what it's worth.... On v2.1 on a genuine ESP 32 WROOM I got rock solid performance with literally continuous uptime from the moment it landed..it never rebooted. 

That is until I got updated to v2.1.1 via an OTA update a few weeks ago....now it reboots regularly (every few hours). :-(

Unless there's something weird about the way the OTA update came through then something is not quite right...

Any thoughts?

CP.

needs-coffee, 2022-05-06:
Hi @ChirpyTurnip 
Without logs of the crash ( and preferably a decoded stacktrace) it would just be guesses as to the cause, if you are able to attach logs i could have a look.

closed 2022-07-12 by stale[bot]