ESPresense/ESPresense #448: Groups

wifispray, 2022-04-14:
Hey @DTTerastar 

This is maybe more of an implementation question than FR, but here we go.

When implementing the solution to a multiroom environment where users may have multiple devices. Is there a way to group sensors together? Like if I am tracking (via ESP) my iPhone, watch and car keys can you group the devices together as one entity?

Im also keen to understand some best practices for this, maybe the above approach is not good practice. Something for the readme maybe?

Any advise would be warmly received

kirel, 2022-04-15:
I have a template sensor for a similar situation in home assistant. I track both my watch and my phone but the watch is not advertising when connected to the phone. So I assume I'm at the room of the watch unless it's not_home. Then I'm at the location of the phone.

DTTerastar, 2022-04-17:
I'm thinking about this.  I think I want to make something that works like the WLED Sync where all the ESPresense nodes talk to each other to get some extra intelligence.

closed 2022-11-05 by stale[bot]