ESPresense/ESPresense #450: Tractive dog collar not detected

denko81, 2022-04-18:
I have a Tractive dog collar witch is not found when i search after it. The device shows up in nRF Connect. 

**Config**

- Version: 2.1.1
- Active scan enabled: Yes/No
- Include filter:
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor:
- Device (be specific): Tractive dog collar

DTTerastar, 2022-04-24:
Can you show info about the advertisements it sends in nRF Connect?  Screenshots?

denko81, 2022-04-30:
[image]

It's the device at the bottom.

DTTerastar, 2022-04-30:
If you put f3ef8ef0acd1 into the known macs area: 
[image]
and then use device id: "known:f3ef8ef0acd1" in hass mqtt_room does it work?

denko81, 2022-04-30:
How do you mean use in hass mqtt_room?

DTTerastar, 2022-05-01:
```
sensors:
  platform: mqtt_room
  device_id: "known:f3ef8ef0acd1"
  name: 'Dog'
  state_topic: 'espresense/rooms'
  timeout: 60
  away_timeout: 120 # number of seconds after which the enitity will get status not_home
```

denko81, 2022-05-01:
I think I have tested it without success.

DTTerastar, 2022-05-01:
Has the mac address changed?

denko81, 2022-05-01:
It's not changing

denko81, 2022-05-07:
I'm sorry for the late reply but I have the known Mac option but it didn't work, the sensor is not showing up in Home Assistant.

I saw you made a pull request regarding this issue. Is it something that I can test?

denko81, 2022-06-05:
I'm still not able to find the collar.

DTTerastar, 2022-06-18:
I don't understand what your issue is.  ESPresense sees it and it work in MQTT Explorer but it doesn't work in HASS?  Have you followed: https://espresense.com/home_assistant ?

denko81, 2022-06-22:
No, I don't see it in ESPresense or MQTTExplorer.

DTTerastar, 2022-07-18:
If you hook up the serial console and bring the collar within 1 foot of the node, what do you see under Close | .  Please paste in the line.

DTTerastar, 2022-07-18:
If you see absolutely nothing when brought close then it's not advertising and we can't track it.  Sorry.

denko81, 2022-07-30:
I can see it in nRF however.

DTTerastar, 2022-08-04:
Can you show logs of esp32-verbose?

denko81, 2022-08-25:
How do I do that?

DTTerastar, 2022-09-01:
Hook usb to device, go to https://espresense.com/firmware pick Verbose and capture the output via https://espresense.com/terminal

PR #461 (2022-05-23): Add tractive
Fixes #450
closed 2022-11-08 by stale[bot]