ESPresense/ESPresense #452: ESP32 shows up at "disconnected" after a several days to weeks - autoreboot?

Anto79-ops, 2022-04-19:
Hi!

Thanks for the awesome script and work here! I just wanted to to know everyonce in a while my ESP32 shows up as disconnected on Home Assinstant, and I have to resort to unpluging and plugging it back in, and it works fine again. It seems like its just loosing wifi

Is there a way to include in the script the rebooting of the ESP32 say, every 24hrs or it would be nice if the reboot happened when it doesn't connect to Wifi anymore.

DTTerastar, 2022-04-24:
It does already reboot when it loses Wifi or MQTT.  It also has a watchdog that reset's it if it get's stuck. I have seen a few devices get a little too hot, or have out of spec capacitors actually get stuck.  Is it more than one board?  Is it a quality manufacturer? Can you hook it up to usb and capture a log of what's happening when it occurs?

Anto79-ops, 2022-04-24:
This is a good point your raise. They to get warm/hot, I guess I could measure the temperature? I bought 4 from the same place/vendor and they all do it so, if there is a hardware design, it would make sense. To be honest, I don't know the reputation of the manufacturer, i purchased all 4 of them on AliExpress, so I guess there's always a chance, here:

https://www.aliexpress.com/item/1005002410521023.html?spm=a2g0o.order_list.0.0.21ef1802Arh7Hl

and the actual photo here:
[image]

I can't tell if there is some incorrect manufacturing on it.

Either way, the log files will be more telling. I'll have to connect to a computer so that it runs 24/7, I guess the log files are not stored?

DTTerastar, 2022-04-24:
Yes, you'd have to hook it to a computer to get the logs.  Make sure you have a lot of lines of scroll buffer :)

Anto79-ops, 2022-04-24:
Ok, this could take a while. Its on the order of weeks as to when this happens. If you want, you can close this and I can repost later with my findings.

DTTerastar, 2022-04-29:
The auto-close bot will take care of it if needed :)

hellcry37, 2022-06-20:
To combat this (that happens on my devkit v1 sometimes, rarely to be honest), I've setup some USB switches with Tasmota on them Sinilink USB Switch Module (XY-WFUSB) , and setup an automation on home assistant as soon as it becomes unavailable, it cuts the power to the esp, waits 10 seconds and powers back on. Then waits 30 seconds and send me a message to verify everything is fine :)

DTTerastar, 2022-06-24:
The latest beta has a fix for a bad memory handling that might be behind this issue.  Please test it and see if it improves your uptimes.

Anto79-ops, 2022-06-24:
thanks for the information. I'd be happy to try the beta:

https://github.com/ESPresense/ESPresense/releases/tag/v2.2.0b2

just wondering, how to push or flash this to the ESP32? I used the web interface for the initial setup.

DTTerastar, 2022-06-25:
https://espresense.com/beta can do it, it doesn't support all types of firmware yet, or you can check the box to allow prerelease.

Anto79-ops, 2022-06-25:
thank you! Done. I've changed the settings to allow pre-release on 3 of 4 of the devices and restarted them. They are all V2.1.2 now, not sure when it will push the update. Time will tell now, keep you all posted here.

closed 2022-07-18 by Anto79-ops

DTTerastar, 2022-07-18:
Awesome!