ESPresense/ESPresense #453: Security for opening an entrance door

AxokGit, 2022-04-20:
**Is your feature request related to a problem? Please describe.**
I would like to set up a system to open my front door with the phone. In conjunction with Home Assistant, I would like to make an automation that triggers the lock to open.
I noticed that ESPresence distinguishes a device based on its ID. But this represents a risk of identity theft

**Describe the solution you'd like**
The solution would be to manage IDs dynamically and use rotating IDs, maybe on the same principle as Google Authentificator or something else. The goal is to prevent the issuance of an ID that could trigger a process by an unauthorized person.

**Describe alternatives you've considered**
Another alternative would be to change the ID every time the device has been detected or as soon as it leaves the detection area and this could be managed by Home Assistant.

**Additional context**
A system like this already exists with the unlocking of Tesla vehicles with the phone but I don't know on what principle this authentication is made. I only know that an application (-> Tesla) is necessary for the proper functioning of the process.

Thanks

DTTerastar, 2022-04-24:
I think this is a great idea, we would have to make it as a custom app, and have some sort of enrollment process but it would be a good way of having trackability in ESPresense while having no real trackability by other people.  It should work similar to covid tracking beacons, but we'd know the seed values for the hash rotation so we could actually figure out which device is which.  As a side benefit it'd also be much higher security so simple replay wouldn't work.

DTTerastar, 2022-08-30:
Apple has a "Car Key" program that maybe someone could investigate on this?

closed 2022-11-05 by stale[bot]

tehXor, 2022-11-25:
Sorry, I'm a total noob on bluetooth but since I'm looking for a solution for the same use case:

Is there no challenge response possibility for a default bluetooth paring? I mean there are these parings which I do have to confirm with a PIN given by the device I want to pair with. So till now I assumed that this is somewhat secure and no other device could simply "overtake" that paring and spoof to be the original device. And under that assumption no custom app would be required. We'd only need the possibility to actively pair with the ESPresense device and mark this pairing as trustworthy. Once our original device comes into ESPresense proximity it will automatically connect to it just like it would to a headset or whatever former paired bluetooth device, we can check RSSI and run or MQTT based automations accordingly...

(I'm writing this without even having an ESPresense device since I'm still looking for the best solution for exactly that use case. So if this is already possible, sorry again...)